jcl5m1/ventilator #10: Supply of Nano V3 Module

Brashdan, 2020-03-17:
As a biologist, it is amazing to see how doable this is. My question is how quickly could the Nano modules be acquired. The demand for a million is going to hit like a freight train possibly as soon as two to three weeks. Is there any way that these could be sourced or scavenged locally?  https://sccm.org/Blog/March-2020/United-States-Resource-Availability-for-COVID-19?_zs=jxpjd1&_zl=w9pb6 

We could easily need a million more after that with a month because the projection in the attached article is with only 20 million infected. I will try to dumb this down so a biologist like me can understand it and so Joe the plumbers can start to build them in their garages. But it looks brilliant to me.

peteflorence, 2020-03-17:
I doubt that Arduinos would be the limiting factor; there are tons of Arduinos (or clones, which work fine too) in the world.  Many different models of Arduinos (Nano, Micro, etc.) could probably all work.  I teach a workshop for kids with my friends and we have bought many hundreds of Arduinos in the past.  Don't know what the actual total supply #s though...

However the best thing to do, if somebody is reading this and would like to take it on,
would be to start to assemble numbers on supplies of all different parts.

And also starting to think through logistics.

reinier1, 2020-03-17:
Arduino is an open source product. So every can make then thereself https://www.instructables.com/id/How-to-make-your-own-Arduino-board/. Also if there is a problem with getting arduino's inside the Netherlands I can help with supplying arduino's I maybe have access to a number of arduino nano /uno range(100-150)

pedrombmachado, 2020-03-17:
@Brashdan ,
  I am happy to help in porting the Arduino source code to any controller board that you may have around you. I can literally program anything from FPGAs to workstations (even old computers).

sb43201, 2020-03-18:
Sourcing of the cloned Nano board is not a problem, the bottle neck is the CPAP motor.

Brashdan, 2020-03-18:
If you guys will help me program it, I am going to try to build one. And if
that works, may try to build more because it looks like there are going to
be a lot of people over 60 (like me) that will be not get critical care in
the hospitals due to lack of ventilators.  I suspect they will just let
their family watch them turn blue and die so there will be lots of people
that could be saved by the family taking their loved ones home to the
garage and using an ad hoc ventilator. Perhaps they will not allow that
though. But as the law stands now, they could not prevent a family from
doing that.