jcl5m1/ventilator #13: How can I help?

pedrombmachado, 2020-03-17:
@jcl5m1 I would like to contribute to the project. We need to rump up these ventilators. How can I help?

aliakanan12, 2020-03-17:
Hello , can we replace pump with rc brush motor and esc ? And 3d print box for motor ? Please help

pedrombmachado, 2020-03-17:
Would a brushless pump like this one work https://www.ebay.co.uk/itm/Water-Pump-280L-H-12V-DC-Brushless-Small-Submersible-Motor-Pump-Garden-Pond-DW/142841106466?hash=item2141fe5022:g:jZ4AAOSwyvtdDZ1W? Do we have a slack group for brainstorming and recruit more people?

aliakanan12, 2020-03-17:
i need to build an respirators device not cpap can u help me some info ?

pedrombmachado, 2020-03-17:
Can you join a new slack workspace that I created for enabling us to chat. https://join.slack.com/t/open-ventilator/shared_invite/zt-cq0c2o4x-gXJLb0sZ4dVTUPt3oZy61Q

pedrombmachado, 2020-03-17:
I will make you admin

jesusguzmanpy, 2020-03-17:
It would be very helpful in countries with few health resources. They could create a community to get a generic prototype with all the details.

danchitnis, 2020-03-17:
UCL (London, UK) is hosting a low-cost ventilator design challenge to be manufactured in UCL and deployed in NHS and globally. UCL runs large hospitals in London including Royal Free which is dedicated to such incidents. Please see [here](https://docs.google.com/forms/d/e/1FAIpQLSftJuZDxgx3dp3jv3ab-i1Xiz7jtUe9pxuI0XXgOcaJd_cJ3w/viewform).

nasocializes, 2020-03-25:
@pedrombmachado feel free to take on one of the to dos.

nasocializes, 2020-03-30:
@pedrombmachado have you seen [issue 62](https://github.com/jcl5m1/ventilator/issues/62) that discusses a basic pump? 

And to dos are now also listed in the [project kanban](https://github.com/jcl5m1/ventilator/projects)