jcl5m1/ventilator #14: Project Clearification

TheAmazingJarvis, 2020-03-17:
I'm a little confused on what the goal is. Are we trying to turn CPAP machines into respirators or are we making respirators from scratch?

axjjienn, 2020-03-17:
Both.

aliakanan12, 2020-03-17:
@axjjienn can we use this project as respirators ?

axjjienn, 2020-03-17:
Ventilators and respirators are essentially the same.

https://medlineplus.gov/ency/patientinstructions/000458.htm

ajcrow, 2020-03-18:
Positive pressure ventilation is tricky.  It.was devoped by Forest Bird based on aircraft oxygen systems developed for bomber aircraft in world war 2. 
Negative pressure ventilators  AKA Iron Lung date from the 1920's.  Much simpler to manage.  They were developed to replace the function of a paralyzed diaphragm due to polio.  Not sure if useful to lungs filled with fluid.

ajcrow, 2020-03-18:
1 these devices need to be able to be operated by minimally skilled personnel (assuming a shortage of respirators also comes with a shortage of therapists.)  Leave intubation to the professionals!  Can cell phone provide computing power?
2 effective to assist lungs filled with fluid (edema).  This is to treat patients with acute respiratory distress syndrome from COVID 19.  Need to understand what those patients need.
3 need to be easy to manufacture.  Could 3d printing create parts?  Can be pneumatic powered.

ajcrow, 2020-03-18:
World health organization has a webpage on acute respiratory distress syndrome with COVID19. CPAP machines are recommended with masks for mild cases.

nasocializes, 2020-03-30:
Project to dos are now also listed in the [kanbans](https://github.com/jcl5m1/ventilator/projects)