jcl5m1/ventilator #16: Professional Insight?

aris-t, 2020-03-17:
So I hit up a friend with connections to people who actually make ventilators and these are some areas of concern they brought up.

- "This seems like a very very dangerous endeavor"
- Inflating patient’s lungs correctly (including being in sync with their respiratory drive) 
- Weaning the patient off support later is a super complex thing with no room for error. 
         Though I believe this really only becomes an issue after ~24 - 72 hrs of use, (Needs confirmation)
- They also pointed out that this wouldn't be a true ventilator as you usually sedate and intubate patients in that case.
- They were also concerned with the potential for the positive pressure gas exiting  the patient to contain aerosolized virus.

mrquincle, 2020-03-17:
What's their suggestion? Can they open source their hardware? Can they ramp up their production enough? Can they under FDA have other companies helping them to ramp up?

peteflorence, 2020-03-18:
### Re: project comparison with other projects
1. The ideal scenario would be to ramp production of FDA approved ventilators.
2. There are parallel efforts to this project to do exactly that.
3. Given the possibility that these efforts might not meet global demand, then we should explore low cost, easy-to-manufacture ventilator options.  Keep in mind this will be a global need, not just for privileged countries like the U.S., etc.
4. Parallel to this project, there are other low cost open source projects as well.

Given the importance of each day, some level of parallel efforts seem great.

### Re: technical questions
1. Regarding intubation with an endotracheal tube -- could somebody point to if we know that COVID patients actually need an endotracheal tube?  This overview on ventilators (https://www.youtube.com/watch?v=gk_Qf-JAL84) points out that there are two different use cases for ventilators.  One (1) is that the patient can't protect liquids and solids from going down the trachea.  Two (2) is that they can't breathe on their own.  If the requirement is only for (2), it seems possible that an endotracheal tube is not required.  Need doctor to advise.
2. If an endotracheal tube is needed, it could be added to the design.
3. "They were also concerned with the potential for the positive pressure gas exiting the patient to contain aerosolized virus." -- yes, need to make sure this is solved for and tested.
4. "Inflating patient’s lungs correctly (including being in sync with their respiratory drive)" -- it depends on the mode of operation of the ventilator.  If the patient initiates the breath, then yes the event of the patient starting to inhale needs to be sensed.  Note though that not all modes of operation require this.  See https://www.youtube.com/watch?v=gk_Qf-JAL84 for more.

jcl5m1, 2020-03-18:
The material I find online is very bipolar.  This person falls in the "if you don't do it right, it's a terrible idea".  The other camp, is bolting pieces of wood together to make something that might help.  There's not much middle ground of helpful information which is "in an emergency, these are really the things that matter...".

Bag-Valve Mask Ventilators don't seem to require intubation to be effective. 

If the environment already has a lot of infected patients coughing, won't there already be a lot of aerosolized virus?

When I tested it, it was very easy for me to match the rate driven by the machine.  How important is it for the machine to match the person vs the other way around?

peteflorence, 2020-03-18:
FWIW -- I am hearing from doctor friends that in severe cases, intubation is required, but not always.  Don't have data on the ratio.

fireb0lt, 2020-03-18:
Pressure support ventilators can be used with or without intubation. They can help prevent pneumonia from Covid19

Arif-I, 2020-03-18:
hi, I am from Bangladesh and following your project. The situation is going worse here and we'll def need more breathers. Even if this works with 40 or 50 % efficacy, that will be great help. I appreciate the effort of the community.

allen-munsch, 2020-03-18:
> "They were also concerned with the potential for the positive pressure gas exiting the patient to contain aerosolized virus." -- yes, need to make sure this is solved for and tested.

https://en.wikipedia.org/wiki/Venturi_scrubber

acab45, 2020-03-18:
Re: pandemic ventilator- i am not a programmer - It is my first time on github and i am a bit lost- would someone, anyone, please call me ANYTIME asap. Anthony Cabrera md, frcsc---Tel: 1-905-923-1766 OR whatsapp: 34-653-08-58-41   (34 is for Spain but i live in Hamilton, ontario, canada) OR give me your number and location and i will call you back- FYI: email: acab45@yahoo.ca
thank you
Anthony Cabrera md, frcsc

aris-t, 2020-03-18:
So i started this thread as an attempt to stir up some technical considerations which we may be overlooking but a more experienced practitioner would point out right away. I was fully expecting the response ive been getting from my more conventional contacts of "if you don't do it right, it's a terrible idea". I don't subscribe to that im much more for if you can help at all you should. Therefore im reaching out to some less conventional but more open minded contacts now.

Overall seeing @peteflorence and @jcl5m1 comments this thread seems to be serving its purpose. Keep up the good work.

rzeimer, 2020-03-18:
i suggest that providing devices that can match ventilators is not fruitful. rather we should get feedback on what are the specifications that need to be added to CPAP like machines to be useful for the cases that are not severe, which will make a large population and put pressure on the resources.
we could then use existing CPAPs which abound

allen-munsch, 2020-03-19:
Also checkout : https://gitlab.com/TrevorSmale/OSV-OpenLung

acab45, 2020-03-19:
https://www.amazon.com/Ventilator-Book-William-Owens-ebook/dp/B07D3142ZM/ref=sr_1_8?keywords=RESPIRATORY+THERAPY&qid=1584587296&sr=8-8

https://www.amazon.com/GREENS-RESPIRATORY-THERAPY-Practical-Respiratory/dp/1593309341/ref=sr_1_2?keywords=RESPIRATORY+THERAPY&qid=1584587393&sr=8-2

I AM CURIOUS IF ONE OF YOU WERE TO BUY THE BOOKS, HOW COULD YOU ALL ACCESS IT TO WORK ON IT.? IF YOU WOLD LIKE TO CALL ME BACK TO EXPLAIN PLS DO SO.

Thank you,

Anthony Cabrera
------------------------
This information is directed in confidence solely to the person named above and may contain confidential and/or privileged material. This information may not otherwise be distributed, copied or disclosed. If you have received this e-mail in error, please notify the sender immediately via a return e-mail and destroy original message. Thank you for your cooperation.

TimVanDyke, 2020-03-19:
I've been talking to acab45 and he also shared this video with me and I thought it might be a bit helpful. I think what we need to know here is basically what the use case we need to fulfill of a ventilator is, correct? 

as in, we need something along these lines (**Please correct me if I'm wrong**): 
ex:
Patient has corona virus and needs a ventilator in AC (or whatever) Mode.
they need:
- a minimum of x breaths a minute
- to monitor from u pressure to v pressure at y level precision
- alarms for a, b, and c, scenarios
- etc

acab45, 2020-03-19:
this book seems better for what you are tring to do:    open free textbook;

The Process of Breathing – Anatomy and Physiology    

 
 
 
   
The Process of Breathing – Anatomy and Physiology

acab45, 2020-03-19:
VENTILATOR PROJECT-------------i will try to find a collaborative  RT in Grand Rapids, MI)

https://www.google.com/search?q=specifications+for+a+ventialtor&oq=specifications+for+a+ventialtor&aqs=chrome..69i57j33.5460j1j7&sourceid=chrome&ie=UTF-8

https://www.sciencedirect.com/book/9780721601861/mechanical-ventilation

--->   http://www.kmscl.kerala.gov.in/ratecontractspec/Ventilator%20ICU.pdf

https://www.venteclife.com/assets/media/1064/vocsn_tech-specs_current.pdf

https://www.nxp.com/docs/en/application-note/DRM127.pdf

---------------------------------------------------------------------------------------------------------------------------------------------------------------------------

A SEPARATE BUT VERY IMPORTANT PROJECT  "facial air flow system" (very easy for you guys to design using off the shelf items, yet so critical for front line health staff in the context of "diy when supplies run out" ). I am working on the gown n hood...see images far below

i need someone who understands eletronics to design /explain how to connect a common drill battery (ie ryobi ) to a fan blower unit and along this wire have a little rehostat turn knob to adjust flow. i will send a diagram of this tomorrow. 

IMPORTANT:--->  Whatsapp for me is the best way to be able to send you all sorts of videos, images, audios, answer questions via audio back n forth  etc from my smartphone. We can setup a group on whatsap so that i can send all of you information at once then you can curate it and put it up on github as you see fit..
my whatsapp is:  

create a new message inside whatsapp :
1)-remove default canada and select SPAIN in drop down  
2)-enter 653085841 and
3)-send a test text

Thank you,

Anthony Cabrera
------------------------
This information is directed in confidence solely to the person named above and may contain confidential and/or privileged material. This information may not otherwise be distributed, copied or disclosed. If you have received this e-mail in error, please notify the sender immediately via a return e-mail and destroy original message. Thank you for your cooperation.

acab45, 2020-03-19:
https://twitter.com/charliekirk11/status/1240449295116931072?s=20

If the above is true and confirmed by other researchers, your ventilation machine may not be necessary 2 weeks from now. 
This malaria drug exists in major quanities across the USA and can be secured overnight  given the emergency powers to cut red tape by Trump decree could start tx very soon for this indcation

ac md
------------------------
This information is directed in confidence solely to the person named above and may contain confidential and/or privileged material. This information may not otherwise be distributed, copied or disclosed. If you have received this e-mail in error, please notify the sender immediately via a return e-mail and destroy original message. Thank you for your cooperation.

jcl5m1, 2020-03-19:
I had a long call with a pulmonologist yesterday.  Here are my notes:

https://github.com/jcl5m1/ventilator/wiki/Notes-from-chatting-with-a-pulmonologist

acab45, 2020-03-19:
thanks i will look at those notes you sent me.
-----------------------------------------------------------Good news. Impressive results from a study out of France showing: (excerpt:
".......At day 6 post-inclusion, 100% of patientstreated with hydroxychloroquine and azithromycin combination were virologicaly curedcomparing with 57.1% in patients treated with hydroxychloroquine only, and 12.5% in thecontrol group (p<0.001)...."
These two medications have been routinely carried in pharmacies for many years.The above was a modification from a successful study done in China many weeks ago so it seems its the real thing.
---------------------------------------At first thought, you might wonder if it is worth continuing with your project. The answer is for sure yes. If not in this country, the poorer developing countries would benefit as well as future viruses. I think world leaders have realized, that when it comes to critical aspects of healthcare, manufacturing your own (ventilators, masks etc) it is a matter of national security. I admire and am inspired by the effort of your group on Github, undaunted in the face of complexity. I really need to find you guys a RT.

Thank you,
Anthony Cabrera md, frcsc------------------------This information is directed in confidence solely to the person named above and may contain confidential and/or privileged material. This information may not otherwise be distributed, copied or disclosed. If you have received this e-mail in error, please notify the sender immediately via a return e-mail and destroy original message. Thank you for your cooperation. 

 
 

https://github.com/jcl5m1/ventilator/wiki/Notes-from-chatting-with-a-pulmonologist

acab45, 2020-03-19:
I did a quick scan of the notes, it strongly appears he is the real deal.  As you know this area of medicine is far outside my scope. I find his comments encouraging and supportive of what you are trying to do. 
ac md------------------------This information is directed in confidence solely to the person named above and may contain confidential and/or privileged material. This information may not otherwise be distributed, copied or disclosed. If you have received this e-mail in error, please notify the sender immediately via a return e-mail and destroy original message. Thank you for your cooperation. 

 
 

https://github.com/jcl5m1/ventilator/wiki/Notes-from-chatting-with-a-pulmonologist

allen-munsch, 2020-03-19:
Some techinical specifications can be found here : 

[Specification For RMVS Challenge.pdf](https://github.com/jcl5m1/ventilator/files/4355316/Specification.For.RMVS.Challenge.pdf)

acab45, 2020-03-19:
https://ca.finance.yahoo.com/news/manufacturers-offer-to-produce-medical-supplies-to-help-with-coronavirus-194049973.html

Thank you,
Anthony Cabrera------------------------This information is directed in confidence solely to the person named above and may contain confidential and/or privileged material. This information may not otherwise be distributed, copied or disclosed. If you have received this e-mail in error, please notify the sender immediately via a return e-mail and destroy original message. Thank you for your cooperation.

acab45, 2020-03-19:
It is quite possible the canadian government that is funding the ventilator manufacturing push may end up keeping it open source. I have no idea. Perhaps during the process of creating the device they might file novel patents and create a company to sell all over the world. No idea. I am not sure how intellectual property issues will pan out.

Thank you,
Anthony Cabrera------------------------This information is directed in confidence solely to the person named above and may contain confidential and/or privileged material. This information may not otherwise be distributed, copied or disclosed. If you have received this e-mail in error, please notify the sender immediately via a return e-mail and destroy original message. Thank you for your cooperation.

acab45, 2020-03-20:
A big automotive manufacterer Martinrea here in Ontario appears to be all in on ventilators
https://www.google.com/amp/s/business.financialpost.com/transportation/canadian-auto-suppliers-in-talks-to-build-ventilators-masks-during-coronavirus-crisis/amp 

   
   - The content of this email is confidential and intended for the recipient specified in message only. It is strictly forbidden to share any part of this message with any third party, without a written consent of the sender. If you received this message by mistake, please reply to this message and follow with its deletion, so that we can ensure such a mistake does not occur in the future.

maday, 2020-03-23:
Johnny, I appreciate what you are trying to do and want to help.  Posting this so that I can be part of the thread.

MordiSacks, 2020-03-23:
This might be helpful
https://www.youtube.com/watch?v=mnIpD1VwyMo

I have been following this channel for a while, very very good information

RWL-69, 2020-03-29:
Hello, all.  

I have not had a chance to look at everything linked in this thread but I wanted to make some comments.

My thought on this looks at why this project is so important at this time.  A growing number of cases and a lack of hardware in many countries.  As the cases increase that require a ventilator, the choice between one that provides all the bells and whistles and one that keeps someone alive disappears.  There is even discussion in the USA about automatic DNR's for victims due to lack of equipment.

I agree with the idea of making one that can be built from automotive parts as a basic system as these parts can be grabbed from the street if need be.  Adding in the various monitors and sensors is possible given a long enough lead time.

A summary sheet of what the limits are for a basic unit to a wish list of what would be nice for a complete unit would be a handy tool.  A chart with check boxes might be the way to go.

If some of this could be attached to Wiki for the project, that would be a great reference.

MordiSacks, 2020-03-31:
Medtronic is sharing its portable ventilator design specifications and code for free to all

https://techcrunch.com/2020/03/30/medtronic-is-sharing-its-portable-ventilator-design-specifications-and-code-for-free-to-all/

ErichBSchulz, 2020-04-05:
Hiya - I'm a medical doctor (anaesthetist/anesthesiologist depending on where you live)- I've been collating note here:

happy to collaborate.

https://docs.google.com/document/d/1sdrKYQ0mDOu4bJum6Fx6piRutIJovo7UqFKYHHxUD5A/edit?usp=sharing