jcl5m1/ventilator #17: Other Similar Projects

aris-t, 2020-03-17:
A quick run around the internet yielded that other people are working along similar designs. 

Seems hackday has some good insights as usual
https://hackaday.com/2020/03/12/ultimate-medical-hackathon-how-fast-can-we-design-and-deploy-an-open-source-ventilator/

https://hackaday.com/2015/09/25/hackaday-prize-semifinalist-individualized-breathing-apparatus/

Some other top links:
https://www.instructables.com/id/The-Pandemic-Ventilator/

https://innovationmanagement.se/2020/03/15/open-source-respirator-and-low-cost-ventilator-efforts-to-fight-coronavirus-covid-19/

http://letmegooglethat.com/?q=open+source+ventilator+projects     (its a more concise link than a search result)

peteflorence, 2020-03-18:
Yes some parallel efforts will be good.  This project may be more mechanically simpler than other projects.

Here are additional links, this is a 2010 paper from MIT and BU, there is work ongoing based around this design: https://web.mit.edu/2.75/projects/DMD_2010_Al_Husseini.pdf

Here's a project last year from Rice: https://www.youtube.com/watch?v=1t2t8d8xtD0&feature=youtu.be

The above two use bag-valves.

Many more linked here: https://www.facebook.com/groups/670932227050506/about/

allen-munsch, 2020-03-18:
found this as well: https://app.jogl.io/project/121#about

aris-t, 2020-03-19:
Re post of @allen-munsch from over on the professional insight thread.

Also checkout : https://gitlab.com/TrevorSmale/OSV-OpenLung

zathras, 2020-03-27:
Hi all,

Here's something I'm working on, through JOGL:  https://app.jogl.io/project/151 .  I'm specifically making an Android app to provide the chart display.  If there's interest, I could adapt that and make it available for other hardware configurations.

My contact info is on my Github page.