jcl5m1/ventilator #4: Filtration for Intake

axjjienn, 2020-03-17:
This is a great project, and I am really excited to help. One thing I note is that it is likely we would need some kind of filter on the respirator intake, so that any kinds of allergens or other harmful particles in the environment are not transferred to the patients lungs. Just a thought.

jcl5m1, 2020-03-17:
Thanks for mentioning this.  I added a few more images and 3D files for filters I had from a precursor project.

TheAmazingJarvis, 2020-03-17:
Could it be important **where** the filter is placed? Contaminates not just from from ambient particles but particles coming form, for example, lubricants from the motor or little bits of printer filament. 

Maybe one filter at the air intake and one at the mask?

axjjienn, 2020-03-17:
I was thinking the same thing. In order to reduce parts, it makes the most sense to put the filter on the outlet of the pump. It will potentially allow contaminants to enter the pump, but I think the pumps can take it.

aris-t, 2020-03-17:
You guys may want to take a look at some of these options:
Cheap:
https://www.directhomemedical.com/cart/merchant.mvc?Screen=PROD&Product_Code=1-H1605-inline-bacterial-viral-cpap-filters&Store_Code=DHM&gclid=EAIaIQobChMIo-G35omi6AIVSrzACh29hwv3EAQYASABEgK9FPD_BwE

https://www.noinsurancemedicalsupplies.com/main-flow-bacterial-viral-filter/?sku=AV-7178&gclid=EAIaIQobChMIo-G35omi6AIVSrzACh29hwv3EAQYBiABEgKaUvD_BwE

Expensive and in demand but the real thing:
https://www.medtronic.com/covidien/en-us/products/mechanical-ventilation/filters.html

These info sheets on the above link's products may at least give some specs to work from 
https://www.medtronic.com/content/dam/covidien/library/us/en/product/ventilator-filters/dar-sales-sheet.pdf

Overall looks like most designs could be rigged up with a 3D printer or PVC fittings and an alternate filter material

jcl5m1, 2020-03-17:
added a link about inline filters to README  thanks!

richovercash, 2020-03-17:
Agree on the inline filter, and issues re impurities coming from a commercial motor. Also the NIOSH filter is VOC compliant which might be overkill, though it does offer an easily sourced mount.

HitoJima, 2020-03-18:
Needs to filter  VOCs, you dont want to allow any additional problems.  Everything post filter should be manufactured in ‘clean room’ conditions if possible.

richovercash, 2020-03-18:
I've got a concept for an inline filter here: 
https://1drv.ms/b/s!AjCTtwReRUpmiZMWODGYKpWxrK-bRA

Uses the 3M bayonet mount, 3M gasket, and 3M P95  filter. Its just a concept and I need some ME help making it work. Got a folder of 3D files ready to upload

richovercash, 2020-03-18:
[image]

Basic Concept

richovercash, 2020-03-18:
[image]
Uses existing P95 filter

clc, 2020-03-22:
Here's a configurable inline filter holder that I made:
https://www.thingiverse.com/thing:4233776 - should be able to take a wide variety of filters

jcl5m1, 2020-03-22:
that's really nice.  what are you thoughts about just making the tapered
press fit ends for the tubes. to get rid of the bayonet?
 also, do you think it would be possible to align the long axis of the
filter to be more parallel with the air flow (not making the tube come out
perpendicular from the filter).  That way more of the filter could be used
it might be more mechanically secure?

[image: image.png]

jcl5m1, 2020-03-22:
I guess the filter doesn't have to stay flat either.

clc, 2020-03-22:
How about:
[image]
for making the filter parallel with the air flow?

Same design file, just with tweaked parameters (increased hepa thickness, decreased hepa length/width).

For the "tapered press ends" are you talking about making the ends that connect to the tube more conical so they will slip into the tubes more easily or did you have a particular design in mind?

clc, 2020-03-24:
Updated https://www.thingiverse.com/thing:4233776 with a tapered end for the tube connector + made tube connector height a customizable property.

[image]

chakahamilton, 2020-03-24:
> You guys may want to take a look at some of these options:
> Cheap: [...]

I'm currently focusing my efforts on a PAPR   to use the Universal Inline Bacterial Viral Filter for CPAP Machines.

oiaohm, 2020-04-10:
The bad news here is that the intake filters as in the filter after the air source and before the mask should not be 3d printed.   These need to be molded  plastic.   3d printing like it or not makes a fairly rough surface that bacteria can in fact love being on.

Please note using silicon or injection or clay molding can all produce clean surfaces.

I am not sure how much airflow each Universal Inline Bacterial Viral Filter for CPAP is designed for.