jcl5m1/ventilator #5: Efficient No Power Airway Shunt

axjjienn, 2020-03-17:
I think with respect to task (6) it is easy to have some-kind of spring loaded shunt, the problem would be making it power efficient. A potentially efficient way to do this would be a small servo which opens a hatch on the airway, that triggers when the power level reaches a certain "low" point. The issue with this is that it functionally decreases the power that can be run from any one-battery, as the "low" power point needs to be still above the complete discharge point of the battery, such that sufficient power remains to trigger and operate the airway shunt fail safe. It might be possible to create a pneumatic hatch, which opens when the flow rate drops below a certain rate, but these typically only work in continuous flow systems, which I am not sure we have here. Also, when starting the system, this hatch would need to be manually shut until steady state conditions are met.

jcl5m1, 2020-03-17:
interesting idea, I like the passive approach.  Perhaps some sort of mechanical latch where if the internal pressure drop down to ambient, the tube pops off.  only under high internal air pressure does it hold secure.  If i was a 3D printable element that would be great.

TheAmazingJarvis, 2020-03-17:
How about a simple flappy door? Airflow pushes the flappy element out of the way. No airflow and the flappy falls down.

[image]

omid-s, 2020-03-17:
> How about a simple flappy door? Airflow pushes the flappy element out of the way. No airflow and the flappy falls down. No electronics.
> 
> [image]

I was thinking of a same model but I was not sure if back pressure from breathing out could mess this up?

TheAmazingJarvis, 2020-03-17:
> I was thinking of a same model but I was not sure if back pressure from breathing out could mess this up?

I was wondering the same thing. One way to find out is to print one and try it.  Wish me luck.

TheAmazingJarvis, 2020-03-18:
> these typically only work in continuous flow systems, which I am not sure we have here

Therein lies a big question. When converting a CPAP into a respirator, would it stay on the whole time as they do now? 

(For people reading this who don't know the C in CPAP stands for Constant.)

TheAmazingJarvis, 2020-03-18:
I hooked it up to my CPAP machine and it works pretty darned well.

nbenci, 2020-03-18:
Use Tesla valve