jcl5m1/ventilator #6: Discussing sleep cycle detection using only the motor's current

Tagge, 2020-03-17:
Hello,
I will try to provide the beginning of an answer to that question. I am not intimate with Arduino nor the exact functionning of the human breathing apparatus, but I'll try to provide the beginning of an answer.

When I first pictured it, it didn't seem feasible at first, because I didn't know if the motor would be directly affected by the process of exhalation. Turn on a servo motor, and rub your thumb in a way that it opposes resistance to its movement. Increased resistance will increase power consumption, within the servo motor's parameters of course. But it has been increasing because you have opposed effective resistance. In the case of a servo motor, an increase in air pressure would not change power consumption (its profile has barely any air resistance), but we'll come back to this later.

I assume the fan is a simplified DC motor with blades hooked onto it, which it probably is. The air flow we're creating goes directly in a tube connected to the patient's breathing apparatus with a mask that has a decent seal. Now, that's where I might be wrong, but that's how I picture it :
- When the patient exhales, pressure temporarily increases in the mask : that's how the air is getting out through the fins in the mask (pressure differential)
- Increased pressure in the mask makes it harder for the fan to pump air in, as there is resistance to the air getting into the mask, therefore pressure also increases in the tube, all the way down to the fan. 
- Increased pressure around the blades increases air resistance, therefore the energy required to spin the blades at the same speed is now higher than it was when pressure was normal. 

Now, I assume that the Arduino can detect these subtle changes in energy consumption by the motor. Therefore, I think it's theoretically possible to detect the different steps of the breathing cycle. Now a lot of what I've said is based on my own recollection of my engineering and physics classes. I might have got some things wrong, but I'm convinced this is more or less how it works. I'm waiting for people who might be slightly more qualified to confirm or infirm what I just said. 

Sum-up : We should be able to detect the steps of the breathing cycle by measuring the fan's power consumption, higher consumption means exhaling, lower consumption means inhaling.

peteflorence, 2020-03-17:
Yes

Tagge, 2020-03-17:
I will try to suggest a system to measure the current on the Arduino board, and will then integrate the logic where appropriate in the .ino file.

jcl5m1, 2020-03-17:
yes, the motor speed is definitely being backdriven by the breath.  Which almost certainly affects the current draw, which should be readable with a sense resistor and the analog pin on the Arduino.  I haven't had a chance to measure it yet.  And you'd have to separate the driven current and the backdriven current somehow.  I don't have any immediate grate ideas.  Will look into collecting some sample data.

fireb0lt, 2020-03-18:
FYI, some masks are vented such that exhalation occurs through the mask rather than back through the blower

sb43201, 2020-03-18:
We can use air pressure sensor to directly measure the air flow pressure in the system instead of using the motor current.