jcl5m1/ventilator #8: DIY CPAP Style BLDC

ntwallace, 2020-03-17:
In addition to exploring off the shelf 12VDC pump motors, it seems worth recreating a CPAP setup with 3D printed/off the shelf parts. Mark Rehorst's [printable blower](https://drmrehorst.blogspot.com/2018/04/the-mother-of-all-print-cooling-fans.html) (specifically v2) + a [cheap BLDC](https://www.amazon.com/DYS-Brushless-Outrunner-Multirotor-Quadcopter/dp/B077HYQLMP) looks like a good option.

Diodeus, 2020-03-17:
Huh. I always expected the blower to be a positive-displacement vane pump.

jcl5m1, 2020-03-17:
yes, this should work.  Will add a link to the README.

necro-nemesis, 2020-03-20:
It appears a centrifugal pump has been used for this application. I concur with exploring the use of off an off the shelf brushless DC motor and 3d printed pump solution. I can see limited access to sufficient CPAP pumps being a criticial factor. Brushless DC motors though should be more readily available as is access to 3d printed parts.

wrt to the controller wonder it there would be benefit in using an Nodemcu vs Arduino in order to add in wifi capability possibly for monitoring and ota programming.

I don't have a CPAP pump in hand to try and 3d model and adapt a brushless dc motor to the design. If anyone comes up with drawings I can work to adapt a brushless motor to I'd appreciate it. Also specs on the motor would be beneficial. I am scouring the internet now looking for something I can work with.

Other designs I have looked at incorporate 3 electrically controlled check values with the pump inflating a diaphragm. Not certain this is requirement if the Arduino can reasonably moderate the flow but in other designs it appear on exhale the airflow goes out a separate route through a value unobstructed. Would there be the need to add a separate route? Is it possible to stop the centrifugal pump rotor at a position in it's rotation during exhale to allow a passageway in the pump to be open at that point in rotation?

https://www.cpap-supply.com/Articles.asp?ID=134

necro-nemesis, 2020-03-20:
This design may be worthy of consideration. https://www.thingiverse.com/thing:39755

mrehorst, 2020-03-20:
I couldn't find the exact specs on the original motor used in the CPAP blower I copied, but it was a 24W motor.  I can modify my CAD file to accommodate any motor you might have, can scale up the rotor and housing for more air flow, too.  

That Thingiverse design looks like it's made to pump water, and uses a brush-type motor- bad news around O2 because of sparks.  Brush type motors wear out quickly.  In the CPAP blower, the motor was located inside the blower with the moving air which probably helped to keep the motor running at a low temperature.

necro-nemesis, 2020-03-20:
It was designed for air. You have a CAD design for the blower on the go?

mrehorst, 2020-03-20:
https://a360.co/2I2Jz3Z

necro-nemesis, 2020-03-20:
Yes for this to work I believe the approach has to be brushless motor for various reasons.

necro-nemesis, 2020-03-20:
> https://a360.co/2I2Jz3Z

Brilliant I'll have a look. How can I assist you?

necro-nemesis, 2020-03-20:
I also code and am conversant with Arduino.

mrehorst, 2020-03-20:
The motor linked on amazon is rated 1120kV, 2-4S.  If you use 14V to drive it (~4S), you'll only get it up to 16k rpm or so.  The motor from the CPAP machine was at least 2000kV.  The impeller needs to spin at 20-25krpm at maximum output (to match the CPAP blower- but was that sufficient output for a ventilator?).  You want to find a motor that's rated for 2000-4000kV, and uses 3S or 4S battery, preferably with ball bearings for long operating life.

necro-nemesis, 2020-03-20:
That sounds like a typical drone motor 2500kV if memory serves. I assume the power requirements of the motor will dictate the supply voltage and the Arduino would be powered at 5vdc using a buck converter off that supply to keep things simple. I was able to bring in your CAD files to work with them. I need to see if the motor is capable of 20-25kpm sustain operation. Unsure on this aspect.

mrehorst, 2020-03-20:
I think this motor might be good:  https://hobbyking.com/en_us/multistar-v-spec-1304-3100kv-multi-rotor-motor-ccw.html  Not sure if the CW or CCW version is better- one of them has a left hand threaded prop mount...

necro-nemesis, 2020-03-20:
Ha ha. You were looking over hobbyking at the same time I was. The issue with kv is that it doesn't take into account load so I'm not sure at this point how to calculate and apply the load to the equation. It may come down to some experimentation with a prototype pump. Should I commence building one from the drawings? I'm sure there will be several iterations built to get it adapted to ultimately the motor required. I have a couple here and an ESC to give me a baseline.

necro-nemesis, 2020-03-20:
I have to dig. I have motor for an RC electric plane that might be suitable for testing. It has a single channel ESC with it I can hook to a receiver and run through it's paces on a bench. The drone motors I have on hand are 2250kv and 2600kv but I have a feeling this other one is bigger if I recall. Hard to imagine a CPAP machine would be as potent as what these motors deliver but possibly. I just have no basis to go by.

necro-nemesis, 2020-03-20:
@mrehorst is there a bearing in the housing or are you just utilizing the bearings within the motor?

mrehorst, 2020-03-20:
Those two parts are the motor assembly.  The one on the bottom is the piece that the discs used to screw onto, and now the printed impeller screws onto it.  I think I modeled them separately because I was considering separating them to mount the impeller directly on the motor shaft, which I never did.  There are no extra bearings- just those that are built into the motor.

necro-nemesis, 2020-03-20:
For my the proof of concept I was consider removing those two components. Sizing the inner diameter of the motor hole + clearances then using the screw in base mount configuration and collet used for propellers. The base and collet could be modeled in or just considered required items. See pic.
[image]

mrehorst, 2020-03-20:
That sounds reasonable.  I'd keep the motor body inside the blower to help cool the motor, and just add a hole for the wires.  The hole can be sealed with a drop of silicone once the motor is installed.

necro-nemesis, 2020-03-20:
There are vanes on the tops of motor cases as the case rotates with the spindle. Some appear not to have vanes. I can look at having slots down the sides of the motor mount area to allow for air to flow around the motor. I am also assuming some pressure will be generated by the impeller that will leak between the impeller and the case and push down along the motor. In the process of re-drawing the base atm. Hope to post something reflecting this concept shortly.

necro-nemesis, 2020-03-20:
I'm also going to have to make several design changes to make this 3d printable. I am hoping to arrive at a design that will not require support material. It will take a few iterations to get this broken down into that but that is the end state I am aiming for.

mrehorst, 2020-03-20:
[The procedure I used to copy the impeller is here.](https://drmrehorst.blogspot.com/2019/02/designing-3d-printed-parts-for.html)

The outrunner type motors are built inside out.  I don't think you have to do anything special to cool them other than keep the motor inside the blower housing.  There's probably enough air movement there to keep it cool.

I broke the original design into multiple pieces so that no support material would be needed.  The nozzle prints as a separate piece so that different size nozzles could be used.

necro-nemesis, 2020-03-20:
[image]
[image]

I can't upload an .stl to the comment area to allow you to manipulate the drawing. There's a 3d repo under master branch. Suggest we either clone and PR it or start a separate repo where the designs can be shared.

mrehorst, 2020-03-20:
I find it best in situations like this to model the motor first, then design the printed parts around it.  I wouldn't bump out the housing that way, I extend the bottom of the housing to keep the whole motor inside it.  Those slots on the sides are going to leak a lot of the air you're trying to push with the unit.

Can you post a model of the motor, or just a drawing with dimensions?

necro-nemesis, 2020-03-20:
The motors seem to vary in length. The one I model around is 35mm tall from RC plane. The quadcopter ones appear to be 18mm tall. Both seem to be 28mm in diameter. To model the motor I just used a cylinder. I need to do some research to see if 28mm diameter is standard but yes a shorter motor would not protrude as far. 35mm is on the long side of these motors. I am thinking the motor house should thread into the case for two reasons. First this would allow for clearance adjustment and secondly it appears that this would allow the design to meet the no support critera. Let me see if refactoring for a shorter motor can achieve the requirement to not protrude from the case.

mrehorst, 2020-03-20:
I thought you had a specific motor in-hand to try out.  I guess it doesn't matter how long the motor is- just move the entire bottom of the case downward and you won't need any support material to print it.

necro-nemesis, 2020-03-20:
[image]
[image]

Yes I'm prototyping around a motor in hand. At any rate here's a couple of pics using a shorter more drone specific motor to appreciate the proportions. This brings up the question of how universal to make the pump design and leads back to my original observations. The base plate cross pattern for the screws is common. The diameter's appear to be relatively common; additional clearance can be added to accomodate. The height of the motor may vary depending on which motor is acquired. Recommend the motor mount thread into the case and that motor mount designs of different lengths be available or the length of customizable simply by changing the length in CAD. If this seems satisfactory then I will go about drawing the thread for the motor to screw into the case.

necro-nemesis, 2020-03-20:
[image]
[image]
[image]

fireb0lt, 2020-03-20:
Excellent work so far! Most ventilators use Micronel blowers, could we learn anything from their design? The trick is getting high static pressure. https://www.micronel.com/products/u-line/

necro-nemesis, 2020-03-20:
[image]
[image]
After doing some additional research I came across a design that was built and tested using a brushless RC motor. I suggest moving down this avenue. The drawing files and test footage are available.

necro-nemesis, 2020-03-20:
[image]

necro-nemesis, 2020-03-20:
https://www.youtube.com/watch?v=LQTW7RxqRWc

https://www.youtube.com/watch?v=-jO_tKHpZec

hobodrifterdavid, 2020-03-20:
Hello, I'll just dump some notes here, maybe you find it helpful.

Nice overview of quadcopter motors:
https://oscarliang.com/quadcopter-motor-propeller/

>> wrt to the controller wonder it there would be benefit in using an Nodemcu vs Arduino in order to add in wifi capability possibly for monitoring and ota programming.

I think it's a good idea to have a ventilator controller MCU running code that is as simple to reason about as possible (having less code also helps). This means you can be more certain that you bug-free software. Software for flight control is written without any interrupts, as far as I know. You have simple loops, where tasks are done in sequence, and you can calculate the maximum possible running time for each task, so you won't run out of CPU time. Running a Wifi stack on FreeRTOS, for example, it's a pretty complex system. I made a proposal here: https://docs.google.com/document/d/1KpPWYRb1A637Cp5KIhhKtuGFS8Cdm5fGra3TTwwrBKo/edit#
I suggested having an control MCU, that can send and recieve simple binary messages over a UART. You could then connect an ESP32 up, and that could relay messages over Wifi or ethernet.

>> Also specs on the motor would be beneficial.

It's a bit tricky that drone motors come in all kinds of sizes. Best if a very common size can be used, as these will be easier to source. A notes:

CPAP motor is rated for 25W (Mike's blog).

See page 244 here: 
https://www.aarc.org/wp-content/uploads/2014/11/19802-001-F-LTV-1200-and-1150-Ops-Manual.pdf
It's the specification for a centrifugal compressor ventilator. It says:
Nominal Power Draw. Startup: 66 watts. Running: 36 - 48 watts
By the specification, most any small quadcopter motor should be able to put out enough power, in terms of watts.

Bigger motors *may* have larger bearings that will deal with sustained use better. I think it's not easy to buy lousy bearings these days, it's a solved problem. But, 25k RPM, day in/day out..

Bigger motors have larger casings and may be able to dissipate heat better.
"when the motor overheats, the coating on the coil will start melting and causing electrical shorts inside the motor."

I see there are two formats for screw patterns on the base, 16x16 and 16x19mm. Easy solution, support both by putting holes for both in the turbine base, with one pattern offset 90 degrees to the other.

There are different propeller mounting arrangements on the motors. Perhaps choose M5 threaded shaft with locking nut? Four screws would also be ok, if the pattern is standard across motors. These little motors can put out terrific power though, that could tear appart 3d printed parts that have screw holes too tightly spaced together. A metal plate on top of the 3d printed part, that spreads the pressure from the screw heads over the surface of the 3d printed part, could help. In the case of the M5 shaft, a large washer could work.

RE no. of poles: "Different sizes motors have different number of poles, 22XX and 23XX motors generally have 12 poles and 14 magnets." Good if we can stick with one type, as the motor controller won't need to be configured for different types.

Regarding cost, I was hoping your design could be used in project I had in mind, described in the google doc (improved $50 ventilator, https://docs.google.com/document/d/1KpPWYRb1A637Cp5KIhhKtuGFS8Cdm5fGra3TTwwrBKo). If the motor is $10 or less online, retail, that's on target.

Browsing banggood.com, 2207 seems very common and inexpensive, on gearbest.com.. 2205 seems at the top. Perhaps a design that can accomodate either of these sizes (with a shim at the base of the motor?). Lot's of 3D models available for these motors: https://grabcad.com/library?utf8=%E2%9C%93&query=2205

Regarding aerodynamic design of the turbine, there's lots of information online. There are scaling laws about how to scale up a rotor design. I'm very rusty on this stuff. Here's a complete course for example: https://nptel.ac.in/courses/101101058/
And there are lots of textbooks.

However, there's a lot to pick up. If you aren't already familiar with it, here's a suggestion. Order one of every suitable compressor design you can find (ok, within reason). Mark knows the next step :-). Don't shed too many tears, you can be fairly sure the rotor design is a product of the same process. Then, benchmark the different rotors, in terms of output, noise, efficiency etc. Please allow me to contribute/cover the costs, email me on davidzweig@hotmail.com

>> I assume the power requirements of the motor will dictate the supply voltage and the Arduino would be powered at 5vdc using a buck converter off that supply to keep things simple.

I think there's a good argument for operating at 12V, as it's easy to find 12V lead acid motorcycle battery with 1km of anyone who would need a $50 ventilator. On the other hand, Laptop power supplies are fairly good quality and everywhere, I think they output 18-22V usually. So not sure. I think maybe 12V is a better bet.

Cooling. Drawing air though the motor is important. The entry to the turbine is an area of low pressure.. behind the rotor, high pressure. You can use this to draw air through the motor, with the right design. Try make something that works with different motor can designs.

Focus on the design of the turbine 'bowl' and rotor.. the outer casing we can consider later as part of the overall design of the ventilator. For know, the outer casing could be minimal, to speed up printing. Maybe adjust overall dimensio [... cut ...] oot/center of the rotor, where it attaches to the motor. Add extra material here, use stress relief radiuses, and do FEA analysis, if you know how. I can also help with this later.

Is vibration an issue btw?

I suggest building a couple of devices, setting them up in front of a webcam, and running at 200% anicipated maximum load, and 400%, for example. Leave them running as long as possible, (probably noisey). See how it fails.

I hope there is something useful here. Thank you for your efforts.

necro-nemesis, 2020-03-20:
[image]
I've modified the base design to make it printable without support by removing unneeded details and ensuring there isn't uprintable overhangs.

hobodrifterdavid, 2020-03-20:
If you want to discuss on slack, I created a group. There are some people knowledgeable about aerodynamics and CFD on helpfulengineering.slack.com , they could be invited too.
https://join.slack.com/t/open-ventilator/shared_invite/zt-cwh6py1x-C5tO4JewMUf2tRN7rwWB~g

I'm trying to gather useful information here in doc, that allows people to get quickly up to speed. It's publically editable, you can record any important results or findings if you like:
https://docs.google.com/document/d/1KpPWYRb1A637Cp5KIhhKtuGFS8Cdm5fGra3TTwwrBKo/edit#

necro-nemesis, 2020-03-20:
I'm nearing the point of prototyping a unit. If you wouldn't mind I will press on as I'm spending most of my time with this design. I would like input on the impeller as the housings for centrifugal pumps appear to be rather consistent but if there's any recommendations regarding the impeller that is where I could use assistance. The remainder of my work is simply adapting the RC brushless type motor to the case which is nearing completion at this time. What has not been established is the required output of the pump. At least with a prototype in hand I can appreciate what may be required to obtain the required output.

necro-nemesis, 2020-03-20:
BTW the libs for various WiFi functionality, webserver, wifimanager etc. on an esp8266 are very fleshed out due to significant interest in it's use in IoT. I have used them and they have proven to be extensive, easy to use and reliable. If it's unnecessary complexity then I would say leave it but I certainly wouldn't be reluctant to use those libs.

hobodrifterdavid, 2020-03-20:
Sure. 
Impeller design, I would use Mark's technique with existing designs, unless you are familiar with the design of turbomachinery. I'm not, really.
Required output, I have read on a comment on helpfulengineering.slack.com , is 240L/min at 40cm H20. This really needs some research, looking at the specifications for ventilators and other soruces. Some effort will need to go into creating tooling to measuring pump output.

necro-nemesis, 2020-03-20:
I intend to 3d print this design to get some data from it. I have that capability.

hobodrifterdavid, 2020-03-20:
Ok. The controller mcu is maybe a different issue, can be discussed elsewhere.  I'll dissapear now and try help another group. Thanks.

necro-nemesis, 2020-03-20:
It is. I saw enough action on that front that I put myself to this task.

hobodrifterdavid, 2020-03-20:
One more relevant specification here: https://www.britishchambers.org.uk/media/get/Specification%20For%20RMVS%20Challenge.pdf

>> Deliver at least 400ml of air/air 02 mix in no more than 1.5 seconds. The ability to change the
rate at which air is pushed into the patient is desirable but not essential.

necro-nemesis, 2020-03-21:
[image]

necro-nemesis, 2020-03-21:
Removed extra bits a bobbles that didn't add to functionality. Trimmed the shell where it was disproportionately thick at the top. Re designed the top flat thus printable and stripped extra outer rings to reduce print time. Base has the legs removed flattened on the bottom. I've resized the dimensions to accommodate the RC motor I have on hand for testing purposes. Radius add for lofting to overhangs iot to print out without support. Time will tell if this will print without support or require any additional modification to to design. My experience tells me it should.
[image]

hobodrifterdavid, 2020-03-21:
I was trying to think of a good way to measure the flow rate. Pressure is easy with a column of water. 

Maybe a 'peak flowrate meter' could give some idea.. cheap enough on Amazon.
[image]

necro-nemesis, 2020-03-21:
Marks design (mrehorst?) would be faster to replicate. My concern with the design is wrt to the impeller in relation to the volute casing. Here's a pic of using transparency to showing how the impeller is located in relation to the case. Maybe Mark can chime in on the design and I can continue to work with it. I was ready to rework the impeller when I noted how much volume was open below it figuring he was still working the design.

[image]

hobodrifterdavid, 2020-03-21:
It's a working design, check out some posts on his blog: https://app.slack.com/client/TUTSYURT3/C010AM8S0D6

The extra space.. perhaps it helps reduces pressure fluctuations at the ouput a bit, by having a larger volume. I don't think it harms anything, the air that passes the rotor blade is only going one place.. the output. :-)

necro-nemesis, 2020-03-21:
I'd like to hear more about that decision. I would have thought the impeller would sit lower and occupy more of the open volume within the volute casing. Where's James Dyson when you need him?

This:

[image]

necro-nemesis, 2020-03-21:
[image]

mrehorst, 2020-03-21:
When I copied the CPAP blower, I copied the whole thing- the impeller (including the slope of the blades and the slope of the "roof", the case/volume, etc., with changes made to accommodate the motor I used.  The original impeller was thin and fit in the top cover with a large, empty volume below it (except for the motor).  

This thing is spinning at 20-30krpm.  If the printed blades are tall, they may break due to centrifugal force.  In the final design, there had better be something that will capture shards of plastic if the impeller disintegrates.  You wouldn't want it blowing plastic bits into the patient's lungs...

I don't know how to design a turbine of this sort, but it isn't hard to believe they may have compromised output volume/pressure for quiet operation- it's for a machine that is used by people when they are asleep.

I have a couple other blowers I can take apart, photograph, and model just to get some idea of how they look.  I'll be back in a couple hours.

necro-nemesis, 2020-03-21:
I see. So based on a working design the open area below the impeller was already incorporated in the design. I'm thinking the only enhancement to the rotor may then be to leave the blades stout but more closely contour them to the volute casing profile. I know it axial flow turbines this case clearance is critical to performance but this is a centrifugal air pump it probably trades off flow for pressure and may be more reliant on centrifugal force to than creating a tight tolerance to the case. If you think there's merit in recontouring the blades to the case profile I can give that a shot and leave the rest as it stands with the motor adaptation. As was stated earlier the lower area would ultimately exit out mind you there may be some explanation wrt to turbulent airflow which would likely have been modeled and tested for mass production of such a pump.

What are all the laterally placed holes in the case for? A specific mounting application?

necro-nemesis, 2020-03-21:
FWIW I prefer your design as it meets a critical requirement in terms of time to reproduce. The other design does not even with me trimming the design down.

hobodrifterdavid, 2020-03-21:
@necro-nemesis I see you're really fast with CAD. Would you be interested in helping me develop concepts for the overal packing for the $50 ventilator? I made a channel, here we can discuss. I'll just take a bit of downtime first. https://join.slack.com/t/open-ventilator/shared_invite/zt-cwh6py1x-C5tO4JewMUf2tRN7rwWB~g

necro-nemesis, 2020-03-21:
I was. I'm troubleshooting my computer atm. I'll let you know when I'm up again. Software issue. Looks to be Windows update related. Way prefer Debian but my CAD is on this piece of crap.

necro-nemesis, 2020-03-21:
Ok. fixed my issue. Something with roaming data corruption it appeared. I'm back up and running. Joined Slack. With mrehorst's input on my impeller concern I believe the course of action I would like to take is the follow:

-Complete the RC motor mount redesign using the motor I have in hand using the stout impeller of the initial design.

-Examine where I can trim the design to reduce material and print time. (possibly post prototype)

-Revise where necessary to print without support material.

-Commence building of prototype to collect data on operation.

-Source a suitable anemometer to measure flow.

*If collected data provides satisfactory results.

-Model various suitable common RC motors.

-Generate versions of mount to accommodate a range of motors.

mrehorst, 2020-03-21:
I took apart another CPAP machine made by ResMed.  The blower is quite small and power input for the machine was 12V @ 2.5A, so the motor would probably be no more than 18-20W.  The impeller design is quite different from the first one that I modeled and printed.  The motor is an out-runner that is exposed to the moving air inside the housing.  I've been messing around for an hour trying to model the case - it's a snail-like spiral that I can't figure out how to model with any accuracy.  The impeller diameter is 52 mm and the inside diameter of the housing is 64.8 mm, so there's lots of room between the impeller and the case.  The input port (the hole in the top cover) is 17.4 mm.  The exit port is 15.5 mm ID, but it's flared a bit- I think the diameter of the exit at the inner surface of the housing is about 9.5 mm, but the shape is complex.

Here's a CAD model of the impeller and motor:  [https://a360.co/33BGX5u](https://a360.co/33BGX5u)

Pictures here: [https://drive.google.com/open?id=1hQaCXwzunUPwm1Hkt-IuBgTkKlh1zWyR](https://drive.google.com/open?id=1hQaCXwzunUPwm1Hkt-IuBgTkKlh1zWyR)

necro-nemesis, 2020-03-21:
Can you take pic of the case convert it to an SVG, clean up the SVG sufficiently then import that. Extrude the 2d SVG to use to give you some proportions to model around? You may have to do that in two dimensions of X, Y or Z axis to get the proportions right. I assume the snail like spiral is going to require some lofting between two separate circles or different diameters around an center axis.

mrehorst, 2020-03-21:
You can generate a coil in Fusion360, but the problem is matching the shape of the blower housing.  It may not be all that critical anyway, since we don't have original specs on the blower.  Approximating the housing without the coil will still work. I can generate that in a few minutes.

The impeller design doesn't really lend itself to 3D printing- I think you'd have to print it in two pieces, one mostly flat with recesses to receive the blades, and the other flat with blades, and cement them together.  Not sure I'd trust that to hold together spinning at 30k rpm.

Car painting and spray tanning use something called HVLP turbine pumps to move lots of air.  It might be better to think in terms of hooking 10 people up to one blower than to make a separate blower for each person...

hobodrifterdavid, 2020-03-21:
Hi, we started a dicussion on packaging on open-ventilator.slack.com , 'design-packaging' channel.

,

necro-nemesis, 2020-03-21:
@mrehorst while you work out the other pump internals I am focusing on parts that can be reused for mounting an RC motor which can be incorporated into whatever design. I have this collet arrangement in mind given the relatively high rpm of the stator. I will just print the collet to ensure it will print as it's rather finely detailed. I may have to make it more reliant on the flexibility of the material and reduce the level of detail to make it printable.

[image]

[12 more replies]