jcl5m1/ventilator #21: any one can help how can connect wires to arduino and esc ?

aliakanan12, 2020-03-18:
via whatsapp +96176653802

aris-t, 2020-03-18:
How you physically connect up to the ardunio is based on what you have available.

Conventional prototyping methods are the use of 22G wire and solder. of course this is simplified by an electrical soldering iron and electronics solder. You can either solder pin header to the ardunio and other components as possible and use a breadboard to make the connections or just solder wires and components together.

Soldering technique: https://www.youtube.com/watch?v=QKbJxytERvg
Breadboarding and perfboarding: https://www.youtube.com/watch?v=w0c3t0fJhXU

In a real pinch you could likely use a screwdriver and a fire in place of the electric soldering iron and solder alloy could be made using lead and tin melted in a ceramic vessel over fire and then poured.
Allow ratios: https://fctsolder.com/eutectic-solder/
Video: https://youtu.be/T_MdxdGr57k?t=176

Less reliable techniques mostly involve using pressure to hold a contact in place. and this can be done by just crimps, clips, wires or even tape. of course these are less reliable than a solder connection as they can come undone.

As in theory this for a life critical device i suggest the most robust method you can which would be soldering. In any method care should be taken to protect the connections from disruption, Also fire is hot and lead is toxic. So use this advice at your own risk.

aliakanan12, 2020-03-18:
no i mean @jcl5m1 project how can i connect wires

aris-t, 2020-03-18:
ok then im a little lost as to what you mean by "connect wires"?

Are we talking how to make a physical connection: see above
Are we talking about which wire goes where: see [circuit diagram](https://github.com/jcl5m1/ventilator/raw/master/images/schematic_20200317.1.jpg?raw=true) 
or something else?

aliakanan12, 2020-03-18:
yes i mean circuit diagram im not understand how can i connect to each other please if you can help me ? whatsapp : +96176653802
i have all parts

axjjienn, 2020-03-18:
The link @aris-t gave and [this](https://i2.wp.com/christianto.tjahyadi.com/wp-content/uploads/2014/11/nano.jpg) diagram should get you where you need to be.