jcl5m1/ventilator #24: Collaboration

douglaswburke, 2020-03-18:
Hi Johnny.  My name is Doug Burke and I run a healthcare IT company called Cognitive Medical Systems (www.cognitivemedicine.com).  Two things:

1) We might be able to help with the arduino request and software needed to run the ventilator at different setting and for different patients and/or diseases.  My co-founder is a neonatologist and our CMIO is an anesthesiologist.  We work a lot with the DoD and VA and so some work in this area around automation and decision support in anesthesia and ventilation.

2) The DoD is about to release a "challenge" grant/competition where, from what I understand, the challenge will be to create a ventilator from common off the shelf parts for less than $200.  You should enter.  Send me an e-mail and I can let you know when the competition is live.

My e-mail address is:  dburke@cognitivemedicine.com

Thanks for doing this work!  Lots of interesting challenges at a critical time in healthcare delivery for the world.

Cheers - Doug

NoahGWood, 2020-03-19:
Hello Johnny and Doug.

I run a small quantum computing startup Spooky Manufacturing, we can assist in the circuit design and PCB prototyping of this device.

hobodrifterdavid, 2020-03-19:
Hello. Please also check Slack here: https://join.slack.com/t/open-ventilator/shared_invite/zt-cq0c2o4x-gXJLb0sZ4dVTUPt3oZy61Q

jcrubino, 2020-03-20:
Hi @douglaswburke and Johnny
I am a former LEVEL III NICU Registered Respiratory Therapist with ECMO transport experience and now professional software engineer working in python.  Let me know if there is a potential good fit for collaboration on this or future collaborations.

jcl5m1, 2020-03-24:
I could use help understand the existing options for ventilator tubing for managing inspirary/expiratory flow.  In particular how expiratory flow is prevented from going down the inspiratory circuit?, and how come inspiratory IPEP/EPAP doesn't create flow out the expiratory circuit?

DaveM135, 2020-03-24:
Hi Johnny

I believe these are valve controls that maintain flow directionality and line pressure (inspiratory pressure and end expiratory pressure).   The valves are generally either time or flow rate controlled depending on the mode of ventilation..  I’m referring to traditional mechanical ventilators or BiPAP machines in the case of noninvasive ventilation.   Perhaps there is a respiratory therapist on the thread who can provide the specifics or point us to the right technical references?

Best
Dave

nasocializes, 2020-03-25:
@jcrubino  Can you point to existing parts that are used in ventilator circuits that are readily available?

jcrubino, 2020-03-25:
For NIV there is an Exhaust valve (expired gas in the image) in the circuit.  Because of the back-pressure of the inspiratory line much of the exhaled air exits the exhale port.  Additional valves "peak plateau valves" are available to add for safety and better CO2 removal.

As mentioned in other issues, CO2 retention is a serious issue in NIV because the CO2 can often get pushed back into the circuit and re breathing of CO2 occurs.  

**Minute volumes in NIV are not guaranteed because of leaks around the mask and tube so monitoring of CO2 and satO2 are essential to gauge effective ventilation and oxygenation**

[image]

Because the lungs are more compliant than the exhaust valve in the circuit the positive pressure moves air into the lung first.  Vents often check or make available a vent circuit compliance test for aid in more accurate minute and tidal volumes.

jcrubino, 2020-03-27:
@nasocializes @jcl5m1 

[Mechanical Ventilator Milano (MVM):A Novel Mechanical Ventilator Designed for Mass Scale Production inResponse to the COVID-19 Pandemics](https://arxiv.org/pdf/2003.10405.pdf)