jcl5m1/ventilator #26: Support arduino quadcopter controller boards with barometric pressure sensor

zobotek, 2020-03-18:
Many flight control boards intended chiefly for quadcopters (such as the atmega-based APM 2.5 and 32-bit boards similar to the pixhawk) have absolute pressure barometers that, with some tubing, could be used to sense pressure to help sense and control the cyclical breathing process.  This hardware is already in the hands of many with the aptitude for DIY respirator construction.

jcl5m1, 2020-03-19:
do you have a pointer to the baro sensor and control board?  Direct measurement of pressure would be nice.

zobotek, 2020-03-19:
MS5607 and MS5611 are common barometric pressure sensors in the older Atmega-based flight controllers such as the Ardupilot 2.X hw installed in many quadcopters that now grace many a garage and many would gladly sacrifice for a noble cause.  The opensource Ardupilot SW is a fertile source of firmware that can read the sensors, control motors, log, interface with remote displays, and has PID controllers that might modulate compressor pressure.  

Links:

APM 2.x flt controller hardware:  https://ardupilot.org/copter/docs/common-apm25-and-26-overview.html  
MS5611/MS5607 pressure sensors:  https://www.te.com/usa-en/industries/sensor-solutions/applications/drone-sensors.html

zobotek, 2020-03-19:
Adding some thoughts on using quadcopter brushless motor controllers to run a fan/turbine:
These controllers are capable of sensing motor current (they need phase-voltage back EMF to time commutation) and RPM.  These typically run open source SW such as BLHeli.  I once submitted a pull request the Blheli code for a specific speed controller to provide pulse RPM output on an unused pin that's then fed back to a digital input on the Arduino-based flight controller, making closed-loop motor speed control possible.  I think this took a grand total of about 6 lines of code.  

Communicating motor current back to the flight controller, if needed, could take more considerable design and development. 

Links:  
https://www.embedded.com/designing-electronic-speed-controllers-for-drones/ 
https://github.com/bitdump/BLHeli

sorki, 2020-03-19:
BMP180 / BMP280 are also quite common. We now have bunch of opensource BLDC controller firmwares as well running on 32 bit MCU ESCs that could be adapted with pressure sensing and other required functionality. Not all quadcopter ESCs can output current sensing data - most new Blheli32 can (via UART telemetry) but Blheli32 isn't opensource anymore.

I can try hacking hacking pressure sensing via BMP180 which I have lying around to my BLDC firmware if needed - though isn't current sensing enough to detect breathing periods?

jcl5m1, 2020-03-20:
the BMP180 seems to only go up to 1100hP. which is only 11cmh2o,  That's not high enough. 
https://www.adafruit.com/product/1603

MS5611/MS5607 pressure sensor claims it can operate uto 1200mbar, which is >1200 cmh2o, which is plenty.   Seeing if it could measure lung pressure would be great.  Could you log some data and share plots? 
https://www.te.com/usa-en/product-CAT-BLPS0035.html

zobotek, 2020-03-20:
will do.

zobotek, 2020-03-21:
Working on this now.  The plan is to have a small diameter O2 hose cannibalized from a cannula run from a Chinese 'Pixhawk LITE' (STM32) flight controller controller board to a CPAP hose attached to either a 3M dual-cartridge respirator or a CPAP mask (I have both to test with).  Note that the breathing pressure is _highly_ dependent on the whatever flow resistance and capacity that the air supply provides.  This resistance may come from filters, valves, hoses, etc.  

The pneumatic design needs some thought here.  It's better from a durability/reliability perspective to sense pressure at the body of the ventilator so that a single (or double) air supply hose is all that runs to the patient -- no separate pressure measurement hose.  But measuring closer to the lungs is probably more accurate.  Since the commercial designs seem to NOT have a separate pressure sense hose, I'm going with the 'measure on the distal side of the CPAP hose - away from the mask'.   But the design depends on some slight resistance to free breathing to generate a pressure drop, even if it's the resistance of air motion in an open area.  More resistance means more pressure drop for a given airflow.  So I'm thinking of using a double  will rig it such that the air supply is open room air on the far side of a common CPAP hose, with the other end connected to the mask.  So pressure 

Step 1:  Design & 3D-print an adapter that allows a tiny hose to be attached to the MEAS MS5611 pressure sensor that's soldered on to the flight controller.  The adapter will be glued around the edges of the sensor in a manner that ensures the edges are sealed.  I will cannibalize  ~6' of small diameter O2 hose from an oxygen concentrator's cannula.  I believe that this is a standard cheap hose that hospitals and drug stores are likely to have some stock of.  

Step 2 is to use the stock logging capabilities of Ardupilot SW to log pressure while I breathe.

Step 3 would be to move away from precompiled Ardupilot firmware to pressure measurement routines supporting an Arduino STM32 tool chain that is compatible with this project's Arduino code.  Hopefully this project's code will run on at least ATmega and STM32 controllers. 

>

jcl5m1, 2020-03-22:
Yes, the look ms5611 like it has plenty of accuracy and range.  Correct, having a measurement to atmospheric would be best. But, I think this could be sample from an initial idle state.

 https://www.youtube.com/watch?v=T3ma1n_jhbQ

jcl5m1, 2020-03-25:
It seems that flow/volume measurement is a critical feature.  I found some flow sensors for ventilators, but they are 1) reasonably pricey 2) the supply of those now dried up.  I can't find a place to buy them.

Do you think you could make an accurate flow sensor using two MS5611s?  This would be quite cheap, and digikey has a large stock of them.

zobotek, 2020-03-25:
Yes, if the air pump characteristics and pressure are known, flow can be
computed from a single absolute pressure sensor.

IMO, a differential pressure sensor attached to a tube of known resistance
a simple setup for flow measurement.  I have at least 3 such sensors in my
possession.  2 are used for airspeed sensors, and one for a septic system
pump controller.

drjgset, 2020-06-28:
If you need to set-up multiple pressure sensors for differential diagnoses in a hospital ward, then is there a Team out there that can use a liquid/air interface, especially viscous fluids, to determine the correct code for this need?

This is a necessary loop code with a graded application.

drjgset, 2020-06-28:
Is there a submersible model and code for determining barometric pressure using available sensors or biosensors with similar applications?

zobotek, 2020-06-29:
Many are submersible.  For info on sterilizing, check the data sheets.  I
3D-printed a hood and hose adapter for my flight controller's built-in
sensor.  3D printed things are typically not airtight and might need to be
sealed, depending on the application.  I typically seal with paint or
superglue.  A tiny airbag connected to a sealed tube should be cleanable
with the bag submersible.  There's lots of code out there - Google is your
friend.