jcl5m1/ventilator #29: Contact request: Dr. Sidney Merritt UCSD - endotracheal tube

jamescnelson, 2020-03-19:
Don't use this site often, looked hard for any other means to contact other than issue requests but a Dr. Merritt at UCSD is looking to speak with the Creator of this device as it's being discussed in med communities across the net regarding the addition of an endotracheal tube

619-543-6240
Email: skmerritt@ucsd.edu

jamescnelson, 2020-03-19:
[image]

AgentSmithers, 2020-03-21:
I'm a software engineer and have experience coding micro controllers to handle this logic. I would like to add my efforts to support this idea. Please let me know if there is anything I can do to help!

jamescnelson, 2020-03-24:
Contact Sidney Merritt at UCSD her email is in the support request. you may
want to work with her directly.