jcl5m1/ventilator #30: Device does not blow off CO2

JoshuaPK, 2020-03-19:
Has anyone done a study on the failure mode of Covid-19 patients with acute respiratory distress?  Is the problem that the SPO2 (oxygen) is too low, or is the problem that the pCO2 (carbon dioxide) is too high?  (It's possible to have an acceptable SPO2 while at the same time having a harmful amount of CO2 in the blood- my first wife passed due to a complication from pneumonia with that exact issue.)  Could this device be engineered to blow off the CO2 as well as increasing SPO2?  Can that even be done without tracheoscopy or intubation?

jcrubino, 2020-03-20:
Increasing the respiratory rate with adequate volumes will blow off the CO2.
Yes, almost any vent system can blow off CO2, but it is not always helpful to correct the underlying cause of the CO2 buildup.

sPO2, needs proper supplemental Oxygen and PEEP to allow the lungs to absorb the O2.  Additionally there are secondary complications that can change how this is treated, i.e. functional hemoglobin

You will need arterial blood gas work done to understand the ventilation (CO2, O2) physiology of the patient.  Blowing off CO2 mechanically is not a long term solution to care for a patient with a metabolic cause of the CO2 retention.   Certain cases could be mediated in a few hours with sodium bicarb iv.

So far the formal data from China could be interpreted that the secondary issues around COVID 19 critical care are primary concerns with COVID19 making them more difficult to treat effectively.

jcrubino, 2020-03-21:
Hi @JoshuaPK I just found this webinar on youtube given live yesterday
https://www.youtube.com/watch?v=rMT97toZNJw

Bullochman, 2020-03-22:
Co2 Retention is a major issue with Non-Invasive Mechanical Ventilation.  If the depth of the breath isn't applied properly, the lower half of the lungs will retain Co2 instead of expelling it.  Often times, if a patient is retaining Co2, their Oxygen levels (SpO2/Oximetry) will appear low and to compensate, the doctors and nurses unfamiliar with Co2 retention and the dangers will increase/aka attempt to titrate the Oxygen Liter flow to the patient to bring their oxygen levels (aka sats) up, but this only worsens the condition and can cause irreversible damage and death.

I'm the director of a Home Medical Equipment company and have respiratory therapists on staff and my father is a Respiratory Therapist by trade.

The only true solution to this problem is to _monitor Co2 levels_ in addition to the oxygen levels.  This requires an End Tital Co2 Monitoring machine (ETCO2) which means it measure the level of Co2 exhaled.  Capnography is the method of measuring the Co2 levels and the least expensive machine I've ever seen is about $500.

However the new Trilogy Evo Ventilators (released this past January) is the very first Ventilator ever on the market with a build-in Co2 monitor and can be used to monitor Co2 levels for such patients/conditions.

I have a few extra Co2 monitors I can send to anyone interested in further developing the project.  If you can send me an e-mail or text Two Oh Five 447 Zero Eight Two Nine with an address, I'll get it out to you quickly to help speed things along.

Hope this helps, just let me know if there's anything I can do in my capacity.  Thanks

jcrubino, 2020-03-23:
It does help, end tidal co2 would be great.  For myself I am trying to sort
out how much scope a project like this can take on to remain helpful and at
various cost points.

There are basic CO2 sensors available to hackers... Might want to save your
for patients in need.

jcrubino, 2020-03-23:
@Bullochman you might want to stick around as a device expert.
The scope of the project started as a challenge to create a ventilator from widely available parts... but as you pointed out there is more to do... like etCO2, sO2, mask, headgear, humidity etc.

jcrubino, 2020-03-23:
@Bullochman checkout my github [page on covid](https://github.com/jcrubino/Respiratory-Care-of-COVID-19) material for health professionals.  Id like to know if there is any material sections, specific information, that will be helpful for Healthcare workers to get up to speed on COVID-19.