jcl5m1/ventilator #32: Build a PAPR!

jcl5m1, 2020-03-20:
As I've learned more about the risk of high pressure care, it's definitely not for the faint of heart and can continue as an engineering thread.

I wanted share a much lower risk, less controversial application of the same parts - which is a Powered Air Purifying Respirator (PAPR).

https://github.com/jcl5m1/ventilator/wiki/Build-a-Low-Cost-PAPR

cbonitz, 2020-03-20:
Just a quick thought: Changing the configuration to push air through the filter, and having only the hose and the mask after the filter would turn any leakage from a safety issue into an efficiency problem. In the design you propose, leakage in the fitting between the filter and the pump or in the pump itself might cause the system to pull in contaminated air.

```
<-----------------+------------------------>
Negative Pressure |   Positive Pressure

    +------+   +----+                 +----+
    |Filter+---+Pump+-----------------+Mask|
    +------+   +----+                 +----+

               +----+   +------+ Hose +----+
               |Pump+---+Filter+------+Mask|
               +----+   +------+      +----+

```

jcl5m1, 2020-03-20:
ah, good point.

jcrubino, 2020-03-21:
HEPA h11+ filters might be a responsible alternative for this project to N95 filters needed in the healthcare center response.

Vacuum cleaners have been using them for years and making them widely available.

The foreseeable problem is deformation under pressure loads.

nasocializes, 2020-03-25:
@jcrubino  These are great ideas if you can prototype & test them.

jcrubino, 2020-03-25:
@nasocializes parts are on their way.

chakahamilton, 2020-03-25:
Here are the parts ive ordered from AMAZON for my PAPR Design. unfortunately, Their decision to only ship essential stuff means I have to wait until mid April to get everything. @jcrubino I too looked into using a UV light Source, but in my research found that it would have to be so intense it would require a lot of power. I could be wrong though in my interpretation. I plan to use HEPA Filters but am also trying to figure out how to "Sterilize" the incoming air which is almost impossible using a small footprint. I was also planning to power this system with Standard Drill batteries but will most likely settle on SLA Batteries 12V 10am for my motor size.

[image]

jcrubino, 2020-03-25:
@chakahamilton I understand the problem with UV light.  It needs to be a non ozone emitting UV light (which is usually the case if I understand, but 20 seconds of exposure is also needed.  I did not do any numbers on the strength of the light but I understand that has to be part of the equation.

The power source is another issue.  Light enough to be on a belt and rechargeable is ideal.  I like the cordless drill battery idea a lot.  We will need 3d printable designs for this.  

The cpap/vent filters are a great idea that I am using as well... but those might be hard to come by someday soon.

I am trying to come up with an inexpensive hood system for clinicians that is not scary for children.  T
T
he full face snorkel mask is indeed practical but for anyone wearing glasses and meeting with patients outside the wards it might not be conducive to clinic orientated tasks with people who may or may not be infected.  

Anyone know what the paper like material is... more solid then a walmart reuseable bag.  You can see the fibers, cheap and air tight?  Then how to fix a plastic material view portal to it with something better than hot glue?

Great ideas... lets keep the momentum up on this.

chakahamilton, 2020-03-26:
> @chakahamilton I understand the problem with UV light. It needs to be a non ozone emitting UV light (which is usually the case if I understand, but 20 seconds of exposure is also needed. I did not [...]

[These](https://www.amazon.com/dp/B07K6BK639) claim to be 395nm-405nm which in theory should work, but because UV light is not my wheel house I am unsure if that is the rating for the whole spool or each individual led.

I made these parts choices to  insure that it would be accessible for everyone to build. The idea of 3D printed parts is nice and all, but not everyone has access to those (I have 2). I also didn't want some one to need a full machines shop as well. So far I've used the following tools.

Drill
Step Drill Bit
Gorilla Tape
Hot Glue Gun
Xacto Knife

jcrubino, 2020-03-26:
I got an idea for respirator hoods.

I saw this today [DuPont expediting production of Tyvek hazmat suits for health care workers](https://www.nbcnews.com/health/health-care/dupont-expediting-production-tyvek-hazmat-suits-health-care-workers-n1169031) which is what I was looking for I believe.  

It can also be found (this and similar materials) in home stores as vapor barrier for new buildings. 

Any origami or seamstress/tailors out there for pattern making?

chakahamilton, 2020-03-27:
i'm using Rain jacket and pant for the suit, as tyvek and other professional jumpsuits are hard to come by or prices are extremely high and in addition to shipping/availability time.

chakahamilton, 2020-04-09:
I just finished building my first prototype of a PAPR based on my post [Above](https://github.com/jcl5m1/ventilator/issues/32#issuecomment-604093852)

[Video 1 Blower](https://photos.app.goo.gl/152azF1NSzZyT4S46)

[Video 2 The Face Mask](https://photos.app.goo.gl/EKKPcUzDHmcFNrf5A)

IT still need a little work but result so far are promising. I can breathe with the unit no problem. 
I plan to add Oxygen & Carbon Dioxide Sensors in the mask at some point.

oiaohm, 2020-04-10:
> @chakahamilton I understand the problem with UV light. It needs to be a non ozone emitting UV light (which is usually the case if I understand, but 20 seconds of exposure is also needed.

The bad news here all UV light that microbes and viruses makes ozone.  Dealing with produced ozone is a not simple.

https://en.wikipedia.org/wiki/Ultraviolet_germicidal_irradiation

Yes different frequencies of UV produce less ozone so make it simpler to remain under safe limits it also effects how much of it the air has to be exposed to.

"2011 Handbook, HVAC Applications, covers "Ultraviolet air and surface treatment" in Chapter 60"

This is noted in the wikipedia but you really do need get your hands on this and understand it before messing around with UV treatment of air.   Time of exposure alters based on flow rate and UV intensity and frequency.   20 seconds is only a really rough ball park.  Yes it comes a balancing act between ozone create and treatment speed.

jcrubino, 2020-04-10:
Modern germicidal lights should/are filtering out the light wave band that
creates ozone...

You need aprox 50 to 200 watts per square meter per second to disinfect
using a flourecent based UV light.

oiaohm, 2020-04-11:
> Modern germicidal lights should/are filtering out the light wave band that creates ozone... You need aprox 50 to 200 watts per square meter per second to disinfect using a flourecent based UV light.

This is horrible wrong on what is filtered out this will lead you into very big danger.

https://www.oxidationtech.com/ozone/ozone-production/uv-lamp.html
 The only section of the UV spectruim that does not produce ozone at all is UV-A band.     
UV-A does not work as germicidal .    Germicidal lights are UV-C that is 200 to 280 nm .   Yes link above 100nm-240nm ozone is made one way and 240-315nm ozone is made another there is not a single bit of the UV-C spectrum that does not produce ozone.  Yes 240nm makes ozone both ways.  So every bit of the germicidal lights frequency will produce ozone there is no way around this as it the nature of ozone production.  

Lets get to what they are in fact filtering out.

Now there are a few frequencies you avoid because UV way more effective at making ozone at those points.  Like you are going to want to filter out 254 nm out of UV-C Germicidal light because you get massive peak in ozone production at that point with that freq at 50-200 watts you are not going to stay inside safe values for ozone.

Basically the UV filtering on germicidal lights lowers the ozone produced does totally not stop ozone being produced.   This is the dangerous part you take a light design for a square meter of air and use it to process 1l of air instead you might end up with ozone above safe percentages.

Do also note Germicidal lights don't come with the one set of standard filters on them either the amount of watts a surface needs to be treated is linked to allowed frequencies of the germicidal lights also how many watts a area being treated can be treated with before you exceed safe ozone levels.  Welcome to nice fun.

Basically Germicidal Lights are very tricky items to play with and very simple things to totally screw up with.   Yes in a breathing system you would want a ozone alarm sensor.

There is also the confined space problem where the ozone in a room where you are using germicidal light can slowly increase over time.   Basically germicidal lights are not 100 percent safe and require serous due care.

richovercash, 2020-04-11:
Chaka,
Progress looks great!
I had similar idea using iRobot filter, Ryobi Battery, and motor/blower.
This is just a basic proportional model without any design.

Thanks + Best,
Rich Overcash
Industrial Design

This e-mail, including attachments, may include confidential and/or
proprietary information, and may be used only by the person or entity to
which it is addressed. If the reader of this e-mail is not the intended
recipient or his or her authorized agent, the reader is hereby notified
that any dissemination, distribution or copying of this e-mail is
prohibited. If you have received this e-mail in error, please notify the
sender by replying to this message and delete this e-mail immediately.

richovercash, 2020-04-11:
[image]

Chaka,
Progress looks great!
I had similar idea using iRobot filter, Ryobi Battery, and motor/blower.
This is just a basic proportional model without any design. Can place waisteband clip on backside. 

Looking for feedback. Attaching images here.

jcrubino, 2020-04-11:
@oiaohm 

I respect your critical take on UV light in general.  In addition to potential ozone production, there is the problem that most materials are not tested for UV-C band light.  

My comment was made in haste, perhaps too *lightly*.... a mention of some prior reading I have done on this.

Yes Ozone production is produced in much of the UV Band, yet using the reference you posted above... the facts you are sharing seem misinterpreted or I am not catching on to your rhetoric.

> Like you are going to want to filter out 254 nm out of UV-C Germicidal light ... 
[image]

The link you posted says 254 nm destroys ozone, and I have read this in other places as well.  I believe the grey area here is that most lights available cannot produce 100% within the specified narrow range, rather the stated wavelength on a UV light is the peak measurement.

oiaohm, 2020-04-11:
> @oiaohm

> The link you posted says 254 nm destroys ozone, and I have read this in other places as well. I believe the grey area here is that most lights available cannot produce 100% within the specified [...]
This diagram is not really fully covering the problem.  Peak of ozone destruction is still creating free Oxygen.   Yes that free Oxygen can link up and make O2 but it still can link up and remake O3 from the 254 nm side as well.  

Yes the grey area is a problem but even a correct UV freq light for ozone destruction will not produce air 100 percent ozone free all the time because of the destruction followed by recreating cycle that is not in basic diagram.

Yes grey area of UV light ratings mean you have some of the low frequency that will 100 percent produce ozone.

Please remember that free Oxygen does not have to just bind with O2 to make O3 or another free Oxygen to O2 like it could react with seals and other things.

> My initial concern, was that this approach is not practical considering that, circulated air has to be exposed to the uv light for a second at such wattage levels. For surfaces cleaning, this seems possible but... not for portable maker made ventilation device.

https://breathequality.com/germguardian-ac4825e-review/

I would not say it impossible devices like this are getting away with a 5watt light but the lot of considerations to get the chamber size to light to airflow rate right the wattage may not be that high.   Please note a device like these air purifiers  is not putting the air straight into a pipe/past seals that may not like free Oxygen.  They are also not aiming for 100 percent ozone free just not more ozone that what went in.

My brain was remembering it was not a free lunch.  I was thinking Ozone but the biggest problem here is free Oxygen reacting with materials its does not matter where in the Ozone altering section of UV you have this problem.  

I was remembering correct that Ozone does not magical cease to be something you can fail to monitor just because you use UV-C.   The free Oxygen problem was what I was getting mixed up on.   Notice both UV levels are going to make this problem remember free Oxygen really loves reacting with things and if it the  wrong things can the result can be start producing lots of heat leading to fire inside device.   Yes fire inside a breathing device would be really bad.

I am not saying don't play with UV but be really careful this can backfire badly.  UV-A is the only area you are not playing free Oxygen problem and that does not work for germicidal.

Sorry my brain was being stupid when I was reading those diagrams some how I was reading seeing O1 as O3.  Yes high volume O1 going into humans lungs does not do those the world of good either.   Basically you don't want O1 or O3 in large volumes going into human lungs.

chakahamilton, 2020-04-11:
> [image]
> 
> Chaka,
> Progress looks great!
> I had similar idea using iRobot HEPA filter, Ryobi 18V Battery, and motor/blower.
> This is just a basic proportional model without any design. Can place waisteband clip on backside.
> 
> Looking for feedback. Attaching images here.

@richovercash I really like that design. i hacked a filter holder together with Mock-up with a Radio shack project box hot-glue and gorilla tape. can you make it so the hole at the top accepts a standar CPAP hose (thats what I am using)

jarvik1984, 2020-04-15:
> I just finished building my first prototype of a PAPR based on my post [Above](https://github.com/jcl5m1/ventilator/issues/32#issuecomment-604093852)
>
> [Video 1 [...]

Wonderful job... I’m a physician and new to github but something similar to your prototype going. I’m trying to find a way to get portable power to the blower. My blower is DC 12V/8A m. I’ve also tried with a brushless radial blower that provides 42 CFM with a shorter tubing and a HEPA in-line filter but was not as good as it was with the electric air blower.

jarvik1984, 2020-04-15:
Chaka,
This is what I’ve tried 
[image]

chakahamilton, 2020-04-15:
> Chaka,
> This is what I’ve tried
> [image]

To make mine portable I used a car cigarette lighter outlet 12 volt. I'm also planning to power it with two sealed lead acid batteries 12 volt 10 amp so my blower will last approximately 10 hours since it only pulls in about an .50 to 1amp. 
I hope this help.

chakahamilton, 2020-04-15:
I've made a slight design change, because it will require less points of failure and moving parts. I found this sand blasting hood on Amazon. I think it will work better than the snorkel mask, but snorkel mask are readily available.
[image]
.

chakahamilton, 2020-04-15:
@jarvik1984 I found this blower on Amazon interesting I like the fact that it is 188 CFM. it would definitely solve the low air pressure issue. I have yet to purchase this though. I am still building the power supply.
TerraBloom 6" Silent Inline Duct Fan, 26W, 188 CFM, Quiet Mixed-Flow Energy Efficient Blower for Air Circulation in Ducting, Vents, Grow Tents

[image]

jarvik1984, 2020-04-15:
> @jarvik1984 I found this blower on Amazon interesting I like the fact that it is 188 CFM. it would definitely solve the low air pressure issue. I have yet to purchase this though. I am still [...]

I’ve been testing it at the hospital with the electric air pump and I can tell you that more than 48 CFM is a lot of flow, you can barely breath with so much positive pressure on your face... as far a the battery, I’ve thought about that too but I found those batteries at Homedepot and each are 3 kg/5lbs more... happy to hear from you again and thanks for your reply

chakahamilton, 2020-04-15:
> I’ve been testing it at the hospital with the electric air pump and I can tell you that more than 48 CFM is a lot of flow, you can barely breath with so much positive pressure on your face... as far [...]

Thanks for the Airflow 48 CFM I was having an issue where the blower I currently use wasn't able to push past the HEPA Filter, so I was trying to overcome that. I think Ill add a speed control to my setup to see if I can control the flow more efficiently that way.

As for the batteries, yes that adds quite a bit of weight. but I was thinking instead of a traditional belt worn battery pack, a knapsack might be better suited for the challenge. A someone wearing it out in the field, is pain strain an issue?

jarvik1984, 2020-04-15:
2 issues added with bigger blowers in my opinion. 1) Too noisy... we have to be able to communicate, I think anything past 60db is a lot. 2) for the batteries, we usually have to be running around the hospital and a very heavy battery could make it hard to get things done. How about lithium battery packs?

chakahamilton, 2020-04-15:
I've been looking at lithium ion packs, to save on weight obviously but my expertise is very limited when it comes to those they actually kind of scare me. I'll perhaps revisit looking at them and doing some more research. I also found that they are also extremely expensive when you look into the multi-amp hour aspect of them The higher the amp hour the more expensive they seem to go. 

Yes as you saw my video my blower is very noisy, so I'm looking for something that's very quiet as well I'll continue my search and report back.

chakahamilton, 2020-04-22:
> Chaka, how are you doing? Any new improvements on your prototype?

Jarvik1984, Yes! I have to new blower designs that I am going to test out once they arrive. Both will offer an nearly silent or quieter operation. When they arrive I will build and test both.  One involves a series of Mid range computer fans & another a continuous Blower. I am also going to tweak the filter idea to use polypropylene filter fabric. 

Stay Tuned.

jarvik1984, 2020-04-22:
Awesome. Please keep me posted, I’m working on it too.. I’d love to share ideas and join efforts to get this done.

chakahamilton, 2020-04-22:
> Awesome. Please keep me posted, I’m working on it too.. I’d love to share ideas and join efforts to get this done.

Will do.

chakahamilton, 2020-04-24:
I've Started working on A 2nd version of the Blower using 2x 12v 2.20A CPU Fans  for air intake and a smaller 12v 0.16A general computer fan to blow air into the hood attachment. I am keeping the CPAP hose and Filter Attachment through all designs. 

I am still waiting for the polypropylene  Fabric to arrive so it can be tested as a filter material. I have 

**Pictures of Version 2:**
[Picture 1](https://photos.app.goo.gl/27BrRJEyY2dSoxCJ6)
[Picture 2](https://photos.app.goo.gl/wYwPpkrxgrpGtZGy9)
[Pciture 3](https://photos.app.goo.gl/keqvEpdS2iaghPyF6)

After all of this I will tackle the electronics 
I'll definitely follow this Thread #62 more closely.

chakahamilton, 2020-04-26:
I'm moving all of my development to my [Git.](https://github.com/chakahamilton/Powered-Air-Purifying-Respirators-PAPR-) you can find further updates there over the next few days..
Regards.

StevenACoffman, 2020-08-14:
You may want to check out this open source PAPR Hood design: https://making.engr.wisc.edu/papr/