jcl5m1/ventilator #33: PID Controller

HDElectronics, 2020-03-20:
Hi!!
I love your initiative, a humanitarian gesture <3
I would like to add something to your project, it's a PID controller
this controller which will be implemented in the arduino program will simply keep a constant pressure while playing on the variation of the speed you motor.
You can read some documentation to do it or i'll help you to do it.

necro-nemesis, 2020-03-20:
sure sign me up.

peteflorence, 2020-03-20:
Hi there, yes some type of closed-loop control will be useful.
I recommend reading Johnny's notes from a call with a pulmonologist: https://github.com/jcl5m1/ventilator/wiki/Notes-from-chatting-with-a-pulmonologist
To have more context on what control is possible / needed.
At present Johnny's prototype doesn't have a direct pressure sensor, so it isn't currently possible to close a PID loop on a pressure setpoint.
But such a sensor could potentially be integrated.

If possible, someone could build a copy of Johnny's device and prototype pressure sensor options.