jcl5m1/ventilator #36: Construction with homemade elements.

enrutador, 2020-03-20:
Hi from Spain
Many people do not have and may not be able to purchase any CPAP units, but they do have mat inflation pumps. Is it possible to use it to breathe and expel air into the lungs?

Type: 
https://www.jardinitis.com/bomba-electrica-quick-fill-220-240v-invertible-bec83/?otcountry=ES&gclid=EAIaIQobChMIhKeZ0vqo6AIVSFXTCh1KFQPbEAYYAyABEgLKUfD_BwE

Thanks all the people

HDElectronics, 2020-03-20:
Yeah you're right, some people cant even buy from the internet because of the delivery is stopped right now, however we can fin more solutions by using the hair dryer for example to transform it into a Ventilator-ish Device.

zobotek, 2020-03-20:
These lack the control and monitoring capabilities of brushless systems such as CPAP air pumps, computer fans and quadcopter motors coupled with BLDC controllers.