jcl5m1/ventilator #37: Hairdryer + Arduino with a relay

kyrgyz-nlp, 2020-03-20:
I attached a tube to a hairdryer (with heating off) –– it seemed to work, meaning it blows really well. Planning to buy a relay for automation with Arduino.
Do you think this kind of build would be more accessible to a broader audience?
Also, I'll make the air go through a container with warm water to humidify it.

I'll be happy to hear back!

jcl5m1, 2020-03-20:
That's a really nice thing to try.  I would recommend find a way to measure either the pressure testing using the water tube fixture, or a flow rate measurement.  I suppose you could use a garbage bag to estimate Liters/min of air flow.   Time how long it takes to fill the bag up?

Would recommend first try making a PAPR with it. 
https://github.com/jcl5m1/ventilator/wiki/Build-a-Low-Cost-PAPR

You can probably use a standard N95 face mask as a filter on the intake.  If you do this, take photo and describe your test results.

PatoZG, 2020-03-25:
That’s a great idea, hairdryers offer good power output. Hairdryers usually employ AC motors, in order to modulate the fan speed a variable frequency drive is required. I recommend buying a solid state relay to modulate the speed using Pulse Width Modulation, which can be output from an Arduino DO pin. Here is a sample relay Register a new account on the Banggood APP and get $100 of coupons. I just found an amazing item with free shipping.https://banggood.app.link/iGvLLhSk84

chakahamilton, 2020-03-25:
i recently explored the hairdryer idea. in order to get it to work on batteries, I just needed to break apart the circuit board on top of the motor, a simple full bridge rectifier. Once that was done im now running the dryer on my drill battery.

chakahamilton, 2020-03-29:
I've tried the hair dryer will HEPA  filter and its not strong enough to allow enough airflow to breathe.

iaankrynauw, 2020-04-22:
@chakahamilton what was the wattage on your hairdryer? 

If it is in the upper spectrum of normal hairdryers, is this then unfeasible? 

Is the N95 mask still an option to use as a filter, will it let more air through?

chakahamilton, 2020-04-22:
> @chakahamilton what was the wattage on your hairdryer?
> 
> If it is in the upper spectrum of normal hairdryers, is this then unfeasible?
> 
> Is the N95 mask still an option to use as a filter, will it let more air through?

I use a 1875 watt hair dryer with a vacuum cleaner filter not (synthetic fiber)
I'm currently testing two new designs you should follow the build PAPR thread #32 for progress updates.