jcl5m1/ventilator #38: Alarms: low airflow

amanzimdwini, 2020-03-20:
Not sure how easy - but you'd need an alarm when the airflow decreases (such as when the mask comes off the patient). Please add to your "to do list".
And yes, (1) last effort, and (2) probably best for PROVIDERS, not patients. GREAT idea, though - I am ripping an old CPAP apart right now

Istria1704, 2020-03-22:
The most robust and simple way I could think of:
Span a thin copper wire across the air channel and put a small (but constant) current through it.
The wire will heat up until it reaches equilibrium temp.

When air flows, the wire cools down. When air stops flowing, the wire heats back up.

Since resistance (Ohm) of the wire goes up with temperature rise, you could measure this "cycle" of increasing and decreasing resistance in the wire. If the cycle stops, you know something is wrong.

jcl5m1, 2020-03-22:
added to do's.

i ordered some nichrome wire, which gives 26 ohms/ft.  But, i actually
don't think
https://www.amazon.com/gp/product/B07CHTXZYW/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

This reference has consideration of the material for hot wire anemometer.
But, seems have mixed material recommendations.
https://www.sciencedirect.com/topics/engineering/hot-wire-anemometer

I think this can be done with a low cost 24-bit ADC like the HX711
https://www.sparkfun.com/products/13879

I am also curious about ultrasonic flow sensors:
https://www.youtube.com/watch?v=Mkjw-iX6Af4

 I wonder if one of these can be re-purposed to do that.  It claim to have
0.3cm resolution.
rough velocity estimation in the tube is 50cm/s is 1.5% speed of sound.  If
I am doing the math correctly, does that mean to it has to be a 30cm long
tube that varies by 1% due creating a 1% deviation of 0.3cm that it can
detect it?
https://www.amazon.com/gp/product/B01COSN7O6/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

jcl5m1, 2020-03-22:
I also curious about these air flow sensors used by engines:
https://www.autozone.com/engine-management/mass-air-flow-sensor

jcl5m1, 2020-03-22:
Looks like they even have 8 wire MAF sensors, that include flow, humidty, temp, and baro

https://www.youtube.com/watch?v=mFhovSdOY3g

I picked up a Bluestreak MF21029N 5-wire MAF sensor.  Still trying to figure out how to talk to it.

jcl5m1, 2020-03-23:
Got a link to an actual ventilator flow sensor:
https://www.sensirion.com/en/flow-sensors/mass-flow-meters-for-high-precise-measurement-of-gases/low-pressure-drop-mass-flow-meter/

jcl5m1, 2020-03-23:
https://www.sensirion.com/en/flow-sensors/sfm3xxx-sensor-platform/

jcl5m1, 2020-03-25:
worried about supply on these.  I can't buy anymore. I am curious if a differential pressure flow valve can be make with two MS5611s

jcl5m1, 2020-03-26:
I tried an automotive flow sensor.  It provides an analog flow signal.  

https://photos.app.goo.gl/UkC9ceGeRCVUVESo6

 For the part I used: MF21029.  This pinout is starting from flat side:
1. flow analog out (0-5v)
2. 5v reference from controller (unclear this is needed)
3. GND
4. 12v supply
5. temp analog out put (though this wasn't working for me)

jcrubino, 2020-03-27:
@jcl5m1 You want to make a "Manometer" for differential pressure measurements.
From this, along with the dimensions of the circuit you can use Poiseuille equations to determine the flow and volume.

MeatLasers, 2020-04-02:
The SFM3200 supply might indeed by tight for a while.
An alternative would Sensirion SDP800 sensor, which is used to measure the airflow in CPAPs. This should be available in very high quantities, and is also now available without any problems at Digikey.
More info:
https://www.sensirion.com/en/flow-sensors/differential-pressure-sensors/sdp800-proven-and-improved/
https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/0_Corporate/Specialist_Articles/Sensirion_Specialist_Article_Gas_Flow_Bypass_EN.pdf

MikeJaworski, 2020-04-03:
I'm also interested in the issue of flow metering here - I'd already decided hot wire is the way to go as that's what commercial machines use but looks like the SFM3XXX series is out of stock globally and wouln't want to tax the medical supply chain anyway.

Can I ask how testing with automotive MAFs is going?  Which model?  Are they sensitive enough considering they are intended for use at much higher flow rates?

PS:  I also found ultra-cheap (£5.50 = $7 each) **bidirectional** MAFS:

https://www.te.com/commerce/DocumentDelivery/DDEController?Action=srchrtrv&DocNm=LMM-H03&DocType=Data+Sheet&DocLang=English&DocFormat=pdf&PartCntxt=G-MAFCO-005

This way you could easily measure inspired and expired flow, and calculate leak rate by difference.

Only issue is the linear portion of the sensor response is over the velocity range 20-40m/s

For standard 22mm ID breathing tubes, the velocity at 100LPM is only 4.3m/s so you'd be on a highly non-linear part of the response curve...

MikeJaworski, 2020-04-03:
OK, even better I found this DP flow evaluation kit from sensiron with flow element.

The bare DP sensors are about $12, evaluation kit with flow element is about $70.  You could 3D print flow elements, probably.  Evaluation kit flow element fits to medical tubing.

https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/8_Differential_Pressure/Application_Notes/Sensirion_Differential_Pressure_Sensors_EK-P4_Flow_Element.pdf