jcl5m1/ventilator #39: Over pressurization

fireb0lt, 2020-03-20:
I don't think that 20cm h2o is very high, nor would cause any lung trauma. My personal NIV mask setup tops out at 25cm and isn't even fully inflating my lungs. Cough assist machines go into the 60s-70s of pressure, even if it is for brief periods.

All this to say, we likely will have problems getting high enough pressures, don't need to really worry about over pressurization. As a FYI, I'm from Washington State where Covid-19 is out of control. My pulmonologist is one of the top in the nation (Dr. Joshua Benditt) and we talked about too high of pressures on my ventilator. He wasn't concerned.

jcl5m1, 2020-03-20:
interesting, i get mixed info on this.  I agree 20cmh2o doesn't feel high, but 45cmh2o kind of hurts as a healthy individual. I don't know how sensitive COVID patients can get.    I'd be interested in Joshua's read on the following:

https://github.com/jcl5m1/ventilator/wiki/Notes-from-chatting-with-a-pulmonologist

nasocializes, 2020-03-25:
Johnny: it seems like going above 20cm requires additional sensors to ensure safety. we'll continue to use 20cm as a cut off guidance until we have more capabilities.

Thank you! Closing for now to clean up the list.

closed 2020-03-25 by nasocializes

bradtem, 2020-03-25:
Consulting with some doctors who do ventilation, pressures are high in ARDS.   The alveoli are closed up and flooding.  Lungs are not compliant.   As such, ARDS ventilation is done with a very high base pressure (known as PEEP.)  Doctors are setting this to 16cm in may cases.  This is the lowest pressure, for exhalation.  The peak pressure varies, enough to deliver the required volume of air (or rather oxygen.)  Because the alveoli are so stressed, it is desired to deliver a very low volume of air, which means having a very high concentration of oxygen.   I have heard reports patients start at 95%, which can't be sustained because it is toxic but they get them going this way.   If the patient starts at 16cm, I suspect a peak pressure of 20cm will not help many patients.  Of course there are patients not needing the 16cm, or the higher peaks.    Also, you end up pumping nearly pure oxygen, which clearly is a risky thing to do near heat sources.  Just about anything can burn in that.