jcl5m1/ventilator #40: Test Lung Research

NAlexC94, 2020-03-20:
In regards to Test lung research, what needed to be done.  I'm a biomedical equipment tester at a hospital with access to many different types of TMDE (Testing Measuring and Diagnostic equipment) and we frequently test ventilators and Cpaps.  Are there specific tests that need to be ran?

I'm going to try building this in our facility and once it's made I can test it using the same test setups we use for normal ventilators, even "dumb" ventilators.  Once made i'll post what models and results i got.

nasocializes, 2020-03-21:
That's awesome. What is the normal set of tests for a ventilators? We can program it to do that.

jcl5m1, 2020-03-22:
I did order one of these, to do basic testing.  But, it don't think it will be me precise intermediate volume measurement.  Just a way to test pressure.

https://www.amazon.com/gp/product/B07F331GDS/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1

nasocializes, 2020-03-25:
If folks have leads of access to test equipment in the bay area that would be great.

nasocializes, 2020-03-27:
@DaveM135 know of any in the area?

DaveM135, 2020-03-27:
Sorry, no.   

Best

Dave

MikeJaworski, 2020-04-05:
> @NAlexC94 That's awesome. What is the normal set of tests for ventilators? We can program it to do that.

Have a look at the ISO10651 series standards, they define a pile of test procedures.

You can download them for free from the British Standards Institute website providing you fill out a form declaring that they are for use in COVID response only:

https://www.bsigroup.com/en-GB/topics/novel-coronavirus-covid-19/ventilators/