jcl5m1/ventilator #41: Global Map for Startup Innovations - Get Funding for this Awesome Project

willivision, 2020-03-20:
Invitation to submit this project to STARTUPBLINK’S CORONAVIRUS (COVID-19) INNOVATION MAP
Get money for this project and colloborate globally
https://www.startupblink.com/blog/coronavirus-innovation-map

With the emergence and clear effect of the Coronavirus (COVID-19), people around the world are dramatically changing all aspects of life. This is an unfortunate circumstance that no one can escape, but as human beings are resilient creatures, we survive, and most importantly, we innovate.

No matter who you are, one of the things we need more than ever is to stay well-informed and to be aware of not only the bad but the good that is happening in the world. It is also important to know that public institutions, corporations, and startups around the globe have been working tirelessly since the emergence of the coronavirus in hopes to find a solution or contain this virus.

We at StartupBlink recognize this change, and we decided to do something about it. We are proud to announce the launch of our contribution to help the global crisis, the Coronavirus Innovation Map.

What exactly is the Coronavirus Innovation Map?
Our goal with the Coronavirus Innovation Map is to build a platform with a map of hundreds of innovations and solutions that help people cope and adapt to life amid the coronavirus, and to connect innovators.

We want you to be to use this platform in quick and easy steps, so, all you have to do is type in your location (City and/or Country), and choose the appropriate category wherein you would like to find a solution. 

Each category on the innovation mind map seeks to answer the following pressing questions:

Prevention – How to, track, and prevent the spread of the virus? For example, the “Track Virus” App in Israel helps to see the locations where the virus is widely spread. Allowing people to have this knowledge will prevent them to go to infected places.

Diagnosis – What kind of diagnostic kits are used to identify those with the virus? For instance, E25Bio has developed a paper-based test kit to detect Covid-19 that will be able to deliver results in half an hour. 

Treatment  – What kind of precautionary treatments e.g. medicines/drugs can help while specific treatments and vaccinations are still being developed. Aside from this, this also covers how people can get their medications without leaving their homes. In this sense, telehealth and home medicine delivery solutions will help us. 

Information – How the mass population can get timely and accurate information about the crisis situation and ongoing innovations? Lack of awareness stands in the core of panic among people; thus, crisis communication and provision of data via modern and widely-used ICTs is the potential solution to reduce the chaos. 

Life and Business Adaptation – How to adjust our daily lives to the current situation by going virtual? During a short period of time, educational institutions switched to e-learning, home offices temporarily replaced on-site work offices, and most of the official and entertainment events, as well as travel plans, were canceled. Now, it is time for you – people with innovative minds to come up with ideas on how societies can go online without changing their daily routines and simultaneously keep the quality of their lives.

The categories shown above will help those visiting the platform, quickly find a solution to their problem. As this issue affects all kinds of people and many aspects of our lives, the categories are broad for a good reason. This platform leaves no-one behind, you can be a government official, venture capitalist, entrepreneur, even parents and students. The main factor here is as an individual or a group/community you will have a contribution to this platform that will make your and hundreds of people’s lives easier.

Through the Coronavirus Innovation Map, we would like to especially encourage collaboration. In these challenging times, everyone can be an innovator. You can come up with your idea and directly contact us. Or if you are an innovator who sees a product or service already in the platform that is connected to your idea, feel free to contact them directly through the platform. 

How will we achieve this?
In order to create an innovation directory, we at StartupBlink have developed a technology to research and map innovation around the world since 2014. We know that curating all relevant data in a short amount of time is not achievable by a single person or company, this why we will need everyone’s help. Through crowdsourcing and the help of the community, we will be able to cover the gaps and have as much information as possible. As a community, we will be able to reach this ambitious goal of connecting innovators and people with solutions.

Do you know of an innovation helping you and others cope and adapt in these difficult times? Maybe a product or a resource created specifically to deal with the coronavirus crisis? Something we did not think we needed just a few weeks ago and now it is a must? If you answer any of these questions with a yes, then you are eligible to pitch-in through filling up this form.  

Let’s not forget that innovation requires collaboration. A small idea you think of today may become a solution to the big problems the world is currently facing. Thus, we really appreciate your collaboration with other innovative minds and contributions to the innovation directory.

willivision, 2020-03-20:
Here is the link to the simple form to get this project on the global map. This is a free service
https://docs.google.com/forms/d/e/1FAIpQLSdz4ti13PN6GPKIfcZzb85d1qSQu_r4qz8v2Y1DtK8M4aYBJA/viewform

peteflorence, 2020-03-20:
@jcl5m1, how do you think you'd potentially deploy funding for this project?

Maybe some initial funding could pay for buying developer kits for people to make around the world?

willivision, 2020-03-21:
I don't have any direct affiliation with Startup Blink.
I just know that they are good at helping startups get money and they are
launching this global map.
They asked me to volunteer to tell inventors about the global map.
You fill out a form to have your project included on it.
Here is a form to add a project to the map
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=978857ca2f&e=6467527f85>
.

Here is the email I got about the global map project, which seems like a
good fit for what you are doing.

I don't have any direct affiliation with Startup Blink.I just know that
they are good at helping startups get money and they are launching this
global map. They asked me to volunteer to tell inventors about the global
map.You fill out a form to have your project included on it.
Here is the email I got about the global map project, which seems like a
good fit for what you are doing.

---------- Forwarded message ---------
From: StartupBlink Global Startup Map <feedback@startupblink.com>
Date: Thu, Mar 19, 2020 at 8:03 AM
Subject: Will you help us build the Coronavirus Innovation Map?
To: <willivision@gmail.com>

Hello Startup Heroes,

Firstly, the word heroes apply today more than before. Heroes keep on
building while actively finding solutions for difficult situations, while
others passively watch the news unfolding. By choosing to innovate, you are
adding to our collective ability to change adverse situations. Please keep
building. History shows that innovation matters the most and has most
chances of success in times of crisis, since you will be between a unique
few doing so.

We are also doing our part in StartupBlink. In this newsletter, we will be
introducing our new global resource, ask for your help on adding
innovations related to the coronavirus, and invite you to a
special online pitching event focused on startups that are dealing with the
coronavirus.

*Coronavirus Innovation Map by StartupBlink 🌏*

As for this virus is related, the following questions come to mind: Who is
working on solutions and where?

We want to tackle this question with the creation of a global startup map
of coronavirus related innovative solutions which will be launched next
week. Those initiatives and projects will range on a wide variety of
topics, from treatment, diagnostics, prevention and lifestyle changes
(remote work, EdTech, etc.)

This will allow entrepreneurs who work on similar projects to connect and
fight a common enemy together, in addition to an innovative list of
solutions for those who are currently searching for them.

*How can you help? 🤝*

Firstly, If you have a startup that is dealing with those topics we would
love to add it to the map to make sure it’s there before we launch. Here is
a form to add a project to the map
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=978857ca2f&e=6467527f85>
.

We are also looking for volunteers who can dedicate a minimum of 10
hours to help us curate and add information on the map. If you are
interested, please fill in this form
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=9b7ad90f0d&e=6467527f85>,
if relevant, we will send you a link to join our telegram group so we can
start working together.
[... 70 lines cut ...]

Stay safe and healthy!

StartupBlink Team

P.S. Do you want to get more done? Join us for coworking sessions on
Telegram at https://t.me/workjams
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=36bcb8e03d&e=6467527f85>
and
meet fellow startup founders, entrepreneurs and freelancers like you!
*Follow us on Facebook
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=ba6c540a8a&e=6467527f85>,
Twitter
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=d27bf091c5&e=6467527f85>
or LinkedIn
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=3e4867941e&e=6467527f85>.
You can also visit our website
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=c2e39e3bc1&e=6467527f85>
or send us an email
<https://startupblink.us3.list-manage.com/track/click?u=8ea35dfd8a2f6ea0e2f245838&id=0cc3396c15&e=6467527f85>.*

*Copyright © 2019 StartupBlink, All rights reserved.*

You are receiving this mail because you are Blinking on StartupBlink or
asked to receive information.

willivision, 2020-03-21:
I am also working on my own project that is in the testing phase

thesound.com
Find Live Streaming of Favorite Artists at Home and Live News Updates About
Coronavirus

I filled out the form for my project to be on the global map and it was
very easy.
They just asked for the information you want about your project on the
global map and how you want to be contacted.
We can easily start a crowd funding.