jcl5m1/ventilator #49: Request for 3D models

jcl5m1, 2020-03-21:
I've added a section on the wiki that contains a list of 3D printable parts that are already available, or models that I have received to design.  Please take a look if you can help.  The wiki page will likely get updated as requests come in.

https://github.com/jcl5m1/ventilator/wiki/3D-Printable-Parts

RomanTheLegend, 2020-03-22:
I made a draft CAD model for CPAP outlet filter, feedback appreciated:

Based on this model:
https://www.cpap.com/productpage/generic-bacteria-filter-cpap-machines-10-pack

Files attached to this PR:
https://github.com/jcl5m1/ventilator/pull/44

As far as I've seen all of these parts have same "ports", that is
Male Connector End Outer Diameter: 15mm
Female Connector End Interior Diameter: 22mm

Submitted as draft because I do not have filters themselves to check the dimensions (waiting for delivery)
Also because I set wall thickness to 2.5 it will somehow affect the volume of airflow, so it might require a bit of an extra pressure to produce same throughput.

Recommend to print it with transparent PTE-G

Additionally attaching Fusion360 project in case if somebody could test and do the adjustments sooner

jcl5m1, 2020-03-23:
Great, thanks! Would anyone be able to this?

RomanTheLegend, 2020-03-25:
I've got the filter, reworking the model

axjjienn, 2020-03-30:
I submitted a pull request #89 for CAD based on the filter in the link (they are now back ordered). I want to build the adapter, but I guess I am not sure what we are connecting it to? I can throw together some quick 3D printed barbed hose adapters with some basic o-rings to provide a simple 0.5 atm seal. Let me know if that is what we're looking for.

RomanTheLegend, 2020-03-31:
Let me check your pull request. Did you build it off the actual model or some online blueprints?
I made a CAD based on model as well, but it needs some refinement for better printability

RomanTheLegend, 2020-03-31:
Replied in PR comments. I'll try to push my updated design today along with source project, so we could collaborate and improve it further. Right now the biggest challenge for me is how to build filter's supporting frame without relying on Support structures during 3D printing

RomanTheLegend, 2020-04-01:
@jcl5m1 could you please create a new branch for CPAP ventilators project? Now that there's more than one collaborator we'll need to cooperate

jcl5m1, 2020-04-01:
I added you as a collaborator, so you should be able to push new branches
and merge to master.

axjjienn, 2020-04-02:
@RomanTheLegend That is just the CAD for an off the shelf, verifiable viral-bacterial filter for CPAP machines. (I bought some from the link in the project before they went out of stock) I added it as a reference. I am still not clear what we want to adapt that too. I was imagining that this would be on the output side of the air compressor/pump, but I don't have any CAD of the compressor/pump to create the mating coupler (which would be a 3D printed part). If there is a part number for one, I can build the coupler relatively quickly.

RomanTheLegend, 2020-04-03:
I made a branch for the model, please PR your CAD here:
https://github.com/jcl5m1/ventilator/tree/cpap-ventilator

You approach idea of having reference model is good and will be useful. I'm putting a quick README with guidelines and suggestions for model design & production and your CAD will be a good starting point.

Apologies for delay in replies, I'm involved with few other projects and sometimes it's hard to give proper answer in time

RomanTheLegend, 2020-04-04:
I created a Wiki page for the part:
https://github.com/jcl5m1/ventilator/wiki/Bacterial-Viral-Filter-for-CPAP-Machines

And pushed first working version to cpap-ventilator branch (guess should've named it cpap-ventilator-filter though)
Once design #2 - a one-piece part with embedded filter is ready I'll merge the PR into master.  BTW, @jcl5m1 for this repo do you prefer merge or rebase?

nimamoslemy, 2020-05-01:
Hi, we are a team that needs your feedback to complete our idea.
Please visit our pages and comment on it. 
[github](https://github.com/nimamoslemy/ventilator)
[gitlab](https://gitlab.com/nimamoslemy/avp-ventilator)