jcl5m1/ventilator #53: Alarms display

ASdOliveira, 2020-03-23:
Hi! I'm Arysson Oliveira, an electronic engineer with some experience in firmware developement. 

I'd like to help your project developing an software interface, that will show in a LCD display(16x2) some informations to the operator, because i think that should have any human-machine interface. 

So, how could i help you?

Julhh, 2020-03-23:
Hi, 
We were inspired by the Code-Life challenge and started working with a team of engineers on a concept as well. I have gotten started on building a Qt application as a HMI to run on a Raspberry pi to monitor/adjust settings and display alarms. Once my concept is a little further with more detailed plans I will share the info here for anyone to collaborate. There apparently is another group somewhere working on this and I am trying to get in touch with them.

Johnny @jcl5m1 , We were thinking about a very similar direction to your work when we found out abut it and are working on figuring out some of your TODO's as well. Would you be interested in getting in touch to discuss and collaborate with our team or is it easier for you if we just share our results in your repository whenever we have something concrete? Please let me know so I gan share my contact details with you.

Cheers,
Julian

aliakanan12, 2020-04-08:
> Hi! I'm Arysson Oliveira, an electronic engineer with some experience in firmware developement.
>
> I'd like to help your project developing an software interface, that will show in a LCD display(16x2) [...]

hello can you help me with display ? i add display but lcd.print is slowing down functions
can i contact you on whatsapp ?

ASdOliveira, 2020-04-08:
> hello can you help me with display ? i add display but lcd.print is slowing down functions
> can i contact you on whatsapp ?

Of course!! Tell your whatsapp, and i'll contact you

aliakanan12, 2020-04-08:
+96176653802 thank you :)

jcrubino, 2020-04-08:
Hi Folks, some people in my network from San Antonio, Texas are working on a related [automated ambu bag](https://therivardreport.com/city-innovators-prepare-for-ppe-shortage/) and looking for a arduino developer.  Ping me if interested and local enough to help.

ASdOliveira, 2020-04-09:
> Hi Folks, some people in my network from San Antonio, Texas are working on a related [automated ambu bag](https://therivardreport.com/city-innovators-prepare-for-ppe-shortage/) and looking for a arduino developer. Ping me if interested and local enough to help.

can you send me your whatsapp man?

hkg1512, 2020-06-14:
I have worked on many LCD Display based systems. Is there anything I can contribute to?