jcl5m1/ventilator #55: Manufacturability

harrison-caudill, 2020-03-23:
Would appreciate a community opinion on this thought.

It appears as though 99% of the complexity would reside within a few components:
 * control unit
 * interface
 * flow sensor (for Tidal Volume control)
 * pressure sensors

All of those components are available in the hundreds of thousands on digikey right now.  A build-your-own-kit seems manufacturable en-mass permitting doctors to utilize whatever pumps and filters and hoses happen to be on-hand.

 * Control Unit: The software could be theorem-proven and verified.  It could include necessary calibration routines, and calibration kits would be needed in much smaller quantities.

 * Interface: The single interface would permit improvised ventilators the world around to share an interface, making training easier.

 * Flow Sensor: There's probably a reasonable way to rig up a hot-wire anemometer with appropriate sensitivity within the necessary flow-rate ranges to be useful (haven't one the math on that one yet)

 * Pressure Sensor: Available on digikey

If all items were combined within a single control unit that triggered external relays for pumps and valves, then it might empower groups to build their own using supply-chain segments that are unlikely to be affected.  For example, several impeller-style bilge pumps for marine pleasure-craft can produce the necessary pressure gradient, work on 12 volts, and are unlikely to be otherwise necessary for the pandemic.  Diesel fuel filters may not stop virus particles (they are 2 micron), but would catch many droplets.

This entire concept, of course, is predicated upon the concept that availability of physical pieces such as pumps and control units, are the limiting factors.

qsullivan11, 2020-03-28:
Interfaces and air flow sensors are available in the automation industry. 
For interfaces, a Red Lion, PanelView, or C-more all work well and are fairly user-friendly.
For air flow sensors, some trusted brands I've worked with are [IFM](https://www.automationdirect.com/adc/shopping/catalog/process_control_-a-_measurement/flow_sensors/thermal_flow_sensors/fts200-1001), Sick, Allen Bradley, and sometimes [Automation Direct](https://www.automationdirect.com/adc/shopping/catalog/process_control_-a-_measurement/flow_sensors/thermal_flow_sensors/fts200-1001).

harrison-caudill, 2020-03-28:
There are only 85 in-stock in that location -- is there a good place to purchase those in the 10's or 100's of thousands?

My intuition is that opamps, nichrome wire, etc are likely to be things that could be sourced immediately and globally in quantities sufficient to meet demands.  Glances at digikey generally confirm that, but I have not yet done a proper analysis.

qsullivan11, 2020-03-28:
That kind of part volume is the greatest challenge, as I've only been an end-user for specific applications, never bought more than ten of any part at a time. There are three other [similar models](https://www.automationdirect.com/adc/shopping/catalog/process_control_-a-_measurement/flow_sensors#Media_Type_ms=%22air%22&start=0), but the total is still only about 320 sensors. 
[DigiKey has 15 different flow sensor models](https://www.digikey.com/products/en/sensors-transducers/flow-sensors/520?FV=1129%7C312356%2C-8%7C520&quantity=0&ColumnSort=-1000009&page=1&stock=1&k=flow+sensor&pageSize=25&pkeyword=flow+sensor) of 2k each for 30k. They are all very expensive ranging from $145-$335 each. 
Since flow is a fixed variable during assist control, it might be more realistic to eliminate a flow sensor and use a mechanical air flow control valve that has a known flow rate.
I'm an electrical engineer and haven't ever used a mechanical air flow control valve because they can be a bit more crude, but they can be researched.
I've started some research on ventilators on a [Google doc](https://docs.google.com/document/d/16_ljhqduKkBL8wyL6JS9CZubEWPDG3U1NrUIbJ3w_3Y/edit?usp=sharing) I made.