jcl5m1/ventilator #58: Love to connect and collaborate from Cleveland

jephos, 2020-03-24:
Hello Johnny,

This is amazing work you've done on this thus far, and I woke up this weekend from my own CPAP wondering the same questions and started making calls.

I've also been in touch with Matt Hearn who has written a great medium article himself on the topic and has referenced your work, but yours has been the most comprehensive effort compiling facts and feedback that corroborates with many of my conversations today about the core issues around adapting CPAP's purposefully.

I'm working on getting a hold of Robert Chatman, whose book on fundamentals of ventilation mechanics you linked, who is from Cleveland.  I also have medical industry design firm interested and reaching out to their respiratory industry contacts.  BioEnterprise in Cleveland has also expressed interest and support, and there may be a proposal to Vulcan (US Govt) due this Wednesday that might be worth entering.

What is the best way to be in touch?

All the best,
JT Tan
JT at Case dot edu

nasocializes, 2020-03-24:
Per Johnny on another response:
Feel free to choose a to do, and reply to an existing issue, or create a
relevant one.

It will be easier to track work on github than via email.

DaveM135, 2020-03-24:
Hi both - am a pulmonologist/ICU specialist in SF Bay area.   Happy to set up some time to talk through what you've done.   I'm new to Github so may be missing better ways to collaborate. LMK how to be most helpful.   I have a background in biotech, medical devices, and clinical care.   Currently work in VC.

nasocializes, 2020-03-24:
Hi Dave. Thank you so much for the ping! Perhaps you can share your linkedin page so we can contact you through there without you having to leak your email :-) Johnny had a great chat with Michael McPeck, a respiratory therapist on Sunday and is now awaiting new parts to arrive. I'll check on when he's ready to chat with another pulmonologist. Thank you so much for reaching out!

jephos, 2020-03-24:
Hi @jcl5m1 @DaveM135 and @nasocializes , love to connect as you suggested.  Perhaps we could arrange a joint zoom video conference?

We love the PPE route, which we believe is one key outcome, but we're also exploring what the minimum set of adaptations could get us outcomes around the original mission that could be used 2/3 of the course of disease to alleviate the "1 respirator, 3 patients" problem that Italy faced.

I've talked with a few groups here who are working on the enclosure / filtration problem, and I've been in touch with Mike Hearn (who wrote this compile: https://blog.plan99.net/cpap-for-covid-d47886bf978c) who has been working the digital front hacking his Dreamstation.

@jcl5m1 @nasocializes Nasocializes, love to get your recommendation on how best to combine forces.

nasocializes, 2020-03-30:
FWIW, kanbans have been created to help project manage: https://github.com/jcl5m1/ventilator/projects