jcl5m1/ventilator #60: Taking a stab at the To-Do's

sbate, 2020-03-24:
I'm a programmer and a rookie on the hardware side, but I took a stab at some of the todo's from a product design perspective related to volume, pressure alarms, etc.  The product is different after that and certainly no longer portable... however, it may steer someone smarter than me in the right direction and the pieces that were added are definitely low cost and readily available.  I'd love to hear feedback!

[Low Cost Ventilator Design.docx](https://github.com/jcl5m1/ventilator/files/4373545/Low.Cost.Ventilator.Design.docx)

nasocializes, 2020-03-24:
Hi sbate,
Awesome!
Can we use google docs so we can collaborate in real time?
I converted your word doc here:
https://drive.google.com/file/d/1zwDcfDHgT7lZHVNuVxvYmhZJPMkktJme/view?usp=sharing

You can request access after clicking on the link.

sbate, 2020-03-24:
Absolutely!  I requested access.

Also a question as to whether the need for an exhaust filter could be redirected.  An intake filter would still add value but at that point it wouldn't need to be N95.  If we have a pressurized system exhaust/exhaled air can be redirected to a container of alcohol, bleach, boiling water, or other deadly trap which would accomplish the same task without using scarce resources like an N95 filter.

jcl5m1, 2020-03-25:
@sbate - This looks great! Thanks for sharing the notes!

jcl5m1, 2020-03-25:
the biggest concern I have is around measuring the volume.  I haven't found an off the shelf way to do that easily. will comment in doc.

bradtem, 2020-03-25:
How accurately can volume be computed by knowing exactly how many revolutions the blower did, how many watts that took, and the measurement of pressure at one or more places?

ameeuw, 2020-03-25:
Hi everybody! Great project.

I think volume is usually integrated over the flow rate to the interface. The flow rate is measured using a differential pressure such as [this one](https://www.sensirion.com/en/flow-sensors/differential-pressure-sensors/sdp800-proven-and-improved/). Maybe cheaper alternatives from Aliexpress such as the MPX5010DP could work if they offer a similar pressure range.
Maybe calibration could get tough though..

sbate, 2020-03-25:
Sure thing - so there were two issues that made me shy away from measuring air.... first there were complications with patient pH if you weren't increasing the amount of oxygen in the air mix, and second measuring air is hard especially at variable pressures.  Measuring liquid is comparatively easy, so filling a an air reservoir (balloon) under water turns the air volume problem into a liquid volume problem which can be solved with a level gauge aka "how full is my bucket".  That full amount of air can be delivered consistently because we're not trying to calculate it on the fly while the patient's lungs are providing variable resistance.  Measuring before the inhale also opens this design up for many alternate forms of air compression, because the blower/compressor/inflater won't be fighting the patient lungs in real time, just a balloon under constant pressure.