jcl5m1/ventilator #61: EUREKA! We discovered the at-home treatment for coronavirus that works! PLEASE READ CAREFULLY AND WATCH THE VIDEO

willivision, 2020-03-24:
[How to Kill the Virus With Hot Air.zip](https://github.com/jcl5m1/ventilator/files/4375642/How.to.Kill.the.Virus.With.Hot.Air.zip)

I have recommended for decades that anyone feeling the first sign of a cold or flu, to take an immediate hot bath to kill the virus and many of the symptoms diminish much more rapidly. This has proven to be very effective in many [research studies](https://www.sciencedaily.com/releases/2011/11/111101130200.htm).

The problem is that the coronavirus can withstand temperatures that are too hot for the normal person's body. The [coronavirus dies at a temperature of 130F](https://www.who.int/csr/sars/survival_2003_05_04/en/) to 133 F (56C). At hot bath, will not get the body to that temperature inside.

It turns out that the coronavirus lodges in the sinus cavities **before** the pneumonia starts. This is why the test for the coronavirus is done by sticking a swab up your nose and you can be without symptoms for up to five days after infection. Once it infects your lungs, you are likely to get extremely ill. 

HOT AIR will kill the coronavirus. We can safely breathe in hot air that is a temperature of 130F (56C). People who live in desert environments, where it gets really hot, do this all the time. People who sit in a dry sauna regularly breathe in air that is 140F+. If you have a dry sauna at home you can use it. If you do not have a dry sauna, you can use a simple hairdryer, **IF** you use it carefully.

I heard about using a hairdryer to make hot air, but my concern was that people would burn their faces. Now, we have a **SAFE** technique that can be done at home with just water, a clean water spray bottle, and a hairdryer.

1. Use the water spray to **moisten the face**. 
2. Then, use the hairdryer on the lowest setting and **use one hand to block some, not all, of the air** coming in the back of the hairdryer to reduce the volume of air. 
3. **Wave the hairdryer** in front of your face but do not let it blow directly in one place to burn you. You can feel the air get hot. 
4. **Breathe in the dry, hot air** for five minutes through the nose with the mouth closed. 
5. **Close your eyes** as much as possible during the procedure (peeking only if you need to) to keep them from drying out. Use eye drops afterward, if your eyes become dry.

Attached is a video showing the safe technique. Please watch it carefully, to the end, to learn how to do this properly.

If you experience fever, headache, dry cough, or shortness of breath, these are the signs of the coronavirus infection. Fever is usually the first sign. Take your temperature regularly, and take the temperature of the ones at home with you as well. If you do not have a thermometer, use your hand to touch your forehead. You can easily feel a strong fever this way. 

Use this breathing dry hot air technique the moment you notice a fever beginning. Continue to use it as described in the video.

You will not kill all the virus and you will still be infected, and you still may get sick. However, you will lower the viral load as it enters the body through the nose. This will help your immune system withstand the viral attack.

_**I have verified the information in this video by reading the scientific research papers (published and peer-reviewed) and it is 100% accurate.**_

**Precautions:**

If you are sick, always consult with a health practitioner over the telephone. This is not a substitute for receiving adequate medical care and not medical advice, only a healthy defensive technique that you must do properly, to do it safely. YOU CAN BURN YOUR FACE IF YOU DO THIS INCORRECTLY! Use caution and follow the instructions in the video. If you feel discomfort STOP! Wave the hairdryer to have it create hot air in front of your nose that you can breathe in. Do NOT hold the hairdryer still, while it is pointed at your face. Elderly people and children need assistance to perform this technique correctly.

**Sources:**
Elevated body temperature helps certain types of immune cells to work better, evidence suggests  
[https://www.sciencedaily.com/releases/2011/11/111101130200.htm](https://www.sciencedaily.com/releases/2011/11/111101130200.htm)

Heat at 56°C kills the SARS coronavirus at around 10000 units per 15 min
[https://www.who.int/csr/sars/survival_2003_05_04/en/](https://www.who.int/csr/sars/survival_2003_05_04/en/)

Thanks to Robert in Romania for finding this video for us.

**PLEASE PASS THIS INFORMATION ON TO OTHERS
We can defeat this virus with accurate information.**

satchell78, 2020-03-24:
Sounds painful. [cough]

willivision, 2020-03-24:
It is safe, if you do this properly.  You can visit Death Valley to breathe
in hot desert air, to do this naturally too.

willivision, 2020-03-24:
Please review the scientific papers before dismissing the idea. Temperature of 130 F (56 C) is one way to kill the virus. The study on temperature for killing SARS coronavirus is a good proof for SARS-COVID 19. This is not conspiracy nonsense. I spend all day fighting misinformation. 

Elevated body temperature helps certain types of immune cells to work better, evidence suggests
https://www.sciencedaily.com/releases/2011/11/111101130200.htm

Heat at 56°C kills the SARS coronavirus at around 10000 units per 15 min
https://www.who.int/csr/sars/survival_2003_05_04/en/

willivision, 2020-03-24:
Please follow these precautions written by the video producer
Dan Lee Dimke, PhD, CEO, Future-world.com

CAUTION: The experimental procedure presented in this video is NOT a cure for coronavirus! It has not yet been tested or endorsed by any health authority. It should be ONLY be considered as one more layer of potential prevention IN ADDITION TO ALL OTHER PRECAUTIONS recommended by the CDC, WHO, or other health authorities – including hand washing, social distancing, coughing and sneezing into a sleeve, avoiding contact with others who display symptoms, getting tested for COVID-19 if fever and other respiratory symptoms develop, and seeking immediate medical advice if tested positive for COVID-19, or if symptoms persist or worsen. NOTE: Once the virus has migrated to the lungs, which may occur in up to 20% of patients within one to two weeks, the virus is too far from the external heat source to be substantially impacted by this procedure. Benefits are likely to be highest when used for prevention and early treatment. THIS VIDEO CONTAINS NO MEDICAL ADVICE AND SHOULD BE REGARDED AS EDUCATIONAL INFORMATION ONLY.

willivision, 2020-03-24:
This is not a cure. This is preventative help that is only useful when you first notice the virus while most of it is still in the sinuses. it is to slow it down and make it less severe if possible. Once it enters the lungs, there is no way to heat your entire body to 130F (56C) to kill the virus inside body without harming the person.

I am starting a research study to track the coronavirus spread in desert areas with dry, hot air. The virus is NOT stopped in hot, humid areas at all. I believe the dry natural air may help and my study is now underway to check the statistics.

chakahamilton, 2020-03-24:
This is junk science it is already been disproven a few days ago. Please be more responsible when contributing.

willivision, 2020-03-24:
You have the right to your opinion but I read scientific journals.

willivision, 2020-03-24:
Hi Chaka, Please read this new scientific paper if you have the time.
It was published by the University of Maryland. Regards, William
IHV Researchers Use Climate to Predict Spread of COVID-19
https://www.umaryland.edu/news/archived-news/march-2020/researchers-predict-potential-spread-and-seasonality-for-covid-19-.php

willivision, 2020-03-24:
Please also give me the links that have "disproven" the effects of 130 F
temperature does not kill the virus. The hard part is getting that level of
heat inside a person.

Here are some examples of other myths and how these are disproven by
studies.

https://www.livescience.com/coronavirus-myths.html

willivision, 2020-03-24:
Here is the study about climate temperature and humidity
[Climate Study.pdf](https://github.com/jcl5m1/ventilator/files/4376689/Climate.Study.pdf)
  

Temperature and latitude analysis to predict potential spread and seasonality for COVID-19

willivision, 2020-03-24:
The temperature of 130 F (56C) is the key to unlocking more about this idea. Dry saunas could be very helpful.

chakahamilton, 2020-03-24:
Your links are from 2011 9 years ago here is the link your requested.
[Here](https://www.nydailynews.com/coronavirus/ny-coronavirus-florida-politician-promotes-fake-cure-theory-20200323-iai3oo6mznd5tj33wdaopzlxsy-story.html)

willivision, 2020-03-24:
Thanks for your info, I will read them.

Yes, the one link I used was for regular flu that was proven 11 years ago,
but the coronavirus is stronger.
The temperature studies are what temperature kills SARS coronavirus.
The technique of killing the virus in humans with hot temperature is
needing improvement but that is what inventors do.

willivision, 2020-03-24:
Yes, I am aware of this. Snopes says curing the coronavirus with a
hairdryer is FALSE. see
No, a Hair Dryer Won’t Stop Coronavirus -
https://www.snopes.com/fact-check/hair-dryer-coronavirus/

The misconception is that this is NOT a cure. It is a preventative aid to
slow getting sick. It needs improvement.

*Dan Lee Dimke, PhD* is a scientist, not a medical doctor. Anyone with a
PhD is also called a "doctor" or a professor if they teach.

Here is what he actually said to me in an email:

"Some internet sites have duplicated this video without permission and more
worryingly, are promoting it as a coronavirus cure, rather than as an
additional prevention and early intervention step. We want to do everything
possible to prevent this misconception."

Thank you for writing to request this.

Warmest regards,

Dan

Dan Lee Dimke, PhD

CEO

Future-world.com

-------

Gargling with warm water does not cure a common cold but does help to
recover from one.

It took ten years for the research to show that raising the body
temperature for those with the common flu helped them and that is why I
shared the link to the old study.

I don't think we can stop this investigation into the temperature effects
on killing coronavirus because one exaggerated claim was discredited by
Newsday and Snopes.

I am a founding member of Snopes! The exaggerated claim is FALSE. Even the
guy who created it says this.

The usefulness of using a higher temperature to lower viral load in the
human body is NOT disproven and is NOT proven either.

People could be saying Johnny Lee's invention of the DIY ventilator would
kill the patients!

He knows it is not ready but that is not stopping him from making
improvements and asking for help.

Perhaps *warming up the ventilator air* would also help the patients. Have
you considered that?

This is why I put the discussion here because it might help and it needs a
serious and fair investigation.

What we need is a random large scale trial of how breathing in warm air
over 130 F impacts the coronavirus.

Luckily, we do not have to organize this huge human trial.

It is happening for us naturally as the virus spreads to warmer parts of
the world.

And we are studying it.

“I have not failed. I’ve just found 10,000 ways that won’t work.” Thomas
Edison

"Don't throw the baby out with the bathwater."

A kernel of truth may be hidden in BS.

willivision, 2020-03-24:
At the moment we do not know. As long as you do not accidentally burn yourself, breathing in warm air cannot hurt. It is certainly less dangerous that taking the anti-malarial drug [chloroquine ](https://thehill.com/policy/healthcare/489097-man-dies-after-taking-chloroquine-phosphate-in-effort-to-prevent-coronavirus)that Pres. Trump is bragging about, which is killing people.

from [https://www.livescience.com/warmer-weather-slow-coronavirus-spread.html](https://www.livescience.com/warmer-weather-slow-coronavirus-spread.html)
Could the summer bring an end to COVID-19?
By Yasemin Saplakoglu - Staff Writer 8 hours ago 3-24-2020

Warmer temperatures and higher humidity could help to slow spread of the virus.

 
 
 
 
 
 Comments (3)
Could summer mean a slower spread of COVID-19?
(Image: © Shutterstock)
Like some other respiratory viruses such as the flu, is there a chance that the new coronavirus will spread less as temperatures increase? 

A new study has found that the new coronavirus, named SARS-CoV-2, didn't spread as efficiently in warmer and more humid regions of the world as it did in colder areas. Though the early analysis, published in the journal Social Science Research Network, is still under review, it provides a glimpse into what we might expect in the warmer months to come. 

Qasim Bukhari and Yusuf Jameel, both from the Massachusetts Institute of Technology, analyzed global cases of the disease caused by the virus, COVID-19, and found that 90% of the infections occurred in areas that are between 37.4 and 62.6 degrees Fahrenheit (3 to 17 degrees Celsius) and with an absolute humidity of 4 to 9 grams per cubic meter (g/m3). (Absolute humidity is defined by how much moisture is in the air, regardless of temperature.)

Related: 13 coronavirus myths busted by science

In countries with an average temperature greater than 64.4 F (18 C) and an absolute humidity greater than 9 g/m3, the number of COVID-19 cases is less than 6% of the global cases. 

This suggests "that the transmission of 2019-nCoV virus might have been less efficient in warmer humid climate so far," the authors wrote. Humidity especially might play a role, given that most of the transmission of COVID-19 has happened in relatively less humid areas, they wrote.

CLOSE
But that doesn't mean that when summer rolls around, social distancing will be obsolete and people will once again pack into bars and concerts like sardines.

For most of North America and Europe, the effect of humidity on the spread of the coronavirus would be negligible until June, when levels start to increase beyond 9 g/m3, the authors wrote. Still, with over 10,000 cases of COVID-19 being reported in regions with average temperatures of 18 degrees C (64.4 degrees F) after March 15, the role of warmer temperatures in slowing the spread might be observed only at much higher temperatures.

"Therefore its implication will be limited at least for northern European countries and northern U.S., which do not experience such warm temperatures until July, and that too for a very short time window," the authors wrote. So the chances of reducing the spread of COVID-19 due to these environmental factors would be limited across these areas, they added.

"It's unreasonable, I think, at this point to expect that the virus will quote-on-quote disappear during our summer months," said Dr. William Schaffner, an infectious disease specialist at Vanderbilt University in Tennessee, who was not part of the study. Still, "I think it might give us a little bit of hope," Schaffner said.

The spread of some respiratory viruses, such as the flu viruses, diminishes in high humidity and high temperatures. It's not exactly clear why temperature and humidity affect the flu virus or other seasonal viruses, but it's in part because when you exhale, some virus at the back of your throat gets pushed out into the air, Schaffner told Live Science. "If we were to get a microscope and look at that virus, we would discover that it's surrounded by a microscopic sphere of moisture" called a droplet, he added.

Related: How does the new coronavirus compare with the flu?

When you have low humidity in the wintertime, that sphere of moisture tends to evaporate, which "means that the virus can hover in the air for a longer period of time because gravity won't pull it to the ground," Schaffner said. But in the summer, when you exhale a viral particle, the surrounding droplet doesn't evaporate, which means it will be heavier and gravity will pull it out of the air much more readily. In other words, "it doesn't hover as long as it does in the winter," making it less likely to infect the person close by, he said.

Transmission of the flu goes down to very low levels during the summer, so we don't typically have to worry about it very much in warmer months, he added. But other viruses, such as the coronavirus strains that cause the common cold, "have a seasonal distribution that is not as dramatic as influenza," Schaffner told Live Science.

Still, "we can't count on" the warmer and humid months to slow the spread of the virus, Schaffner said. "We have to beware of wanting to walk only on the sunny side of the street — there's another side that's shadier."****

willivision, 2020-03-25:
debunking myths from WedMD  says NOT Sure for the temperature affects on coronavirus
Experts Sort Fact From Fiction on COVID-19 Myths
[https://www.webmd.com/lung/news/20200324/experts-sort-fact-from-fiction-on-covid-myths](https://www.webmd.com/lung/news/20200324/experts-sort-fact-from-fiction-on-covid-myths)

TUESDAY, March 24, 2020 

'Like colds and flu, COVID-19 will fade with warmer weather.'

**False -- maybe.** There is no conclusive proof that the coronavirus will die off once the weather turns warm. The reverse is also true. There is no conclusive proof that warm weather and warmer air will diminish coronavirus. A good scientist will wait and see.

"Because this is a new virus, we aren't sure," said Catherine Troisi in the release. She's an epidemiologist and associate professor in the Department of Management, Policy, and Community Health at UT Health's School of Public Health.

One recent study, led by virologist Dr. Mohammad Sajadi of the University of Maryland School of Medicine, has suggested that coronavirus might prefer cooler, more humid climes.

"Based on what we have documented so far, it appears that the virus has a harder time spreading between people in warmer, tropical climates," Sajadi said.

But even if that's so, the fact that humans have no immune experience against the virus means it will probably continue to spread during the Northern Hemisphere's summer, other experts countered.

"We hope that warmer weather will help, but there is no guarantee," Troisi said. "What ultimately helps is that summertime means kids are not in school anymore, and they are less likely to pass viruses around."

jcl5m1, 2020-03-26:
I would like to keep the topics on this github focus on engineering and medical guidance around the requirements of ventilators and PPE.  Thanks.

closed 2020-03-26 by jcl5m1

willivision, 2020-03-26:
Are you may be aware that there are heated tubes that raise the air
temperature for CPAP machines.
Some work with humidifier and others heat dry air.
This is related to what you are doing. If warmer air is a
potential treatment that might help reduce viral load, your device might be
able to deliver it effectively.

Please keep an open mind because we are still testing the theory about
reducing viral load (NOT a cure) with heat to make recovery outcomes
improve.

Here are the heated CPAP notes from a CPAP company and example of the CPA
heated tubing.
The devices are already used to deliver warm air in winter for more patient
comfort.

Using a hair dryer is silly. Using a heated CPAP tube might be helpful.

"*Heated Tube:* You can connect a heated CPAP tube
<https://www.cheapcpapsupplies.com/cpap-supplies/tubing-hoses> to your
device to heat the air passing through it. It can add moisture to the air
if used with a passover humidifier."

   - *Respironics Heated Tubing see *

https://www.cheapcpapsupplies.com/dreamstation-heated-tubing
[image: Respironics Heated Tubing]
<https://www.cheapcpapsupplies.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/h/t/ht15.jpg>
More Views[image: Respironics Heated Tubing]
<https://www.cheapcpapsupplies.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/h/t/ht15.jpg>
Respironics Heated Tubing

Heated Tube for the DreamStation and SystemOne PAP Machines.

This tubing can only be used with DreamStation and SystemOne CPAP Machines.

willivision, 2020-03-26:
You have been upstaged by ELon Musk. Oh well I through you could see things
in a new way guess not.