jcl5m1/ventilator #62: Basic pump / inflated use :why not?

fredohmygod, 2020-03-24:
What about using a basic pump like the one on amazon proposed in the read me ? Has someone tried ?
Any help with which Arduino extension card to purchase to pilot such a motor?

adamcullen, 2020-03-25:
Hi, I've just had a go at building an open source ventilator using the main thread as inspiration. I didn't have access to a CPAP pump, so I decided to give an air mattress pump a go.  Here are my findings.

I used (all prices in Australian dollars):
- 12v DC air pump (A$19)
- Arduino (I've used a mega as that's whats in my test lab, but in reality I'd use a nano)  (about $5)
- WB291111 motor driver board (which looks to me like an L298N knock off)  (about $2)
- 4 potentiometers (less than $1)
- washing machine flexible hose  (about $30, but I'm sure there must be cheaper options)
- 3M 6800 full face respirator mask (about A$90, but I had one following the bushfires!)
- 1 1/4" to 1" BSP reducer (A$2.50)
- 1" to 3/4" BSP reducer (A$2.50)
- 3/4" to 20mm pressure pipe adaptor (A$2)

Here are some pics of the testing setup:
[image]
[image]
[image]
[image]
[image]
[image]

I decided to re-use my 3M 6800 full face respirator mask, which I have following the last disaster to strike Australia earlier in the year - bushfires.  This mask has 3 filter ports, and the large one at the front takes a very course 1 1/4" thread. I found the 1 1/4" BSP thread will get about a 3/4 turn and make a reasonable seal, threadtape or plumbing sealant would make a solid seal in an emergency. This mask has the added advantage of being able to hook up an air feed line to the other 2 ports, which I can add in concentrated oxygen if needed. 
https://www.ebay.com.au/itm/Indusry-Safety-Spray-paint-Supplied-Air-Fed-Respirator-System-For-6800-Gas-Gask-/264207440778

I used the 4 potentiometers to control the strength of the pump on in and out, as well as the length of the in and out breath.  As you'll see in my crude arduino code, the pots basically divide the analogue signal by 100 and round (so instead of reading 0-1028 on the analogue port, they get a value of 0-10) making the pots able to have 10 steps from 0 to max volume, and I have arbitrarily set the breath (in and out) length to 4000ms.

The only other tricky part was wiring up the L298N motor dirver so that it used the A and B motor outs in parallel.  The L298N will only drive up to 2A, I needed at least 2.5A, so by running it in parallel I could get up to 4A.  This is possible per the spec sheet on page 7, by bridging inputs 1 & 4 and 2 & 3.  https://www.st.com/resource/en/datasheet/l298.pdf

Here's my code for the arduino.
```
int breath = 1;

void setup() {
  Serial.begin(9600);
  pinMode(9, OUTPUT);
}

void loop() {
  int inP = analogRead(A0); // pot 1 - in breath pressure
  int outP = analogRead(A1); // pot 2 - out breath pressure
  int inT = analogRead(A3); // pot 3 - in breath time
  int outT = analogRead(A2); // pot 4 - out breath time

  int inP_v = round(inP/100)*25.5;  // take the value from the pot, divide by 100 and round, then multiply by 25.5 so the output value is a multiple of 255
  int outP_v = round(outP/100)*25.5;  // take the value from the pot, divide by 100 and round, then multiply by 25.5 so the output value is a multiple of 255
  int inT_v = round(inT/100)*400;  // take the value from the pot, divide by 100 and round, then multiply by 400 to get a proportion of 4000ms
  int outT_v = round(outT/100)*400;  // take the value from the pot, divide by 100 and round, then multiply by 400 to get a proportion of 4000ms

  if (breath !=0) {breath = 0;}
    else breath = 1;

  switch (breath) {
    case 0:
      Serial.print("Breath In. Pressure: "); Serial.print(inP_v); 
      Serial.print(". In Time: "); Serial.print(inT_v); Serial.println("ms.");
      analogWrite(9, inP_v);
      delay(inT_v);
      break;

    case 1:
      Serial.print("Breath Out. Pressure: "); Serial.print(outP_v); 
      Serial.print(". Out Time: "); Serial.print(outT_v); Serial.println("ms.");
      analogWrite(9, outP_v);
      delay(outT_v);
      break;
  }
}
```

Hopefully this helps someone.  I've spoken with an ER doctor friend, who said using an inflation pump cant be any worse in an emergency than someone having to manually "bag" a patient for a couple of weeks!  I haven't been able to run a water lift test as my hose isn't opaque, but I'll try to find some and report back. I'd also like to do a test running the pump overnight to see if the motor burns out cycling up and down.

jcl5m1, 2020-03-25:
this is really great!  eager to hear how it runs over night going through breathing cycles.

fredohmygod, 2020-03-25:
Exactly what I have been looking for. Thanks !
Thinking about mixing it with decathlon snorkeling mask recently adapted by some hospitals in Italy 

https://www.google.fr/amp/s/www.01net.com/actualites/comment-decathlon-pourrait-sauver-des-malades-du-coronavirus-grace-au-plan-de-son-masque-de-plongee-1881292.html/amp/

adamcullen, 2020-03-25:
I hope bit helps you guys in Europe.

adamcullen, 2020-03-26:
Final update: pump has been tested continuously now for more than 32 hours. It is still going strong, so in a dire emergency, I think this represents a viable solution.

Fearfulbot, 2020-03-26:
Where did you source the pumps?

adamcullen, 2020-03-26:
> Where did you source the pumps?

Just at a local department store like Target. I'm sure you could also source from camping, sports, hardware stores. (Target/Big W in Australia, but something like Walmart in the US)

https://www.target.com.au/p/bestway-sidewinder-ac-dc-air-pump/58980521
https://www.bigw.com.au/product/hinterland-electric-air-pump-12v-dc/p/8296727/

DaveM135, 2020-03-26:
FDA waiver just issued for alternative ventilation assistance devices.   Sponsor role still required but will allow broader use by hospitals and guidance by commercial manufacturers.   

https://www.fda.gov/media/136423/download <https://www.fda.gov/media/136423/download>

Col68, 2020-03-31:
Hi friends

Very great respect to *****Adamcullen***** for sharing this wonderful tutorial with the Arduino code as a gift. This is exactly what i was looking for and my research was successful, thanks to you dear *****Adamcullen***** 

I wish you good health, peace, success and happiness. I would share images of the final result, i would have liked to use it with an STM32F103 OR STM8 if possible with the Arduino IDE but as my knowledge is limited i would be content to use an Arduino Uno.

Thank you so much.

Carter7868, 2020-03-31:
Hello, I have been playing around with controlling a DC motor via a Mosfet. The code I have is fairly basic but if possible could someone look over it? I haven't setup potentiometer's yet for settings / speed, currently its just code variables.
```
// the transistor which controls the motor will be attached to digital pin 9
int motorControl = 9;

//variables
int bpm = 16;
float maxPercent = 1.00;
float minPercent = 0.30;

// the setup routine runs once when you press reset:
void setup() {
  Serial.begin(9600);
  // make the transistor's pin an output:
  pinMode(motorControl, OUTPUT); 
  //led pin 
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop routine runs over and over again forever:
void loop() {
  if(true){
    //LED On
    digitalWrite(LED_BUILTIN, HIGH); 
    // ramp up the motor speed
    for(int x = 255*minPercent; x <= 255*maxPercent; x++){
      analogWrite(motorControl, x);
      delay(60000/bpm/2/(255*(maxPercent-minPercent)));
      Serial.println(x);
      Serial.print(" ");
    }
    digitalWrite(LED_BUILTIN, LOW);
    // ramp down the motor speed
    for(int x = 255*maxPercent; x >= 255*minPercent; x--){
      analogWrite(motorControl, x);
      delay(60000/bpm/2/(255*(maxPercent-minPercent)));
      Serial.println(x);
      Serial.print(" ");
    }    
  }
}
```
You can set the BPM (beats per minute) as well as the min and max air flow percentage.
(Sorry if I did something wrong this is my first GitHub post)

adamcullen, 2020-03-31:
An update - added an OLED screen to give some visual feedback about in and out breath times and pressures. This requires the use of some libraries, so not quite as straight forward as a raw Arduino code.  The inclusion of the OLED screen forced me to be slightly more elegant with the code too, instead of using a raw delay for the timing, I use a set of loops with a constant delay. The number of loops is calculated based on the potentiometers. This allows me to make real time adjustments on the pot, which are output to the pump and to the display, without having to wait until the next cycle before the change takes effect. The most you have to wait is 400ms.  Also included comments to make the code more readable.

[image]

```
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
#define OLED_RESET     4 // Reset pin # (or -1 if sharing Arduino reset pin)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET);

int breath = 1;
int i = 0;
int inP=1028; int inT=1028; int inP_v=1028; int inT_v=1028;
int outP=1028; int outT=1028; int outP_v=1028; int outT_v=1028;

void setup() {
  Serial.begin(9600);
  pinMode(9, OUTPUT);

  // SSD1306_SWITCHCAPVCC = generate display voltage from 3.3V internally
  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C)) { // Address 0x3D for 128x64
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }

  // Show initial display buffer contents on the screen --
  // the library initializes this with an Adafruit splash screen.
  display.display();

  // Clear the buffer
  display.clearDisplay();
}

void loop() {
  // clear the display and set up initial values

  

  // is the next breath an in or out?  
  if (breath !=0) {breath = 0;}
    else breath = 1;
    Serial.println(breath);

  // apply the in breath or out breath code
  switch (breath) {
    case 0: //breathe in
    i = 0;
    Serial.println(i); Serial.println(inT_v);
      while (i <= inT_v) {
        // read the potentiometers and determin the in breath pressure and time
        inP = analogRead(A0);
        inT = analogRead(A3);
        inP_v = ceil(inP/100)*25.5;
        inT_v = round(inT/100);

        Serial.println("Breath in");

        // used to show the values on the OLED display
        display.clearDisplay();
        display.setCursor(0,0); display.setTextSize(1); display.setTextColor(WHITE);
        display.print(F("DIY Ventilator v1.0")); 
        display.setCursor(0,16); display.setTextSize(2);
        display.print("BREATH IN");
        display.setCursor(0,40); display.setTextSize(1);
        display.print("Pressure: "); display.print(ceil(inP_v/2.55)); display.println("%");
        display.setCursor(0,50);
        display.print("Time: "); display.print(inT_v*400); display.println("ms.");
        display.display();

        // we will output the PWM voltage and delay only 400ms here and loop through a maximum of 10 periods 
        // this allows us to re-read the pressure and time every 400ms instead at the end of each breathing cycle
        analogWrite(9, inP_v); 
        delay(400); 
        i++;        
      }
      break;

    case 1: //breathe out
    i = 0;
      while (i <= outT_v) {
        // read the potentiometers and determin the in breath pressure and time
        outP = analogRead(A1);
        outT = analogRead(A2);
        outP_v = ceil(outP/100)*25.5;
        outT_v = round(outT/100);

        Serial.println("Breath out");
        
        // used to show the values on the OLED display
        display.clearDisplay();
        display.setCursor(0,0); display.setTextSize(1); display.setTextColor(WHITE);
        display.print(F("DIY Ventilator v1.0")); 
        display.setCursor(0,16); display.setTextSize(2);
        display.setCursor(0,16); display.setTextSize(2);
        display.print("BREATH OUT");
        display.setCursor(0,40); display.setTextSize(1);
        display.print("Pressure: "); display.print(ceil(outP_v/2.55)); display.println("%");
        display.setCursor(0,50);
        display.print("Time: "); display.print(outT_v*400); display.println("ms.");
        display.display();
        
        // we will output the PWM voltage and delay only 400ms here and loop through a maximum of 10 periods 
        // this allows us to re-read the pressure and time every 400ms instead at the end of each breathing cycle
        analogWrite(9, outP_v);
        delay(400);
        i++;
      }
      break;
  }
  
}//end loop
```

Col68, 2020-03-31:
Hello dear *****Adamcullen***** 

You continue to spoil us, thank you very much for your very informative, understanding and very well explained sharing. 

Too bad for me for not having an Oled screen, is it possible to modify it for use with the 1602 I2C screen ? I will do research and shared ... 

In any case, Bravo and Respect for the unforgettable *****Adamcullen*****.

Thank you.

adamcullen, 2020-03-31:
> Hello dear _****Adamcullen****_
>
> You continue to spoil us, thank you very much for your very informative, understanding and very well explained sharing.
>
> Too bad for me for not having an Oled [...]

A 1602 i2c led will work, it just needs some code modification. The oled is i2c as well, so if you get your addressing right, set up the screen size and how the output works. Sorry, I can't be much more specific than that because every screen will be different.

Col68, 2020-04-01:
Hello dear friend

Thank you very much for the confirmation, i will do my research to succeed in adapting the screen to your project, it is for a personal experience. 

Here is an image during the test editing in amateur mode :)

[image]

I go to the job to modify the code in order to adapt it, I would share my final results and my tests to contribute to this very good project.

Respect at Adamcullen

Col68, 2020-04-01:
Hi friends

Result, does not work for me, i notice that the L298N module is different from that of Adamcullen, i do not know if it is also necessary to supply the motor with an external power supply because the outputs of the L298N module do not supply the motor .

Here is the assembly carried out:

Pinout L298N ---> Arduino 
    ENA / ENB ---> D9
       IN1 / IN2  ---> +5V ---> To powered the breadboard
       IN3 / IN4  --->  GND           

Pinout Pot ---> Arduino 
        Pot 1 ---> A0
        Pot 2 ---> A1
        Pot 3 ---> A2
        Pot 4 ---> A3

Line 5v breadboard line ---> Arduino +5V
Line GND breadboard line ---> Arduino GND
 
Pinout L298 --->  To motor
          OUT1 ---> Red 12V
          OUT2 ---> Black GND

          OUT3 ---> Red 12V
          OUT4 ---> Black GND

The potentiometers are noted 10K i do not know if it is good.

[image]
[image]

Thank you so much.

adamcullen, 2020-04-02:
> Hi friends
> 
> Result, does not work for me, i notice that the L298N module is different from that of Adamcullen, i do not know if it is also necessary to supply the motor with an external power supply because the outputs of the L298N module do not supply the motor .

My advice is to just start with your arduino and the L298N board. If you have a small 12v motor, just wire up one of the motor outs to it and see if you can just control one small motor. if you dont have a motor, a voltmeter or even a 12v light would help fault find.
What you are aiming for is to see if you can control the voltage from one of the motor outputs to make it different from the 12v in. You can run a code that just goes High Low High Low on the pin 9 output and see if you can see a change in voltage at the motor output terminals.
Once you can do that, you can then try bridging both sides of the L298N so that the voltage at motor 1 and motor 2 is the same and changes at the same time with your code.

Have a look at this video: https://www.youtube.com/watch?v=wjFW-TNq8Og and this how to: https://howtomechatronics.com/tutorials/arduino/arduino-dc-motor-control-tutorial-l298n-pwm-h-bridge/

> The potentiometers are noted 10K i do not know if it is good.

Any resistance on the pots is fine. You're just supplying 5v from the arduino and reading it back on the arduino through the pot. All pots vary from 0 to their max which will just scale the voltage.

Col68, 2020-04-02:
Hello Adamcullen 

Forgive my amateurism, thank you for your patience and your instructions which i will follow to the letter to make tests of good functioning of the module. 

Indeed the dissipator was very hot I think that maybe i made a bad assembly or that the module L298N is unfortunately out of service. On the Arduino monitor i can see the unanswered expectations of the module and the potentiometers ... 

Thank you again for your confirmation of the value of the potentiometers, thank you for your follow-up, sincere greetings to King Adamcullen.

alexandre-leblanc, 2020-04-03:
I have an open source design that was intended to use an off-the-shelf DC pump, check it out here:
https://gitlab.com/alex.leblanc/ventilificator

Col68, 2020-04-03:
Hello dear Adamcullen, following the functional test of the module L298N + Arduino + motor as you advised me, I notice that the module does not work, no output voltage is available on the outputs OUT1 and OUT2 towards the motor by supplying with + 12V and GND, I will order new L298N modules, no joy for me for the moment. Again thank you again for your support, my respect for you.

------------------------------------------------------------------------------------------------------------------------------------
Hello Alexandre-leblanc 

Thank you for sharing, it's cool to use a more powerful STM32xx than Atmega-328P. I visited your link but i found nothing of "Open Source", no necessary components, no assembly diagram, no code or sketch to inject, this could not be useful for us (the world of Open Source sharing) please specify the components and steps so that people in need try this project to evolve it ... 

Thank you for your understanding.

alexandre-leblanc, 2020-04-03:
Hi Col68, 

I've just done the hardware (no mech or software) and provided the raw design files (in Altium format). If I have some time later I can generate the outputs (gerbers, fab drawings, assembly drawings, BOM). I haven't set up the outjob file for it yet (nor have I created my fab/assembly notes).

adamcullen, 2020-04-04:
> Hello dear Adamcullen, following the functional test of the module L298N + Arduino + motor as you advised me, I notice that the module does not work, no output voltage is available on the outputs [...]

Col68, I burned up one of my L298N modules too playing around with them! So don't worry, you're not alone.

The main issue is how to supply 5v to the board so it runs, while using 12v for the motors.  I found that if you keep the jumper on the board to supply 5v through the onboard regulator, it also means the maximum output voltage is 5v. I dont know why.  Anyway, a way around this is to supply 12v through the screw terminal and 5v and a common ground (by this i mean the wire together your ground from the 12v supply and the ground from your arduino) between the screw terminals on the L298N and the 5v output on your arduino.

In summary, you need to use (from the Arduino) 5v going to IN1&4 **as well as** the 5v screw terminal. This is the red wire in the diagram below.
**Plus** you need to run ground from the Arduino to IN2&3 as well as the ground on the screw terminal. This is the black wire below.
**Finally**, you need 12v going just to the 12v screw terminal. This is the purple wire below. Take all the jumpers off the board.

Then you can control the voltage coming out of the Motor 1 and Motor 2 screw terminals by using pulse width modulation from pin 9 on the arduino to EN1 and EN2 on the L298N. This is shown as the yellow wire below.

[image]

[image]

Col68, 2020-04-04:
Hello dear Adamcullen

Respect and happiness for you, i will try your modification, thank you very much for your support, the diagram and the detailed instructions. 
I will try to see if hope is reborn. 

I did other experimental tests with a PWM module max 24v but the motor capacitor was extremely hot :), then i used an IRF520 module + 50K potentiometer. 

[image]

The main power supply is a 12V 5 Amp transformer, i used 12V + GND on the VIN / GND outputs of the Arduino and motor inputs on the IRF520 module, i was impressed by the good functioning of the modules. 

[image]

The PWM setting with the 50K potentiometer is very precise and the IRF520 chip has not heated but hey it's still a basic experiment. 

I have a lot of STM32F103 and STM8S106 cards but only one arduino uno and one mega available.

Thank you for all your sharing, i wish you peace, success and happiness. 

Thank you.

Col68, 2020-04-04:
Adamculleeeeeeeennnnnnnnnnnn :)  

Thank you from the bottom of my heart, it works very very well, it's incredible the settings step by step, you are a Genius, it only remains for me to integrate the project in a box and add the tube and the mask for continue the experience in practice to understand the proper functioning, it's incredible i respect you, i will share the result of the final assembly. All the happiness for you.

Long live ADAMCULLEN.

adamcullen, 2020-04-04:
> Adamculleeeeeeeennnnnnnnnnnn :)
> 
> Thank you from the bottom of my heart, it works very very well, it's incredible the settings step by step

No problem. Can I ask what country you are in?

Col68, 2020-04-05:
I am of Turkish origin and I live in France, thank you again Adamcullen, I love you :)

adamcullen, 2020-04-05:
I've seen how bad this is in France, Spain, Italy and the UK. Hopefully all our work can help save lives. Good luck.

Col68, 2020-04-05:
Thank you dear friend, it is difficult because the government did not do its job, the hospitals were on strike before the appearance of this virus to denounce the lack of means but the government to shut the ears, then during the epidemic for lack of protective mask, the government to shout that it was useless to wear masks ... 

Then the patients doubled, tripled. Especially for Italy and Spain, we pray that it stops, man has no strength in the face of this tiny virus, long life to the people of the world.

Col68, 2020-04-05:
Dear Adamcullen, are you from Australia?

adamcullen, 2020-04-05:
Yes.

⁣Get BlueMail for Android ​

Col68, 2020-04-05:
In Australia there was a Turkish friend who makes homemade cheese, he shares other useful experience, I had realized a lot of these recipe :) 

[http://artizanpeynirci.blogspot.com/](url) ;)

Peace, success and happiness to the people of Australia.

Col68, 2020-04-06:
Hello friends 

I combine with the code to succeed but it is very difficult, the address of the LCD i2c is 0x27 after the scan but it does not display anything :( 

I am sorry to be null, if a friend could oriented? 

Thank you very much.

[image]

`#include <Wire.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16,2);

int breath = 1;
int i = 0;
int inP=1028; int inT=1028; int inP_v=1028; int inT_v=1028;
int outP=1028; int outT=1028; int outP_v=1028; int outT_v=1028;

void setup() 

{
  Serial.begin(9600);
  
  pinMode(9, OUTPUT);
  
  lcd.begin(16,2);
  
  lcd.setBacklight(HIGH);
  
  lcd.setCursor(0,0);
   
  lcd.print("Adamcullen");
  
  lcd.setCursor(0,1);  
  lcd.print("-Respirator-V1");
  
  delay(5000); 
  
  }

void loop() {
  // effacer l affichage et configurer les valeurs initiales

 // le prochain souffle est-il une inspiration ou une expiration? 
  if (breath !=0) {breath = 0;}
    else breath = 1;
    Serial.println(breath);

  // appliquer le code d'inspiration ou d'expiration
  switch (breath) {
    case 0: // Inspirer
    i = 0;
    Serial.println(i); Serial.println(inT_v);
      while (i <= inT_v) {
        // lire les potentiometres et determiner la pression respiratoire et le temps
        inP = analogRead(A0);
        inT = analogRead(A3);
        inP_v = ceil(inP/100)*25.5;
        inT_v = round(inT/100);

        Serial.println("Breath in");

        // utiliser pour afficher les valeurs sur l'ecran OLED
   

        // nous allons produire la tension PWM et retarder seulement 400 ms ici et boucler sur un maximum de 10 periodes 
        // cela nous permet de relire la pression et le temps toutes les 400 ms à la fin de chaque cycle de respiration
        analogWrite(9, inP_v); 
        delay(400); 
        i++;        
      }
      break;

    case 1: // expirer
    i = 0;
      while (i <= outT_v) {
        // lire les potentiometres et determiner la pression respiratoire et le temps
        outP = analogRead(A1);
        outT = analogRead(A2);
        outP_v = ceil(outP/100)*25.5;
        outT_v = round(outT/100);

        Serial.println("Breath out");
        analogWrite(9, outP_v);
        delay(400);
        i++;
      }
      break;
  }
  
}// boucle de fin
 `

zero48, 2020-04-06:
> [image]

Very nice setup adamcullen . 
I think the confusion comes up from your original Fritzing SETUP PICTURE ,  which has IN1 & IN2 ARE tied to High and IN3 & IN4 to Low. which would make the motors to stop.

https://user-images.githubusercontent.com/23371950/77723103-9bf70b80-7043-11ea-8c95-ec787002c5e1.JPG

In this latest wiring is the correct way to do parallel wiring in my opinion (is the one I use normally). 

Kudos !!

Col68, 2020-04-06:
Hi friends

Small update,  i go step by step, now i managed to have 2 display lines :)
The first line "Adamcullen" then the second "Respirator V1.0" nothing else. 

The Void Setup / Void Loop section must contain errors, i will continue my research and continue to modify.

` 
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16,2); 

int breath = 1;
int i = 0;
int inP=1028; int inT=1028; int inP_v=1028; int inT_v=1028;
int outP=1028; int outT=1028; int outP_v=1028; int outT_v=1028;

void setup() 

{ 
  lcd.init();
  
  lcd.backlight(); 
    
  Serial.begin(9600);
  
  pinMode(9, OUTPUT);
   
  lcd.print("Adamcullen");
  
  lcd.setCursor(0,1);  
  
  lcd.print("-Respirator-V1");
  
  delay(5000); 
  
  }

void loop() {
  // effacer l affichage et configurer les valeurs initiales

 // le prochain souffle est-il une inspiration ou une expiration? 
  if (breath !=0) {breath = 0;}
    else breath = 1;
    Serial.println(breath);

  // appliquer le code d'inspiration ou d'expiration
  switch (breath) {
    case 0: // Inspirer
    i = 0;
    Serial.println(i); Serial.println(inT_v);
      while (i <= inT_v) {
        // lire les potentiometres et determiner la pression respiratoire et le temps
        inP = analogRead(A0);
        inT = analogRead(A3);
        inP_v = ceil(inP/100)*25.5;
        inT_v = round(inT/100);

        Serial.println("Breath in");

        // utiliser pour afficher les valeurs sur l'ecran OLED
   

        // nous allons produire la tension PWM et retarder seulement 400 ms ici et boucler sur un maximum de 10 periodes 
        // cela nous permet de relire la pression et le temps toutes les 400 ms à la fin de chaque cycle de respiration
        analogWrite(9, inP_v); 
        delay(400); 
        i++;        
      }
      break;

    case 1: // expirer
    i = 0;
      while (i <= outT_v) {
        // lire les potentiometres et determiner la pression respiratoire et le temps
        outP = analogRead(A1);
        outT = analogRead(A2);
        outP_v = ceil(outP/100)*25.5;
        outT_v = round(outT/100);

        Serial.println("Breath out");
        analogWrite(9, outP_v);
        delay(400);
        i++;
      }
      break;
  }
  
}// boucle de fin
 
 `
Thank

adamcullen, 2020-04-06:
I don't think it matters as these pins are the forward/reverse pins. You can have either of them high or low.  The spec sheet says though, that to do it in parallel 1&4 must be tied and 2&3 must also. But you can reverse the polarity, the motor would just spin the other way.

⁣Get BlueMail for Android ​

zero48, 2020-04-06:
I'm Integrating the ATmega328P-AU with the L298 altogether in a PCB (including Oled). I left a FTDI connector in case I want to communicate with Atmega328P-AU, also I exposed the remaining GPIO to external world.
Also I integrated a solder wire system in case I want to control the motor (FWD/RVSE) from software. Took in consideration the current handling of the PCB . So far this is what I have , without the PCB Zones yet.
[image]

Col68, 2020-04-07:
Hi zero48

You are very gifted, bravo for the professionalism, this project is really cool.
Thank you for your sharing. 

Respect to King Adamcullen for the gift of his project. 

Thank you.

[image]

Col68, 2020-04-07:
Hi friends

Mission accomplished, here is the final result of the Adamcullen project, Respect to Adamcullen for sharing, the code is compatible for 1602 I2C LCD screen with the address 0x27. 

I was helped by a friend on Arduino France for the correct code, Thank you and Respect to the friend Lesept. It works perfectly, i still have found and adapted a mask and continue the project to add a nebulizer using a mini humidifier (banana) integrate in a small box with water heating, it seems difficult but not impossible, in all thank you very much to everyone. 

Long live sharing and respect for everyone, health, peace, success and happiness for all people, special thanks to Adamcullen the king of the project "Respirator V1" and to "Lesept" for the help with modification of the code to make it compatible with a 1602 I2C screen and without forgetting. 

Thanks and respect also to Johnny Lee for sharing in the service of all peoples. 

Here is the compatible code 1602 I2C and some photos ... See you soon.

`#include <Wire.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16, 2);

int breath = 1;
int i = 0;
int inP = 1028; int inT = 1028; int inP_v = 1028; int inT_v = 1028;
int outP = 1028; int outT = 1028; int outP_v = 1028; int outT_v = 1028;

void setup()

{
  lcd.init();
  lcd.backlight();
  Serial.begin(9600);
  pinMode(9, OUTPUT);
  lcd.print("Adamcullen");
  lcd.setCursor(0, 1);
  lcd.print("-Respirator-V1");
  delay(5000);
}

void loop() {
  // effacer l affichage et configurer les valeurs initiales

  // le prochain souffle est-il une inspiration ou une expiration?
  breath = 1 - breath;

  lcd.clear();
  lcd.setCursor(0,1);
  lcd.print(breath);
  Serial.println(breath);

  // appliquer le code d'inspiration ou d'expiration
  switch (breath) {
    case 0: // Inspirer
      i = 0;
      Serial.println(i); Serial.println(inT_v);
      while (i <= inT_v) {
        // lire les potentiometres et determiner la pression respiratoire et le temps
        inP = analogRead(A0);
        inT = analogRead(A3);
        inP_v = ceil(inP / 100) * 25.5;
        inT_v = round(inT / 100);

        Serial.println("Breath in");
        lcd.clear();
        lcd.setCursor (0,1);
        lcd.print("Breath in");

        // nous allons produire la tension PWM et retarder seulement 400 ms ici et boucler sur un maximum de 10 periodes
        // cela nous permet de relire la pression et le temps toutes les 400 ms à la fin de chaque cycle de respiration
        analogWrite(9, inP_v);
        delay(400);
        i++;
      }
      break;

    case 1: // expirer
      i = 0;
      while (i <= outT_v) {
        // lire les potentiometres et determiner la pression respiratoire et le temps
        outP = analogRead(A1);
        outT = analogRead(A2);
        outP_v = ceil(outP / 100) * 25.5;
        outT_v = round(outT / 100);

        Serial.println("Breath out");
        lcd.clear();
        lcd.setCursor (0,1);
        lcd.print("Breath out");
        analogWrite(9, outP_v);
        delay(400);
        i++;
      }
      break;
  }

}// boucle de fin
`
[image]

[image]

[image]

Thank you .

zero48, 2020-04-07:
Well , My PCB is ready to be ordered. 
I think it is more practical Than the ones using ESC.

Thanks Adamcullen , your post Inspired it.

[image]
[image]

brian-mckeon, 2020-04-11:
Adam, very interesting and can't get much simpler than this!

I'd been looking at various blowers etc and thought I'd let people know to be careful that some blowers put the motor in the air path for cooling.  Lots of cheaper devices use brushed motors which can spark a bit under load and this could result in ozone in the output to the patient which is bad.    Fortunately the mattress pump seems to be good in that it seems to bleed some of the output air off and run this through the motor - so it doesn't go to the facemask.  But people should check this on whatever blower they choose.   Or choose a brushless motor.   Brushless would also definitely be preferred in environments where oxygen might be being administered.    Lots of risks in these low-cost quick designs but, as you say, if things are going crazy and no place for you at the hospital then the balance of risk changes.

Here are some numbers that I found with the inflator I bought that looks very similar to yours.  It seems to be tuned for lower pressure and higher airflow in that it draws around 3.5A with not much resistance on the output.  This is probably where it does most of it's work if inflating a mattress.  This drops to about 2.5A when there is a facemask and a person on the other end, higher pressure and lower airflow.   I had attached a water U-tube manometer I made from some PVC tube attached to a piece of wood and I was seeing a bit more than 20cm pressure at 2.5A before I backed off so the pump seems quite capable.  20cm feels like a lot however it seems that this or even a bit higher  can be necessary for people in trouble.
[image]

The pump instructions say to give the pump a rest for 10mins for each 20mins of operation but the 2.5A compared to 3.5A peak and the intermittent operation in a respiratory cycle cycle which should allow continuous operation, consistent with your finding it was OK after 24hrs.

Small motors as in this inflator seem to be expected to last in the region of 1000hrs or so with the brushes being the weakpoint however 1000hrs is 40days which sounds plenty for the sort of situation you are considering.

chonghorizons, 2020-04-11:
@adamcullen Thank you for the extensive design. This is really good.

### Power draw question:
 I'm wondering if you've done a power draw and have a watts needed?  Looks like 30W-rated fan, and so maybe 70watts total?  If 70watts, that's 14amps at 5V, and that's going to be difficult to power (off standard USB pack that have a max of maybe 5A, of course it's not hard to power if you build your own battery pack for that spec.)

I know one can build their won 12V battery pack, but it may be easier to use widely available USB chargers.

chonghorizons, 2020-04-11:
You might be able to steal a few ideas from this recent design...

We are promoting a design from 4/6 of an improvised PAPR. Importantly:
1. Using commonly available hospital viral filters
2. Tested in a hospital setting -- so user story has been validated.
 MAIN HALO DESIGN 
https://www.youtube.com/watch?v=JMXP8jAmDKA&feature=youtu.be
https://3dprint.nih.gov/discover/3dpx-013544
https://bunnyscience.dozuki.com/Guide/Bunny+Science+HALO+Respirator+(Buildable+PAPR)/4

We are looking to iterate focusing on cost and to try to limit the 3d printed parts needed. Innovation to test: baseball caps, common irrigation fittings (PVC), and silicon repair tape coupling. Cardboard.
## THE MACGUYVER-COVID19 DISCORD CHANNEL
If you are interested in a channel where we have vetted people for (a) helpfulness and (b) teamworkiness and (c) stick around for more than 72hrs, check this discord server out.

I'm @Hocho on this discord server: https://discord.gg/H3jUmar (all welcome)
or chongare@gmail.com or 607-279-3623 (text/phone)
I'm working 12hr days, 7 days a week to organize and move the needle. Small team, but good team (vetted)

adamcullen, 2020-04-11:
My pump only has a 12v 2.5A power supply, so it can't be much more than that.

Adam
⁣

Pinponcito, 2020-04-12:
some one have the sketch for a basic pump air 12v dc, using in arduino nano v3.0, and a motro driver control L298N 
also coulbe have a  sketch  2 steps. 1 turn on the motor, and second or doble push, start a breath resperation.
thanks I'm amateur Lol.

adamcullen, 2020-04-12:
Ricardo, you should be able to use my sketch with an arduino nano.  Use the same wiring and the same sketch. You just need to select the nano from the boards manager.

Adam

Pinponcito, 2020-04-12:
Ok ok thanks My friend i will do it

3dseed, 2020-04-12:
Hi everyone :)
Is it possible to use this [USB desktop fan](https://www.amazon.es/Fancii-Ventilador-Escritorio-silencioso-circulaci%C3%B3n/dp/B072DSHKCH/ref=sr_1_6?__mk_es_ES=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=USB%2Bfan&qid=1586683037&sr=8-6&th=1) ?
Would it have enough suction to pull through a decent N95 style filter ?

adamcullen, 2020-04-12:
> Hi everyone :)
> Is it possible to use this [USB desktop [...]

I really don't think so. Its aperture is way too large and running off 5v USB is probably way underpowered.

Just buy one of these https://www.amazon.es/Appearanice-Electric-Inflate-Mattress-Inflator/dp/B083NV2SZK/

3dseed, 2020-04-12:
Hi @adamcullen , thanks for such a quick reply :)
I considered this 5V fan as it's easy to run off a small battery, as the 12V battery pack is a bit [pricey](https://www.amazon.com/gp/product/B07H8F5HYJ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) :)   But ok, I'll buy the 5V fan as well..  AND stick with what works the 12V blower .. thanks so much for this ..!

Pinponcito, 2020-04-12:
My friend the diagram that you use is for  BLDC driver and this for a BLDC motor 3 cables.
that setup in my case is using a L298, the conections are diferent.

do you have a schematic for my setup 12v pump air?

my email. is:
 ing.morenoricardo@gmail.com.  thanks my friend

qsullivan11, 2020-04-12:
BLDC is brushless 3 phase, L298N is brushed DC, diagram attached. The [full video is here](https://www.youtube.com/watch?v=dyjo_ggEtVU).
[image]

Pinponcito, 2020-04-12:
Yes my friend you are rigth, and thanks for excelent video, but the video show using arduino Uno, I have arduino Nano V3.0 and sketch arduino from @adamcullen project.

thanks for help to make my project cut work.

> BLDC is brushless 3 phase, L298N is brushed DC, diagram attached. The [full video is here](https://www.youtube.com/watch?v=dyjo_ggEtVU).
> [image]

Pinponcito, 2020-04-12:
no voltage in arduino nano 3.0 and L298N, using the diagram from adamcullen

any idea?

> Yes my friend you are rigth, and thanks for excelent video, but the video show using arduino Uno, I have arduino Nano V3.0 and sketch arduino from @adamcullen project.
>
> thanks for help to make my [...]

adamcullen, 2020-04-12:
> no voltage in arduino nano 3.0 and L298N, using the diagram from adamcullen
> 
> any idea?

My advice is to just start with your arduino and the L298N board. If you have a small 12v motor, just wire up one of the motor outs to it and see if you can just control one small motor. if you dont have a motor, a voltmeter or even a 12v light would help fault find.
What you are aiming for is to see if you can control the voltage from one of the motor outputs to make it different from the 12v in. You can run a code that just goes High Low High Low on the pin 9 output and see if you can see a change in voltage at the motor output terminals.
Once you can do that, you can then try bridging both sides of the L298N so that the voltage at motor 1 and motor 2 is the same and changes at the same time with your code.

Have a look at this video: https://www.youtube.com/watch?v=wjFW-TNq8Og and this how to: https://howtomechatronics.com/tutorials/arduino/arduino-dc-motor-control-tutorial-l298n-pwm-h-bridge/

Pinponcito, 2020-04-13:
Thanks my friend,
the setup and sketch mots have the 4 potentiometter? or can run with out them, I Mean not exist a sketch that can changue the speed (simulation breath ) with out use potentiometer 

thaking the exaple from sketch of Johnny Lee @jcl5m1 

https://github.com/jcl5m1/ventilator/blob/master/ventilator_control/ventilator_control.ino

thanks Adam

> My advice is to just start with your arduino and the L298N board. If you have a small 12v motor, just wire up one of the motor outs to it and see if you can just control one small motor. if you dont [...]

adamcullen, 2020-04-13:
> Thanks my friend,
> the setup and sketch mots have the 4 potentiometter? or can run with out them, I Mean not exist a sketch that can changue the speed (simulation breath ) with out use potentiometer [...]

Try this. I've just hard coded values where the analog read would be in the sketch to simulate values it would read from the variable pots. It should breath in for about 2 seconds at full power, then breath out for about 1 second at half power. This should get you started.

```
int breath = 1;

void setup() {
  Serial.begin(9600);
  pinMode(9, OUTPUT);
}

void loop() {
  int inP = 1028; // was: analogRead(A0); // pot 1 - in breath pressure
  int outP = 512; // was: analogRead(A1); // pot 2 - out breath pressure
  int inT = 512; // was: analogRead(A3); // pot 3 - in breath time
  int outT = 256; // was: analogRead(A2); // pot 4 - out breath time

  int inP_v = round(inP/100)*25.5;  // take the value from the pot, divide by 100 and round, then multiply by 25.5 so the output value is a multiple of 255
  int outP_v = round(outP/100)*25.5;  // take the value from the pot, divide by 100 and round, then multiply by 25.5 so the output value is a multiple of 255
  int inT_v = round(inT/100)*400;  // take the value from the pot, divide by 100 and round, then multiply by 400 to get a proportion of 4000ms
  int outT_v = round(outT/100)*400;  // take the value from the pot, divide by 100 and round, then multiply by 400 to get a proportion of 4000ms

  if (breath !=0) {breath = 0;}
    else breath = 1;

  switch (breath) {
    case 0:
      Serial.print("Breath In. Pressure: "); Serial.print(inP_v); 
      Serial.print(". In Time: "); Serial.print(inT_v); Serial.println("ms.");
      analogWrite(9, inP_v);
      delay(inT_v);
      break;

    case 1:
      Serial.print("Breath Out. Pressure: "); Serial.print(outP_v); 
      Serial.print(". Out Time: "); Serial.print(outT_v); Serial.println("ms.");
      analogWrite(9, outP_v);
      delay(outT_v);
      break;
  }
}
```

With wiring like this.

https://user-images.githubusercontent.com/23371950/78415437-69fe2e80-766d-11ea-85b1-486723096fc9.JPG

Pinponcito, 2020-04-13:
Excellent Adam I wil try.
in my case Im using a arduino nano V3.0  the wiring are sames pines?
thanks for time to help me on my proyect 

> > Thanks my friend,
> > the setup and sketch mots have the 4 potentiometter? or can run with out them, I Mean not exist a sketch that can changue the speed (simulation breath ) with out use potentiometer
> > thaking the exaple from sketch of Johnny Lee @jcl5m1
> > https://github.com/jcl5m1/ventilator/blob/master/ventilator_control/ventilator_control.ino
> 
> Try this. I've just hard coded values in my original sketch where the analog read would be to simulate values it would read from the variable pots. It should breath in for about 2 seconds at full power, then breath out for about 1 second at half power. This should get you started.
> 
> ```
> int breath = 1;
> 
> void setup() {
>   Serial.begin(9600);
>   pinMode(9, OUTPUT);
> }
> 
> void loop() {
>   int inP = 1028; // was: analogRead(A0); // pot 1 - in breath pressure
>   int outP = 512; // was: analogRead(A1); // pot 2 - out breath pressure
>   int inT = 512; // was: analogRead(A3); // pot 3 - in breath time
>   int outT = 256; // was: analogRead(A2); // pot 4 - out breath time
> 
>   int inP_v = round(inP/100)*25.5;  // take the value from the pot, divide by 100 and round, then multiply by 25.5 so the output value is a multiple of 255
>   int outP_v = round(outP/100)*25.5;  // take the value from the pot, divide by 100 and round, then multiply by 25.5 so the output value is a multiple of 255
>   int inT_v = round(inT/100)*400;  // take the value from the pot, divide by 100 and round, then multiply by 400 to get a proportion of 4000ms
>   int outT_v = round(outT/100)*400;  // take the value from the pot, divide by 100 and round, then multiply by 400 to get a proportion of 4000ms
> 
>   if (breath !=0) {breath = 0;}
>     else breath = 1;
> 
>   switch (breath) {
>     case 0:
>       Serial.print("Breath In. Pressure: "); Serial.print(inP_v); 
>       Serial.print(". In Time: "); Serial.print(inT_v); Serial.println("ms.");
>       analogWrite(9, inP_v);
>       delay(inT_v);
>       break;
> 
>     case 1:
>       Serial.print("Breath Out. Pressure: "); Serial.print(outP_v); 
>       Serial.print(". Out Time: "); Serial.print(outT_v); Serial.println("ms.");
>       analogWrite(9, outP_v);
>       delay(outT_v);
>       break;
>   }
> }
> ```
> 
> With wiring like this.
> 
> https://user-images.githubusercontent.com/23371950/78415437-69fe2e80-766d-11ea-85b1-486723096fc9.JPG

adamcullen, 2020-04-13:
> Excellent Adam I wil try.
> in my case Im using a arduino nano V3.0 the wiring are sames pines?
> thanks for time to help me on my proyect

Ricardo, you should be able to use my sketch with an arduino nano.  Use the same wiring and the same sketch. You just need to select the nano from the boards manager.

timhurdle, 2020-04-18:
Hi All, I have been working on a similar device in the UK. I tried to engage with the One Million Ventilators Group with not much response. I am still working on it anyway but been considering that carbon and copper particles from the motor can contaminate the air as there is deliberate low pressure pull through the motor core through the backplate of the impeller rotor (for cooling). Also I heard that brushless motors are needed as 03 (ozone) can be generated by sparks from brush commutation. I think the voltage are too low for that but I have come up with a possible solution if anyone is interested.

brian-mckeon, 2020-04-19:
Hi Tim,
have a look back at my post on Apr 11.  I was also concerned about ozone.  However I found that the motor cooling airflow in this mattress inflator seemed to be bled off the output pressure side and then out the vents below the motor.  A bit different to vacuum cleaners etc but makes sense as the small motor ducts could often be facing a dirt campground and you wouldn't want to be drawing cooling air in from there.  I used deflection of a small piece of paper to try and determine airflow and it seemed to be exiting these vents.  Let me know if you find different.   If you want to go brushless you could simply buy a 12V CPAP motor and pump from DHgate or others, https://www.dhgate.com/product/daniu-wm7040-dc-12v-24v-high-pressure-blower/505193494.html?dspm=pcen.sp.list.1.E97VPnotXnnR3Wuw0dtf&resource_id=#s1-0-1;searl|2412079574:1.  They are not that expensive although shipping times would be pretty poor at the moment.

Pinponcito, 2020-04-19:
Hi Tim, Im from Mexico.
with the sipporth of this guys I make my own sketch and make a ventilator made in home with a air pump rigth now are in 24hr work test, and next day we makeing test in pressure PEEP.
Sent to me a email and i cant share my projetc, i will post here in next days but for more quick anser and you can get a ventilator contact to me.
bes regard

> Hi All, I have been working on a similar device in the UK. I tried to engage with the One Million Ventilators Group with not much response. I am still working on it anyway but been considering that [...]

[20 more replies]