jcl5m1/ventilator #63: Medical issues surrounding using repurposed CPAP gear and parts

bradtem, 2020-03-24:
I've been researching ventilators and talking to doctors to learn whether it's possible to just reflash existing machines to do ventilation.   I have the information in a [blog post found here](https://ideas.4brad.com/creating-plan-reprogram-smart-cpap-machines-become-emergency-ventilators).

However, I will also reproduce the current article below for convenience but check the article for the lastest

As I'm sure, you've heard about the need that many Covid-19 patients have for ventilators which keep them breathing when their lungs fail, as they do during the "ARDS" (acute respiratory distress syndrome) phase of Covid-19 which is the thing that kills you.  The problem is that there are at most around 200,000 ventilators (including tapping older models sitting in storage, a government strategic reserve and a military reserve.)  It is feared that as many as 900,000 could be needed if the worst projections are true.  Around the world, many more.

Companies that make ventilators are ramping up manufacturing as fast as they can.   It still won't be enough.  That's even true if one applies a special technique devised years ago to put 2, 4 and even 9 patients on the same machine, if the patients can be matched so they need the same pressures, airflow and oxygen.   This technique was tested under fire during the Las Vegas massacre and saved hundreds of lives, but it's still only barely tested for something like this -- ARDS patients may need to be on the ventilator for up to a month, if they live.

One tragedy for the ventilator makers -- if they are able to produce hundreds of thousands of their FDA approved modern ventilators, they will save many lives -- but then no hospital, anywhere in the world, will need to buy a new ventilator for years, because there will be such a huge surplus after this is over.  Stepping up means a big sales windfall, but then destroying their business, unless some special government intervention comes in.   There is much debate if manufacturing at this volume is even possible.

Several interesting projects have sprung up to create designs for cheap ventilators, [included from CPAP parts](https://github.com/jcl5m1/ventilator) and there's [even a contest with a prize](https://www.agorize.com/en/challenges/code-life-challenge). What's even more important than cheap is that they be reliable and that it is possible to make vast numbers of reliable ones quickly.    These range from hacker projects to fully manual ventilators which literally require an attending person to squeeze an air bladder to breathe for the patient.   They must do this 24 hours a day, and will be exposing themselves to the virus when they do so.  If they let up, if they make a serious mistake, their patient might die.  It's a last resort.

In any event, the cost is very worth it, so we don't need the machines to be cheap, but they must be simple enough to be made fast, from off the shelf or easily mass-manufactured in factories which can be brought up to full production without putting their workers at major risk.   Not easy.

## Smart CPAP hardware with new firmware = Ventilator?

One option is being explored by several people.  It is not yet verified to be practical, but if it is, it has many attractive elements to it.   Around 20 million people in the USA have a condition known as obstructive sleep apnea, which means their airway can close up when they sleep, temporarily shutting off air and waking them up.   Millions of them treat this with what is called a CPAP machine -- which blows air into their nose via a mask at higher pressure, and keeps the airway inflated.   Over the years, these machines have become much more sophisticated, and several million of them are made each year.

The current machines can blow air with a computer controlled blower that can change the level of pressure instantly.  They have sensors to detect the air pressure, sound and air flow. The older machines did not do this, they just stuck one one basic pressure, but the newer ones tend to have this ability.

While CPAP machines are not ventilators, and don't have software to deliver assist control or pressure control ventilation, they may contain much of the *hardware* needed for certain types of ventilation.   What they don't have is the software.

Normal CPAP machines deliver a max pressure of 20cm of water.  (That's not much, about 2% of atmospheric pressure.)  Many ventilation patients use much less than that.   Ventilators usually are rated to deliver over 40cm, though they don't normally go to that level, as it's dangerous.  

We [have discovered](https://youtu.be/n57u1NvXBgw), though, that the blowers used in some CPAP models are capable of 45cm and up to 400 l/min of airflow, much more than is needed for CPAP, and enough for ventilating many ARDS patients.  Less established is whether these blowers and their electronics might fail if used 24 hours a day at these levels.  My intuition is that most medical devices are built with wide tolerances and this will not be the case, but it must be confirmed.   If 24 hour operation is not practical, there is a solution -- patients can have two or more machines, and have their machine switched every 8-10 hours, which is the normal duty cycle of CPAP machines.  Because there are so many millions of CPAP machines out there, this remains an option.

Some patients can be helped with the CPAP or BiPAP therapy these machines deliver with their current software, and that's good.  However, under discussion is **not** using CPAP/BiPAP to treat patients, but rather converting such machines, by replacing their internal software, to apply ventilation.

In fact, if two machines are available, they can both be hooked up to the patient's air tube, through a valve that allows either one (or both) to blow air but which prevents blowback into the unit.   This allows automatic handoff between two machines, and also means that **if failure is detected in one machine, it can be turned off and the other activated to take up the slack, also making an alert to replace the bad machine in reasonable time**.  This could result in an extremely high level of reliability.

In particular Covid ARDS patients are being ventilated at a fairly high pressure.  In many cases, the minimum pressure (PEEP) is set at 16cm, which is high for a CPAP.  The upper pressure limit varies -- it is not set, but is whatever it takes, up to a limit, to assure sufficient air flow through the lungs.

To act as ventilators, the machines would be converted by loading new software in their controllers.  Fortunately, most of them can be field upgraded with new firmware.   Many of the leading CPAP vendors, such as Medtronic/Puritan Bennett and Phillips Respironics, also make ventilators and are already familiar with the algorithms and software needed. 

Some of these machines feature return tubes which can be used to measure pressure at the interface instead of in the machine.  This is superior, but was largely eliminated from newer machines as no longer necessary.  It is not known if this is needed.

## Monitoring and control

Most of these machines also have a USB port, which can be used to set up the machine and to upgrade its firmware.   New ventilator firmware could also use this port to communicate with a control and monitoring computer, which could either be based on a standard mobile phone or an old laptop running a free operating system such as Linux from a USB stick.   This monitoring computer would not be life-critical.  It would be used to:

1. Provide a GUI to set up the ventilator and all its parameters
2. Receive data from the ventilator to display what's going on, including detecting any problems.   Display a graph of pressure, volume and flow over time.
3. Sending this data and any alerts to central nursing stations (more than one in case of failure) and anybody else that needs to know.   In addition, they would send, and relay from the ventilator a "heartbeat" regular signal so that the failure of either device would be noticed at the nursing stations, causing somebody to be dispatched.

While the monitors would communicate status over Ethernet or WiFi, they probably would not receive commands back that way, or if so, only over a non-internet verified channel, to avoid risks of computer intrusion changing parameters of the machines.   Of course that means problems require dispatching of a qualified staffer which has its own costs.

[... cut ...]
## Key Questions

* Can 95% oxygen, or even 50%, be safely inserted into the air flow?
* Can the output vent be safely filtered to avoid spreading viruses?
* Can CPAP blowers and electronics operate at 20-30cm pressure on a 24 hour basis for days and weeks at a time with very low failure?
* Can CPAP vendors be convinced to release new firmware, and when?
* Can independent parties write new firmware and reflash these units?
* Can enough air:oxygen blenders and patient interfaces and more be supplied?

DaveM135, 2020-03-24:
Outstanding summary!!

Best

Dave

mikehearn, 2020-03-25:
Sounds like we have done very similar analysis, and reached very similar conclusions:

https://blog.plan99.net/cpap-for-covid-d47886bf978c

DaveM135, 2020-03-25:
Thanks for sharing.  Excellent summary.   I am discussing locally with my practicing critical care colleagues and mechanical ventilation experts.   I’m also not sure why this isn’t being prioritized.   Making these machines should be MUCH simpler than making full ventilators.   I am currently focused on the mask issue as a full face mask would be more efficient, I think, as long as the patient is fully conscious and cooperative.    The current nasal masks for CPAP and BiPAP are designed to not cover the full face which is a limitation in the ICU.   The nose and mouth masks require an uncomfortably tight seal for longer term use.   I think the adaptation of the full face snorkeling mask looks like a realistic option but need to explore a bit more.  

Best

Dave

deepaksethu, 2020-03-25:
Can we modify this project to work like this https://www.helmetbasedventilation.com/post/diy-a-helmet-ventilator-can-be-used-with-wall-oxygen-alone-to-help-covid-19-patients?fbclid=IwAR1Rcl-NlfT5YiA1fkeWrdMIQZdVNTf8wqLAcndSHmOZ3ErMsKRvKVlbeLA

bradtem, 2020-03-25:
Deepak, what is your source for there being only  330K people on CPAP in the UK?    In the USA it is estimated there are 22M people with OSA, though of course many are not prescribed CPAP.   However, the CPAP industry is about $4.5B per year globally, which suggests 2-3 million machines per year made though I have not found a firm number.   Further, less than half of patients who get prescribed a CPAP keep it up, so there are millions of machines sitting on shelves.

Also the use of non-invasive masks is already underway but doesn't do it for the patients who need full ventilators.   The oxygen ports you showed are small ports on the mask or line, and it is unclear how you could introduce enough O2 to get 95% if most of the air is coming from the blower.   However, adding O2 on the line or mask does not have fire or other such risks.

But it's not going to be CPAP masks.  I think that's fine.  I suspect if they need to manufacture a million ventilator interfaces, they can do that.  That's just plastic parts and silicone.   It's a million ventilators that's the problem, getting them manufactured to bet your life reliability.  With the CPAPs, I propose having two of them connected in a Y through one way valves, and software which monitors them and switches one off and the other on in case of failure.

bradtem, 2020-03-25:
One thing worth investigating if we can't risk putting high O2 concentration through this gear.  What if the oxygen is coming it at the higher pressure (25cm or more) but flow limited so that it can't inflate the lung on its own, but then is assisted by the CPAP putting out 16-26cm as needed at higher flow.   Would that work?  What's the maximum FiO2 you can get in the resulting mix with 21% O2 in the air and 100% O2 in the feed?   I never took fluid dynamics so this is outside my knowledge.

mikehearn, 2020-03-25:
The 330k figure comes from here:

https://www.blf.org.uk/support-for-you/obstructive-sleep-apnoea-osa/health-care-professionals/health-economics-report

> However, fewer than half of these (around 330,000) are currently receiving the treatment they need

bradtem, 2020-03-25:
My understanding is that helmet masks can have issues with CO2 buildup inside the helmet.   A conscious person could have a mouthpiece they breathe out while breathing in through the nose perhaps.   Anyway if you exhale into a full helmet, your CO2 diffuses in the whole volume and then your exhalation output removes less CO2 per minute unless its flow is much higher.

nasocializes, 2020-03-30:
@deepaksethu yes.

jonschull, 2020-04-02:
Here's a possible approach.  I'd very much appreciate feedback or collaboration.

infection-proofed BIPAP "preventilator".

[image]

Project documentation  [here](https://docs.google.com/document/d/1KWr_RPS04HTDZRTC6QBDoaGjEfw-uBM2XmOOpwxCsnU/edit#).

video here  https://youtu.be/SbYDuheeXZc

bradtem, 2020-04-02:
Johnschull, so the purpose of this is to just capture the exhaust?   Seems like a great deal of effort compared to making a mask which exhausts out a port which can be fed into a filter.   Making millions of plastic parts is not a big deal.  Making millions of complex machines is.  Also, you are feeding the exhaust back into the bipap, which obviously you can't do because it's full of CO2, unless you have scrubbers.

jonschull, 2020-04-02:
Thanks

The purpose is to make existing bipaps deployable  by infection-proofing
them.  Without infection-proofing, the standard design is unusable with
covid.

This IS just two ports.  One is just what you said.  The other collects
leakage arising from imperfect mask seal.  The cowl is to capture and
contain the leakage proper to exhausting.

I think you're right about exhalations.  (I was diluting on each cycle, but
a separate exhaust vacuum/filter is simpler and would put less load on the
bipap).

Make sense?

I don't understand your comment about making millions of complex machines.
There ARE millions of bipaps already extant.  I want to make them
deployable.

bradtem, 2020-04-02:
Yes, this thread is about reprogramming existing cpap/bipap machines. You want to minimize additional hardware.  However, it's reprogramming them to be ventilators, not bipaps/cpaps.  As such they would be using the intubation interface that does not have the exhaust vents of cpap masks.      I would also hope if what you propose is needed, that they could manufacture masks designed to have their exhaust filtered.   While I don't know the reason that this is not done in the past, I suspect there are some complex issues.  In particular, adding suction would alter the flow into and out of the patient and would have to be done with care.    While many CPAP users complain about the exhaust vents blowing on them or their partner, they never make a CPAP mask that vents the exhaust any other way but a hole on the mask, and I am curious why.