jcl5m1/ventilator #64: Android APP to control ventilator via Bluetooth

afraniofilho, 2020-03-24:
Hi, I hope you are well... I'm from Brazil and we are try to catch up to be prepared to face the pandemic.
We reproduced your ventilator project and in the last few days I worked to develop an Android APP to control the ventilator by an ESP32 Bluetooth connection.
[image]
Sorry about the Portuguese language but I want to share with you to contribute with you effort.
If you are interested I can share the code with you (C++ adapted from your, and the Android Studio Project).
Regards,

timafh, 2020-03-25:
I think that's a good approach.
But as an engineer in the field of medical technology, I would prefer an integrated solution.

Even for this type of rather "unprofessional" equipment, we should not completely ignore things like safety and electromagnetic compatibility.

afraniofilho, 2020-03-25:
Thanks for the feedback, Actually, the system have both approach... Mechanical switch and Bluetooth connection as a plus. Regards

ameeuw, 2020-03-25:
For maximum portability a simple control would be best. I am thinking of a three button interface and a simple SSD1306 OLED display to check and change values.

timafh, 2020-03-25:
@ameeuw Yeah, I also think that should be the way to go.

afraniofilho, 2020-03-25:
I'm really appreciate your feedback. I´ll try the OLED and switch...