jcl5m1/ventilator #67: mind map

arezoomehraji, 2020-03-25:
hi, guys,
I hope you are well.  
During the study of this project, I found that we needed a mind map & detailed planning and timetable. as we know, we have many issues and solutions and we have to create a plan for everyone's task because without a conclusion we can not reach the best product. I was very confused about the process and it would be better to organize our tasks on another platform such as Slack or Trello.

nasocializes, 2020-03-25:
If you'd like to provide project management and create links for it, feel free to take lead~

arezoomehraji, 2020-03-25:
@nasocializes I left this comment to know what is your and other volunteers' opinion? what do you think about having full management on this project and sharing tasks? we have good knowledge and need to lead for the perfect result

tlalexander, 2020-03-26:
@arezoomehraji  You may want to ask on this slack channel.
https://join.slack.com/t/open-ventilator/shared_invite/zt-cwh6py1x-C5tO4JewMUf2tRN7rwWB~g

arezoomehraji, 2020-03-26:
@tlalexander is there a slack group? I thought, there is no group on other platforms

tlalexander, 2020-03-26:
Well, that slack group I think is different people, but working towards similar goals. They seemed to be holding meetings etc.

T-Py-T, 2020-03-27:
You can create a Project on this repository to track not only these Open items, but also explicit tasks to be completed. There is even an automated template that closes stale Issues. (Projects Tab). Assign Contributors, etc.

nasocializes, 2020-03-30:
Check it out: https://github.com/jcl5m1/ventilator/projects