jcl5m1/ventilator #68: sorting knowledge and expertise

arezoomehraji, 2020-03-25:
we need to create two categories for the leading project:
at first, I want to ask from all volunteers:
1. name, email,  expertise 
( for add and share on other platforms )

second, we will make an international timetable for whole people around the world, and group for filling the field that we need to have.

DaveM135, 2020-03-25:
Hi 

David Morris  
david.g.morris@sbcglobal.net <mailto:david.g.morris@sbcglobal.net>
Expertise: Pulmonary and intensive care physician; venture capital; drug development
Location: San Francisco, CA

arezoomehraji, 2020-03-25:
Hi @DaveM135 
thank you for sharing your information, we will share the links and plan as soon as others participate and complete whole volunteer data.

greentheo, 2020-03-25:
Did you get any links set up for this?  I'm happy to add contact details to a private document list.

arezoomehraji, 2020-03-25:
hi @greentheo 
I am going to set up but I need each volunteer's data especially for sorting documents and tasks.

Fearfulbot, 2020-03-26:
You guys need to set up a discord. I'm a prototype from Bothell WA with CNC capabilities.

Fearfulbot, 2020-03-26:
Or at least join the directory in the slack groups of open-ventilator

goho50, 2020-03-27:
Georgi Bankov
bankov@gmail.com
General practitioner, Specialist Occupational health
Bulgaria

I am coming from https://diyventilators.discourse.group/. 
I studied yesterday all prototypes here : https://docs.google.com/spreadsheets/d/e/2PACX-1vTYAfldxoIiO46VAWH1NlhrwFBn9mguqS2bh1spnLEu4AVVN1cj1vaEm6vOp5Z6UnaAbUwd8dslCXdM/pubhtml
For me your approach is the right choice and the best compromise between all requirements towards the situation
 I want to make working prototype in order to adjust the software to the different ventilation modes. I will be the "patient".

arezoomehraji, 2020-03-27:
hi @goho50 
I am glad to hear and it would be really helpful to consult with you. I add your information and organize a plan as soon as others respond or more partnerships.

arezoomehraji, 2020-03-27:
hi @Fearfulbot 
thank you for sharing your expertise, would you send your email and your location. we need to arrange all plans and conversations and capabilities with your time. one of the problems that I have been engaged it was time difference and we should set them on a suitable time.

vsbhardwaj99, 2020-04-05:
Hi,
Varun Singh <v.s.bhardwaj99@gmail.com>
Iam not much of an expert but a student of software engineering and would like to contribute my share here. 
Place- India

MikeJaworski, 2020-04-05:
Hi,

Mike Jaworski

(personal email is firstname@lastname.me.uk -> hopefully that makes sense, don't want spambots)

I am an applications / tech support / certification specialist for a diaphragm pump manufacturer.  I design (or help our customers design) complete fluid handling systems built around our pumps as well as being responsible for all EU regulatory compliance.

Experience in:

* Pressure safety (EU Pressure Equipment Directive) certification
* Explosion safety (ATEX / IECEx) for mechanical (not electrical) equipment - I am the ATEX / IECEx certification lead on our complete product line, working to the latest EN 80079 series (trust me, mechanical IECEx / EN 80079 experience is a rare thing).
* Flow metering
* Failure modes and effects analysis
* Relief valve sizing, with some limited valve design knowledge
* Function safety (SIL rating) of industrial fluid handling / monitoring systems
* Industrial automation (PLC programming)

Also have some hobbyist electronics / Arduiono experience although non-recent.  Have an FDM 3D printer arriving in the next few days.

Here's my Linkedin although it's not very up to date:

https://www.linkedin.com/in/michael-jaworski-912b3614b/

Please let me make it clear that I do not represent my employer here and anything I say and do here is not sanctioned by my employer - I am here purely in my capacity as a private individual.