jcl5m1/ventilator #69: Use Decathlon or similar diving mask as C-PAP mask

PaulStefaan, 2020-03-25:
Hi, 

In Europ, we have a project to mod a diving mask to a C-PAP mask 

See link 
https://www.isinnova.it/easy-covid19-eng/

Video
[https://youtu.be/w4Csqdxkrfw](url)

By https://www.isinnova.it/

DaveM135, 2020-03-25:
Very interested in this.   I’m in the SF Bay Area.   I am a pulmonologist.   Can you put me in touch with anyone who is using it in Italy?

Best

Dave

PaulStefaan, 2020-03-25:
> Very interested in this. I’m in the SF Bay Area. I am a pulmonologist. Can you put me in touch with anyone who is using it in Italy? Best Dave
> […](#)

Hi Dave,

Nice to meet you ( I'm located in  Amsterdam)  

Here contact info 
 "the Chiari Hospital"
Dr Fracassi and Mr Romaioli

Article: 
https://www.foxnews.com/world/coronavirus-italy-ventilator-valve-supply-business

closed 2020-03-30 by nasocializes