jcl5m1/ventilator #70: How quickly can the BLDC motor you pulled respond?

mikehearn, 2020-03-25:
From reading around it seems:

* Primarily it's bi-level machines that are interesting as quick ad-hoc ventilator replacements because they can generate much larger pressure differentials, making it easier to tolerate higher pressures.
* Bi-level machines *can* change their pressures by altering motor speeds, but often don't to reduce power and noise levels. Instead they throttle the air intake using a moveable vane/baffle.

Bi-level machines are rarer than CPAP machines. That raises the question of whether a CPAP machine can be quickly converted into a bi-level machine without the additional intake port throttle, by controlling motor speeds. As noise doesn't matter it seems mostly an electrical question. Have you studied the electronics inside the machine you disassembled to see what it would be theoretically capable of? The thinking here being that it's faster to reflash machines than make physical changes to them.

zobotek, 2020-03-26:
Interesting question.  I've started experimenting with mid-sized RC (e.g., quadcopter/airplane) motor controllers and motors in part due to their motor and controllers' closed-loop open-source programming (allowing PID tuning) and ability to deliver high controlled torque capable of rapidly and precisely delivering needed power changes, overcoming the significant angular momentum of props.  I don't know much about CPAP design, but it's not apparent to me why their motor speed controls would require rapid speed adjustments (in within 100ms).  Further, closed source motor control code may make appropriate PID tuning difficult.  

One paper with some data (and abundant dynamic analysis) may be found in International Journal of Mechanical Engineering and Robotics Research Vol. 4, No. 4, October 2015 "Design and Control of Small Quadcopter System with Motor Closed Loop Speed Control"
H. L. Chan and K. T. Woo.  It's available on line - Google for it.  It shows (in Tables I and II) precise and rapid changes in motor velocity that can be achieved with closed-loop control that can be tuned by access.

mikehearn, 2020-03-26:
I guess it's because breathing can turn around from inhalation to exhalation very quickly, so the time between user breathes out + sensor latency + controller latency + motor latency needs to be low enough that the user isn't breathing against the pressure (which is hard).

RWL-69, 2020-03-29:
Normal Breathing Pattern (Normal Respiration Cycle)
[https://www.normalbreathing.org/patterns-normal/](url)

Turn around time in a person is slow in comparison to a drone requirements.  If you look or think about breathing, it isn't instant.  Think in terms of the number of breaths in a minute.  Watching videos and taking a crash course on ventilators has shown me that most motors will work change fast enough.  Normal breathing rates range from 60 breaths per minute (infants) to 12 for adults.  I would look at rates changing in 10 to 50ms would be fast enough.

There is a YouTube video by Medcram that explains how ventilators work with nice drawings of the various breathing cycles under ventilator control.
[https://www.youtube.com/user/MEDCRAMvideos](url)

Motor requirements and control would depend on the mode being used.

In my work, it is usually the other way, looking for pico or nanoseconds, so breathing is in terms of seconds is very slow.

zobotek, 2020-03-29:
I completed a 3D-printed turbine and mounted a 35mm RC plane/quadcopter
motor in it this morning.  I can run tests that characterize its time
constants and pressure/volume delivery.  The speed controller that I am
coupling it with is rated for 20A and outputs a pulse for each motor
revolution for rotational speed measurement and monitoring.  Note that this
motor is compatible with a 2-2kg quadcopter.  The motor torque, closed-loop
controller design and lightweight  (13.3g) 67mm turbine are _much_ more
capable of rapid rotational velocity change than a common computer fan.