jcl5m1/ventilator #75: HEPA FILTERS?

Fearfulbot, 2020-03-26:
According to OSHA 29CFR1910.134, Hepa Filters are considered an alternative when N95 & N100 masks have been depleted.

https://www.sears.com/hoover-genuine-hoover-t-series-exhaust-hepa-filter/p-SPM9466216917

These are normally used in vacuum applications, which is why they're still available

chakahamilton, 2020-03-26:
> According to OSHA 29CFR1910.134, Hepa Filters are considered an alternative when N95 & N100 masks have been depleted.
> 
> https://www.sears.com/hoover-genuine-hoover-t-series-exhaust-hepa-filter/p-SPM9466216917
> 
> These are normally used in vacuum applications, which is why they're still available
> 

> Has anyone tried this? Would they be viable for the PAPR system?
I will be testing them out in a design I am working on. it will use 2 of them  on apposing ends depending on the motors capabilities.

jcrubino, 2020-03-26:
you need h11+ to replicate a N95

H13 filters are fairly standard but there is a risk that the knock off and off brand market are not up to the technical standards of a true Hepa Filter rating at 11+

jcrubino, 2020-03-26:
[Hackaday Article on household filter materials](https://hackaday.com/2020/03/18/homemade-masks-in-a-time-of-shortage/)

Fearfulbot, 2020-03-26:
The term "HEPA vacuum'' is defined as a vacuum which has been designed with a HEPA filter as the last filtration stage. A HEPA filter is a filter that is capable of capturing particles of 0.3 microns with 99.97% efficiency. The vacuum cleaner must be designed so that all the air drawn into the machine is expelled through the filter with none of the air leaking past it."

Is the 99.97% efficiency equivalent to a H11

jcrubino, 2020-03-27:
> Is the 99.97% efficiency equivalent to a H11
  No.  An H11 is rated at 95%.  It was first brought up in the context of an easy to find N95 mask replacement.  We will need a H13 HEPA filter for that comes in at 99.95% total retention rate of 0.3 Micron particles.  This equation becomes a little more difficult if we are to consider the Corona Virus 19 is 0.1 microns in size.  "Better than nothing" and "close" is better than nothing and better than depleting medical supplies from professional healthcare workers and facilities.  Your philosophy may vary.

> The term "HEPA vacuum'' is defined as a vacuum which has been designed with a HEPA filter
   HEPA stands for High-efficiency particulate air AKA high-efficiency particulate absorbing  or high-efficiency particulate arrestance is not defined by the direction of the flow of air.  If this is too liberal a definition let me know.  

The motive for using a HEPA filter is to find easily available parts to create functional respirators and ventilators, not medical grade machines ready to go to market or be put to use in the healthcare setting.

We chose to make ventilators not because it is easy, but because it is difficult and builds a cultural cache of ingenuity using simply available parts.

zobotek, 2020-03-29:
Yes, I've been making bayonet-style 3M 6000/7000 series N100 filter cartridges from HEPA vacuum cleaner bags and 3D-printed ABS frames.  To ensure a sealed surface and low leakage of micro-particles, the frames can be acetone-sprayed or painted.  The filter media I've been using has 3 interior layers of non-woven synthetic fiber surrounded by a synthetic non-woven fiber shell.  It is hot-cut (~190C tool) to the needed shape, sealing the edges.  After cutting, the filter medium is glued to the frame.  The non-foaming polyurethane gule remains somewhat flexible when cured.  The filter medium can be peeled off and the frame reused.  Note that baking an object at 60C/140F for 1/2 hour or so is reported to titrate the SARS coronavirus by a factor of 100,000 or more.   [ https://link.springer.com/content/pdf/10.1007/s00430-004-0219-0.pdf ] .  While this would clearly not kill all bacteria & virus, IMHO it's more than enough to disinfect covid-19, allowing filter reuse without even replacing the filter medium where covid-19 filtration is the objective.  

The 3D-printed air turbines I've constructed and am evaluating for improvised ventilator use are designed to have inlet and outlets filtered by HEPA/P100/N100 filters that are compatible with 3M bayonet masks, whether improvised or factory-built.  

HEPA filter material is considerably more flow-resistant than standard vacuum cleaner filter material.  So much more surface area is needed for the same pressure and flow.  

And yes, there is fraud potential _almost_everywhere_, even in supposedly medical-grade equipment.  While they may not seal well and it's not difficult to understand why an N100 or P100 mask is safer.