jcl5m1/ventilator #76: android app controlling air flow

ta3d, 2020-03-26:
I have written an android app that measures breath activity on airpressure/sound near nose/mouth or in a mask/tube and alarms if breath volume is under a certain threshold. It was designed to detect apnea for sleep laboratory. 
My proposal is to use it for corvid opensource ventilators. Concrete as a controlplane or additional sensor in tube.
I have sourcecode and installable online (released also in GooglePlayStore, but waiting for activation from Google) as well as project proposals and description.
See
http://gitlab.openobjectives.de/gitlab-admin/control-flow

I will continue to work on finetuning measurements and displaying also some kind of intensity curve.

Please review and check if it could be useful for your project.

nasocializes, 2020-03-27:
Johnny would like sensor help as noted in the [To Dos](https://github.com/jcl5m1/ventilator#to-dos-aka-things-i-could-use-help-with):
- Develop a simple low-cost sensor configuration that can provide a similar approximation of lung compliance provided by a ventilator loop display
- Develop Arduino compatible algorithms to align blower cycle to the breathing cycle by looking at backpressure load on the blower, or very simple air pressure sensor that can be inserted into the output of the pump. For ease of developing algorithms see Colab notebook with data visualization.
- Develop a low-cost simple minute volume sensor/approximator. Here is some information about the downsides of being over/under on target volume. Critical care seems to want up to 10-12L/Min.

RWL-69, 2020-03-29:
Looking at ventilator instructions, pressure is a key measurement that will be needed for COVID patients as they are in severe respiratory distress.  From the YouTube videos I have watched on ventilators the pressure needed to get air volume into the patient can tell lots of issues.

With this in mind, I agree to look at the idea of using the pressure sensor to measure the different breathing cycles and control the motor which is what they do in ventilators.