jcl5m1/ventilator #77: Motor controller wire identification.

lebothan, 2020-03-26:
The motor controller shown in the link has on the input side a large red and back wire (presumably supply and ground) and three smaller wires ending in a plug which are not identified. They are brown red and yellow. It appears from the circuit diagram only two are used sig-esc and v-esc.
Can anyone identify which these are, or how to discover?
Thanks

steffes, 2020-03-27:
@jcl5m1 it would be very helpful to remove the electrical tape and take pictures of all wire connections.

nasocializes, 2020-03-27:
Johnny is recommending folks to look at the schematic.
https://github.com/jcl5m1/ventilator#building

lebothan, 2020-03-27:
Johnny is recommending folks to look at the schematic.
https://github.com/jcl5m1/ventilator#building

Which doesn't answer the question does it? If it did I wouldn't have the question.

steffes, 2020-03-27:
Given the following parts how do I wire everything up? I have read @jcl5m1's schematic. 
Do I have to break open the sealed blower to get access to the wires? I also have the same question as @lebothan 

[image]

nasocializes, 2020-03-29:
Hi @lebothan , the 3 wires will correspond to ground, voltage and signal. You'll need to figure out which one responds to what with the part you have. Ground wire should be connected to ground. Leaving only 2 wires as shown in the schematic.

@steffes  it is not recommended to attempt this project if the schematics cannot be understood due to the high risk factor. There are other great projects out there which may be a better fit for folks with various strengths. This project lists some options: https://gitlab.fabcloud.org/pub/project/coronavirus/tracking?fbclid=IwAR1ZptShgjzYM2ifDnxso8rJbFaIrQvOJjA--xrvk3q1Euf4KzQqErWcqLU

closed 2020-03-29 by nasocializes