jcl5m1/ventilator #87: Speaker as alternative blower

madmod, 2020-03-30:
I am completely ignorant of the design constraints here but I think that getting my idea out there quickly to be rejected or improved on is more important than my being an expert.

Could you use a speaker to actuate a bag with valves on it as a precision blower?

It would be much easier to source speakers and amplifiers than the right motors and ESCs and any smartphone or computer could control it. If this isn't the right place for this discussion then I'd appreciate a link to some alternative forums.

jcrubino, 2020-03-31:
This might be a potential high frequency ventilator mechanism where the tidal volumes are very small but the frequency is very high ~ 300 to 900 BPM.

For a normal ventilator you would need to be able to move 6+ liters of air in and out for the average male patient.

axjjienn, 2020-04-01:
It will be really inefficient from a power perspective, and most speakers are not rated to continuous operation.

jcrubino, 2020-04-01:
> High-frequency oscillatory ventilation [Wikipedia](https://en.wikipedia.org/wiki/High-frequency_ventilation)

> In HFOV the airway is pressurized to a set mean airway pressure (called continuous lung-distending pressure) through an adjustable expiratory valve.  Small pressure oscillations delivered at a very high rate are superimposed by the action of a **“loudspeaker” oscillator membrane**. HFOV is often used in premature neonates with respiratory distress syndrome who fail to oxygenate appropriately with lung-protective settings of conventional ventilation.  It has also been used in ARDS in adults, but two studies (the OSCAR and OSCILLATE trials) showed negative results for this indication.

> Parameters that can be set in HFOV include the continuous lung-distending pressure, oscillation amplitude and frequency, I:E ratio (positive-oscillation/negative-oscillation ratio), fresh gas flow (called bias flow), and FiO2.  Increases in continuous lung-distending pressure and FiO2 will improve oxygenation.  Increases in amplitude or fresh gas flow and decreases in frequency will improve ventilation.