jcl5m1/ventilator #94: Connecting brushless ESC to air pump

trackoid, 2020-04-01:
I received the parts listed here... However, how would one connect the brushless ESC to the brushed 12V DC motor of the air pump? (3 connectors on ESC vs 2 on the motor)

Since these are all the recommended parts, I assume there is a way to make it work.

Thanks for the help!

Arthur-Koo, 2020-04-03:
Amone the 3 the entral one is VCC, white/yellow one is PMW signal, and the last one is GND.

trackoid, 2020-04-03:
Thank you Arthur.  My question relates to the connection between the ESC and the motor itself.

The Air pump comes with a brushed motor (2 connectors), and the recommended ESC connection to the motor is for brushless...therefore 3 connectors there.

Wondering if anyone else has solved this yet.

Arthur-Koo, 2020-04-04:
Pls change to a brushless motor with 3 connects. Then u cld control the speed precisely.

zero48, 2020-04-04:
trackoid, I think you mean the actual hardwing from ESC to the Air pump , Which is Invasive. (you may need to open the airpump). I rather use a L298 motor driver which is not invasive. Actually I'm in the process of integrate the Microcontroller and L298 together in a PCB . Ihope this help to clarify your concern

yangoose, 2020-04-08:
I have the same issue connecting the ESC to the Air Pump. Three wires on the ESC and two on the Air Pump. If the solution is to change to a three wire brush-less motor, perhaps the recommended Air Pump should be changed from a two wire model to a three wire model

zero48, 2020-04-08:
> I have the same issue connecting the ESC to the Air Pump. Three wires on the ESC and two on the Air Pump. If the solution is to change to a three wire brush-less motor, perhaps the recommended Air Pump should be changed from a two wire model to a three wire model

I suggest you check https://github.com/jcl5m1/ventilator/issues/62  , That is why I went to this option . I even designed my own pcb . Idea came up from [ Adamcullen](https://github.com/adamcullen) . I hope this help

trackoid, 2020-04-08:
Thanks
I guess we all need a brushed ESC instead or do what you and few others have shared here.