jcl5m1/ventilator #96: Localization initiative from Crowdin

IrynaNamaka, 2020-04-03:
Hi, thanks for the great job you do from all Crowdin team. We'd love to help with translation and make information on ventilator accessible globally. [Crowdin](https://crowdin.com/) gathers all the open-source projects fighting COVID-19, as well as volunteer translators to help with localization management.  
If interested email me at in@crowdin.com
Thanks!

comradekingu, 2020-04-10:
Please don't use Crowdin, it is a lackluster tool in terms of ensuring quality, cumbersome to use, and is closed source software only-as-a-service that spies on users.

I recommend using https://weblate.org/hosting/ which unlike Crowdin offers gratis hosting to all libre software projects, without requiring them to be non-commercial and contribute to their TM.

comradekingu, 2020-04-10:
There are very few libre software on Crowdin, for good reason.
You can tell how much the Crowdin libre software initiative is worth when their representative has an empty account, with faked commits.(!)