jcl5m1/ventilator #100: Oxygen Sensor, why not blood oxygen sensor (oximeter) instead?

leakydata, 2020-04-04:
Since your primary concern is the oxygenation of the patients blood, couldn't you use a solenoid valve on an oxygen tank, which would be controlled by a blood oxygen sensor like the max30100 or max30102. That way you wouldn't have to worry about the oxygen in the flow of air (the inspiration), but instead could just add oxygen, at a constant rate, to the stream until the patients' levels began increasing or hit a threshold set by a clinician. Then the system could continuously open and close the valve on the oxygen tank in response to the patients need; in a feedback loop between the patient and the oxygen tank.

MikeJaworski, 2020-04-04:
2 comments:

1) This is going to be quite laggy

2) All the medical device standards call for direct control of Fi02

jcrubino, 2020-04-05:
A pulse oximeter works by measuring the hemoglobin 02 saturation through light refraction tuned to hemoglobin.

Oxygen sensors for devices are measuring the gas concentration.

sanketkelkar12, 2020-04-06:
I would like to suggest that will it be effective if a feedback from blood oxygen level monitor is taken via Bluetooth and the same could be used to control the speed of blower?

MikeJaworski, 2020-04-07:
See point 1 above.

Response time is going to be >1minute.  Have fun tuning that PID loop.  PS you can't just adjust blower speed to force more air in, you'll rupture the patient's lungs.  You'd need to be much more advanced than that - increasing PIP until a limit is reached, then increasing cycling rate between PIP and PEEP after that.

jcrubino, 2020-04-07:
> Since your primary concern is the oxygenation of the patients blood...

The clinical standard is to get an arterial blood gas which measures paO2, paCO2 and HCO3.  If the patient has chemical poisonings the hemoglobin, specifically the pulse oximeter is not an accurate measure of oxygenation.

marcocamisanicalzolari, 2020-04-19:
There is annoxygen sensor controlled by an App. It could work using an Arduino module to read the value of the sensor

mmcpeck, 2020-04-29:
jcrubino is correct. Pulse oximeters estimate oxyhemoglobin saturation (SpO2) non-invasively. An arterial blood gas (ABG) is an invasive test that measures PaO2 (partial pressure of oxygen in arterial blood) plus other things. There is a relationship between arterial blood oxygen saturation, SaO2 (dependent variable), and PaO2 (independent variable); it is called the oxyhemoglobin dissociation curve. It is an S-shaped, non-linear curve (although the middle is relatively linear). In the upper region of the curve, large changes in PaO2 (which could be diagnostically important in a sick patient) result in only small changes in SaO2. For a patient breathing room air (21% FIO2) a pulse oximeter provides valuable information. But once a patient is on elevated FIO2, the PaO2 can easily be 150 to 300 mmHg or more and the SpO2 doesn't mean as much as changes in PaO2 relative to FIO2. 
     There have been attempts at making miniaturized PO2 electrodes that can be inserted intra-arterially to enable continuous real-time PaO2 readouts, but a reliable device has yet to be developed. All past designs have had difficulty with maintaining accurate measurements over time due to protein buildup over the poles of the micro-Clark electrode or micro-FET. As mentioned by others, real-time SpO2 monitoring is too slow to pick up clinically significant changes in PaO2. And real-time PaO2 monitoring just doesn't work for very long. Thus, closed-loop feedback control of O2 therapy and ventilatory devices has not become an everyday reality at this time. 
     Besides, a ventilator regulates ventilation, which controls CO2 excretion. If you wanted to control the ventilator via feedback from a physiological parameter, you would want to use the arterial carbon dioxide tension (PaCO2) as an indicator since PaCO2 is inversely related to minute ventilation (rate x tidal volume) and directly related to CO2 production. There are devices known as capnographs or capnometers that measure CO2 in respiratory gases and considerable research has been done and is underway with respect to closed-loop control of the ventilator based upon exhaled CO2. However, it turns out to not be quite so simple as there is a multitude of inputs that go into managing a ventilator patient, and doing so on the basis of a couple of physiologic transducer signals ignores many other important factors.
     I'm a respiratory therapist and pulmonary physiologist; most of you are engineers. To solve certain problems, we need each other.  
[image]

mmcpeck, 2020-04-29:
I just spent 20 minutes writing a nice little piece on SpO2, PaO2, the oxyhemoglobin dissociation curve, and closed-loop control of mechanical ventilators. Somehow it got lost in the ether. I'm too tired to do it again.  Anyway, at the end I said I'm a respiratory therapist and pulmonary physiologist and many of you are engineers. To solve many of these issues, we need each other.