jcl5m1/ventilator #101: Moisture or a moisturizer is needed in any ventilator system.

leakydata, 2020-04-04:
Moisture is a necessity in any ventilation system, because it is important that the lung tissue is not dried out. This issue is usually handled by hardware in the system that adds moisture, or special inline filters or devices. Are there plans to manage moisture in the system?

MikeJaworski, 2020-04-05:
Could an off-the-shelf nebuliser be teed into the flow path?  They are cheap(ish) at about £50 and already designed / approved for the job.

For example:

https://www.philips.co.uk/c-p/HH1340_00/innospire-essence-compressor-nebulizer-system

jcrubino, 2020-04-06:
You do not want to have to rely on nebulization for humidification.
COVID and other respiratory pathogen illnesses require hospitals to limit the amount of aerosol produced and in contact with the patient.

A heater water device is most desirable especially for secretion clearance, HME (Heated Moisture Exchanger) is suitable for keeping the internal respiratory tract humidified.  HME are usually a filterpaper device in the breathing circuit close to the ET tube.

jcrubino, 2020-04-07:
@Col68 congratulations on your progress.
I believe you are telling us you are close to finishing a PAPR (Positive Air Pressure Respirator).

For the purposes of clairty,  using a "nebulizer" for humidity is not ideal for ventilator purposes.
A medical nebulizer will create excess vapor that is only useful for short periods of time (medication delivery).  Patients with ARDS and COVID do not need excess vapor entering their respiratory tract as it will consolidate into macro water droplets hindering their already "wet" lung from the immune response.  Additionally, it could create a contact vapor hazard in the room where healthcare workers are.  The healthcare workers will absolutly need a PAPR with respiratory droplet grade filtering in such a situation.