jcl5m1/ventilator #105: Tesla Seems to have a good design

leakydata, 2020-04-07:
As outlined in the video they released today, the Tesla ventilator design seems to address all the issues I have seen on other DIY ventilator sites. I think everybody working on ventilator projects should check it out. 

https://www.youtube.com/watch?v=zZbDg24dfN0

DaveM135, 2020-04-07:
Agree.   Looks pretty impressive

JonTheGreek, 2020-04-08:
Don't know much about 'em, but the sounds that it makes sure sounds legit!

AndrewMbugua, 2020-08-07:
Elon Musk inspires us how make use of everything and never leave anything to go to waste.

AndrewMbugua, 2020-08-07:
And the fact that a university can make them should encourage us

[IEEE member news](https://spectrum.ieee.org/news-from-around-ieee/the-institute/ieee-member-news/nairobi-ieee-students-design-a-lowcost-ventilator-using-locally-sourced-materials)