jcl5m1/ventilator #106: Help with PAPR build please

JonTheGreek, 2020-04-08:
So I have acquired all the parts for the build and I have a some questions regarding the DC brushless motor controller. Is the 5 VIN on the schematic the red (+) wire coming out of the board? Secondly, is the white wire some kind of signal wire? Was also curious if the schematic is mimicking the wires coming out of the board or not.

JonTheGreek, 2020-04-08:
P.S. I work in a dental office as a surgical assistant and I wanted to say thank you to all who contribute their knowledge in designing and engineering to evolve any kind of these ideas! I have been very grateful for stumbling upon Johnny Lee's YouTube video and leading me here.

CoogyEoin, 2020-04-09:
So the motor is controlled by applying a voltage to one of the instructors in it. So using three signals that are out of phase we can move the motor (like dragging a magnet around in circles). Each of the wires is a different signal in a different phase. The picture below shows it a bit better.
[image]