jcl5m1/ventilator #109: PAPR current design issues.  And have you considered cyclonic separation as part of design

oiaohm, 2020-04-10:
1 random electrical tape very bad idea.   Depend on brand of electrical tape this stuff can be treated with different anti rodent chemicals including stuff like Warfarin (Coumadin) so if you want someone  to be using something and end up bleeding out out eyes use any random electrical tape in a breathing air system.   Also some brands of electrical tape have different Flame Retardant chemicals some not human friendly because they can be cancer agents with long term direct exposure.   Also there is a solvent problem from the glues they can use.  Finally some brands will give you a lead problem as well.

There is HVAC/R rated electrical tape there is also the HVAC/R rated clear and duct tapes as well that when in contact with air flow are safer than the HVAC/R electrical tape.   So even when you are using the right tapes you are not meant to use the electrical tape to seal up general air leaks that is what the clear HVAR/R tape or duct tape is for.   Yes for a prototype using the wrong tape I can understand but you should not keep on using the wrong tape long term.  Basically when it comes it air stuff that humans/animals will be breathing electrical tape is only for electrical stuff you have other tapes for the duct work to guide the air flow and it important to use them.

Also don't forget HVAC/R is not any old electrical tape or any old tape for the duct work.

2. you want molded parts if possible.   So parts need to be design with the concept they will be molded at some point.  The surface of 3d prints is not ideal for this usage its normally too rough and allowing of bacteria growth.   Yes in emergencies when you can disregard basic safety they will allow 3d printed parts. 

3.  Question what is the required airflow.  I guess general 120mm computer fan cannot provide it?

4.  Common issue with PAPR systems is when using them in less than clean air that there operational time between filter changes is reduced massively.  In theory there is a solution to this.

https://en.wikipedia.org/wiki/Cyclonic_separation

Do notice how many bagless vaccum cleaners are in fact using Cyclonic separation.   So if first stage filtration was Cyclonic this would reduce the material the rated filter material would have to trap.

Yes the idea of using hepa filters out of vuccum cleaners miss that you have a stage 1 filteration of either the bag or cyclonic separation before the hepa filter this is important to long time operation of that filter.

Yes people sourcing and converting filters design for other things for masks are normally losing the stage 1 filtration this results in that the material will be taking large material than it designed to.

Adding a stage 1 filtration to even to filters designed for masks will improve functional time because they will not come packed with material as fast.

Lets not just try to make a PAPR lets try to make a PAPR that has a feature that most of the common ones are missing. 

There are many materials that can get to  N95/p2 and above like thick towelling but these require some from of forced air system to make them functional.

3dseed, 2020-04-10:
@oiaohm GREAT comments..!!
So going with a forced air system and thick toweling could be a solution ?
Currently we can't even get filters here in Barcelona.. desperately looking for other options.
So we just need the right size air pump and thickness of towel ?
We have the mask files online here.. but need to source the components before we can make the [silicon mask](https://3d-seed.com/open-source-masks/)..

jcrubino, 2020-04-10:
@3dseed [Blue Shop Towels](https://www.businessinsider.com/homemade-mask-using-hydro-knit-shop-towel-filters-better-2020-4) are getting some traction for filter materials

oiaohm, 2020-04-11:
> @oiaohm GREAT comments..!!
> So going with a forced air system and thick toweling could be a solution ?
> Currently we can't even get filters here in Barcelona.. desperately looking for other options. [...]

https://www.researchgate.net/publication/258525804_Testing_the_Efficacy_of_Homemade_Masks_Would_They_Protect_in_an_Influenza_Pandemic

Yes it could be a solution to use fabric towelling of the right weave and density.  Like this study with the tea towel towelling using 2 layers got you to a n95.   Problem is the air resistance to breathing not the filtering.   Of course you may want to use more layers for longer operational duration before filter is breached that going to equal more resistance.

Please note you will need the testing gear as each weave of towelling has different filtering behaviour.

> @3dseed [Blue Shop Towels](https://www.businessinsider.com/homemade-mask-using-hydro-knit-shop-towel-filters-better-2020-4) are getting some traction for filter materials

Sorry to say jcrubino those shop towel are not anywhere near a close to n95 it does not take much to win against a surgical mask on face because they were not design to filter the air you breath in only out.   Material of a surgical mask if you can get it to fit well beat shop towel hands down also comes a hindrance to breathing.

Note the 2013 one went back to 50 percent of a n95 mask on filtering to prevent hindrance to breathing.

Your standard n95 mask materials are particular designed to be low hindrance to breathing while providing the filtering.

Now if we can remove the hindrance to breathing problem with a motor/blower the list of materials that can do N95/R95/P2 filtering and above increases massively.   Of course the resistance is going to cause to reduced battery time in operation.

If you are using towelling I would be recommending Cyclonic separation pre filter so large dust in air are not going into the towelling and wedging the fibers open.

I really don't see why cyclonic separation cannot be used air in take of blower and filter placed after blower.

Lot of ways I can see take your hepa cyclonic separator vaccum cleaner as a model.   Some of those with a washable fabric filter is still above a n95 heck above a r95/p2 in filtering.   When you look at your standard on market PAPR they are not designed this way.   There is no as much profit making a item that has washable filters.

Yes washable filters at n95/r95/p2 and above are not 100 percent safe for a person to use directly due to too much air resistance.   Now powered is a completely different item here as the electronic motor gets to over come the resistance problem.

Please note I don't say what I am suggesting is ideal.   But I am willing to trade battery life/operational time frame to in fact have a operational mask that is not trying to kill me it self and will filter out what is required.

I am after a operational mask not to protect me from covid-19 but to protect me from oil sprays that I have to spray on crops to protect them from pests.   The nightmare is I cannot get p3 or p2 masks here in Australia for that job.  So I am back to old school fabric towelling masks with their risks to life due to excessive air resistance.  Yes hospitals have taken most of the supplies that go to agriculture here. 

So a good design PAPR using fabric towelling would be safer than what I am currently using.

Yes being used in agriculture  the air will not always be dust free like a hospital.

chonghorizons, 2020-04-11:
This is helpful. And, I have a question. My main concern is cost and complexity. Do you know of an easy $20 or less cyclonic configuration? Preferably without 3d printing (I know I'm making wild/hard design demands).  Maybe with two PVC tubes, one inside the other?

The other thing to note is this came out 4 days ago and avoids the filter issue with what they say are readily available hospital viral filters: 

** MAIN HALO DESIGN **
https://www.youtube.com/watch?v=JMXP8jAmDKA&feature=youtu.be
and 
https://3dprint.nih.gov/discover/3dpx-013544

chonghorizons, 2020-04-11:
>>> Question what is the required airflow. I guess general 120mm computer fan cannot provide it?

A 120mm computer fan is probably okay. We have some links (on the Halo, Washington state youtube comments) that says that 20 L/min (1/2 cfm) may be sufficient, but that maximum respiratory load is more like 4-15cfm.  We also suggested a "trademill test" that monitors O2 as an acceptable way to test if a PAPR is sufficient.  

See CoronavirusNotAlone comment that includes this:

##### Youtube comment
f you want to focus on respiration rate, 4cfm is what we found.   Quoted:  4-15cfm is what is needed for human breathing (https://www.google.com/search?q=human+breathing+cfm)   Hard to get exact numbers.  This has a chart for breathing peak by age: https://biology.stackexchange.com/questions/5466/what-is-the-inlet-outlet-speed-of-air-going-in-out-of-our-mouth-during-inhalatio