jcl5m1/ventilator #110: FYI: Washington Hospital version of improvised PAPR

chonghorizons, 2020-04-11:
I'm trying to get a hold of you about your PAPR designs. Specifically, I'm organizing a group of (loosely vetted) people who are seriously working on improvised PPE. 

**Halo design video, and 3-d print files**
Specifically, these doctors/HCP solved a lot of the PPR issues:  https://www.youtube.com/watch?v=JMXP8jAmDKA&feature=youtu.be

Innovations:  existing inline viral filters (widely available in hospitals) and use of disposable plastic bag for hood.

I'm hocho on this discord server: https://discord.gg/H3jUmar  (all welcome, and expect to be vetted lightly)
or chongare@gmail.com or 607-279-3623 (text/phone)

In it to make an impact and move the needle. Working 12hr days 7 days a week. Mainly, i'm organizing and connecting people.

chonghorizons, 2020-04-11:
Here is a picture.
[image]

closed 2020-04-28 by chonghorizons

StevenACoffman, 2020-08-14:
Have you seen this open source design: https://making.engr.wisc.edu/papr/

chonghorizons, 2020-12-16:
Yeah. The madison badger design is great. It is on the more complex and expensive side.

[Http://BunnyPAPR.org] is streamlined down to a production cost of under $30. 

I actually have it under $10 now by cutting out the 3D parts.