jcl5m1/ventilator #123: I got inspired from your ventilator

magicventilator, 2020-07-23:
Johnny I got inspired from your ventilator and I built my own. Let me know what you think of this one: http://magicventilator.com/

Here is some information on how it works:
http://magicventilator.com/2020/07/12/how-the-magic-ventilator-works/

Let me know what you think?!

devyaz, 2020-07-23:
Awesome 👍 keep it up

magicventilator, 2020-07-24:
Thanks @devyaz 👍

devyaz, 2020-07-24:
Have you tested this on yourself? I mean if say patient is failing to breathe normally how does it detect that? Sp02?

magicventilator, 2020-07-24:
@devyaz The way I detect patient breathing is when there is a drop of pressure on the circuit so if I do not detect pressure drops on the circuit I know the patient stoped breathing. Also I constantly monitor the pressure and flow in the circuit and when the pressure goes too high or low for too long or the flow goes to high or too low I know something is going wrong and I can turn on an alarm